iobroker-community-adapters/ioBroker.fronius #7: Fehlende Werte bei CommonInverterData

tobintax, 2017-08-18:
Werte von FAC/IAC/IDC/PAC/UAC/UDC werden nicht angezeigt bzw. sind auf 0 gesetzt. 

Die Ursache liegt irgendwo hier begraben: 

`                    if("PAC" in data) {
                        adapter.setState("inverter." + id + ".FAC", {val: resp.FAC.Value, ack: true});
                        adapter.setState("inverter." + id + ".IAC", {val: resp.IAC.Value, ack: true});
                        adapter.setState("inverter." + id + ".IDC", {val: resp.IDC.Value, ack: true});
                        adapter.setState("inverter." + id + ".PAC", {val: resp.PAC.Value, ack: true});
                        adapter.setState("inverter." + id + ".UAC", {val: resp.UAC.Value, ack: true});
                        adapter.setState("inverter." + id + ".UDC", {val: resp.UDC.Value, ack: true});
                    }else{
                        adapter.setState("inverter." + id + ".FAC", {val: 0, ack: true});
                        adapter.setState("inverter." + id + ".IAC", {val: 0, ack: true});
                        adapter.setState("inverter." + id + ".IDC", {val: 0, ack: true});
                        adapter.setState("inverter." + id + ".PAC", {val: 0, ack: true});
                        adapter.setState("inverter." + id + ".UAC", {val: 0, ack: true});
                        adapter.setState("inverter." + id + ".UDC", {val: 0, ack: true});
                    }`

Wenn ich im Else Statement das value auf "1" setze, dann wird dies auch übernommen. Aus irgendeinem Grund springt er also direkt zu Else. In der JSON Ausgabe ist "PAC" enthalten. Habe es testweise auch mit anderen Werten versucht, leider unverändert. 

JSON Output: 

{
   "Body" : {
      "Data" : {
         "DAY_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 45580
         },
         "DeviceStatus" : {
            "ErrorCode" : 0,
            "LEDColor" : 2,
            "LEDState" : 0,
            "MgmtTimerRemainingTime" : -1,
            "StateToReset" : false,
            "StatusCode" : 7
         },
         "FAC" : {
            "Unit" : "Hz",
            "Value" : 49.979999999999997
         },
         "IAC" : {
            "Unit" : "A",
            "Value" : 27.41
         },
         "IDC" : {
            "Unit" : "A",
            "Value" : 13.1
         },
         "PAC" : {
            "Unit" : "W",
            "Value" : 6261
         },
         "TOTAL_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 5665530
         },
         "UAC" : {
            "Unit" : "V",
            "Value" : 234.90000000000001
         },
         "UDC" : {
            "Unit" : "V",
            "Value" : 488.39999999999998
         },
         "YEAR_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 5665535.5
         }
      }
   },
   "Head" : {
      "RequestArguments" : {
         "DataCollection" : "CommonInverterData",
         "DeviceClass" : "Inverter",
         "DeviceId" : "1",
         "Scope" : "Device"
      },
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2017-08-18T16:18:19+02:00"
   }
}

ldittmar81, 2017-08-20:
Hi tobintax - vielleicht solltest du mit:

if('PAC' in resp)... probieren. Das sollte funktionieren ;-)

tobintax, 2017-08-20:
Hi, funktioniert. Vielen Dank. Pflegst Du es ein oder soll ich es beim nächsten Update mit einpflegen?

ldittmar81, 2017-08-21:
Hi,
du kannst es beim nächsten Update einpflegen. Bin z.Z. Land unter :tired_face:

closed 2017-08-21 by ldittmar81