iobroker-community-adapters/ioBroker.fronius #12: [Request] Moeglichkeit mehrere strings wieder zu geben anstatt nur P-Totaal ?

DutchmanNL, 2017-09-29:
Hi, danke fuer den adapter functioniert super !
Frage: waere es moeglich anstaat nur das P-totaal auch die verschieden ringen wieder zu geben ?
Beispiel: bei meiner anlage benuzten wir 2 strings wodurch ich auf fronius einen wert Mpp1 / Mpp2 und Mpp total bekomme.

Der adapter zeigt im moment nur diesen total wert und nicht die einzelnen stringe. Die data sind interessant weil man damit ermitteln koennte ob ein string ausgefallen ist und evenuteel durch zuviel schatten oder gleich nicht rendable ist.

[image]
[image]
[image]

greetz,

Dutch

tobintax, 2017-09-29:
Hi Dutch, 

alle Werte werden derzeit aus den Livedaten gezogen. Die Abfrage der einzelnen Strings ist bei dem Fronius Inverter über die Livedaten nicht möglich. 

Die Daten müssten aus den Archivdaten gezogen werden. Ich frage diese Stand heute über andere Mittel ab, visualisiere diese aber nicht in IOBroker. 

Allgemein war ich schon einmal dabei, die Archivdaten über den Adapter abzufragen, bin aber nicht fertig geworden. Ich bräuchte wohl etwas Unterstützung oder mehr Zeit. 

Grüße

Tobintax

tobintax, 2017-11-12:
Hi L, 

ich habe es mir angesehen. Du hast ja gute Vorarbeit geleistet. 
Ich habe nur einen Symo 8.2.3, dort erhalte ich lediglich folgende Channels: 

> 
			"channels" : 
				{
					"1046" : 2,
					"13" : 1510516008,
					"22" : 1,
					"65558" : 1,
					"66582" : 0,
					"68419848" : 4760,
					"68485384" : 8317597,
					"68550920" : 8317589.5

Zumindest sind die wenigen Channels die ich habe von der Bedeutung her, mit deinen Identisch. 
Ich weiß nicht ob es an deinem Modell liegt dass da mehr bei rum kommt. Hast Du evtl. eine Sensorcard verbaut? 

Grüße

Tobintax

tobintax, 2017-11-12:
Deine Tabelle mit der Beschreibung der einzelnen Werte ist am Ende leider abgeschnitten (z.B. Votage_AC_PhaseToPhase_). Kannst Du mir bitte mal die vollständige Tabelle bereitstellen? 

Du schreibst ja selbst dass das ganze undokumentiert ist. Die Frage ist, wie lange das ganze seitens Fronius so bleibt. Ich werde trotzdem mal einen Versuch wagen, es einzubauen.

tobintax, 2017-11-24:
Bin ziemlich fertig mit der Implementierung. Fehlt mir nur noch deine Tabelle @R000795

R000795, 2017-11-25:
Sorry, hatte leider keine Zeit in den letzten Tagen. Die Werte sind nicht abgeschnitten, da ich leider die Phasen noch nicht ermitteln konnte, welche das sind. Ich werde in den nächsten Tagen nochmal einen Anlauf starten, die Channels zuordnen zu können.

Vielen Dank schon mal für die Implementierung!!!

tobintax, 2017-12-02:
@R000795 : ok, Danke. Wenn ich die Tabelle habe, ist es in wenigen tagen fertig.

tomac01, 2018-03-03:
Wie sieht es denn aus mit der Tabelle von R000795? Könne wir bald mit der Erweiterung rechnen?
VG

tobintax, 2018-03-03:
Hi, habe seit 4 Monaten nichts mehr gehört. Habe die Entwicklung diesbezüglich eingestellt. 
Jeder der Interesse an dieser Erweiterung hat, hat die Möglichkeit die Daten zu liefern. Da aber niemand geliefert hat gehe ich davon aus, dass die Nachfrage danach einfach nicht da ist.

DutchmanNL, 2018-04-14:
Hi tobinax,

Es tut mir leis ich war Lange Zeit offline... :/
Wen ich es richtig verstehe brauchst du die ausgabe dieser json Von unseren Gerät.
Habe die discussion nicht ganz verstanden was kann ich machen um zu helfen ?!

~Dutch

tobintax, 2018-12-18:
Hi @DutchmanNL, 

der letzte Stand ist hier nachzulesen. @R000795 wollte die komplette Tabelle liefern, hat es aber seit 1 Jahr nicht getan. Seit 2017 gibt es lediglich 3 User, die das angefragt haben und niemand hat die Info's  geliefert. Zusätzlich ist diese Abfrage seitens Fronius undokumentiert. Ich werde hier also keine weitere Zeit investieren.

Aare07, 2020-10-03:
Hallo Tobintax und Danke für den Adapter, funktioniert gut.!
Das Einzige was etwas stört ist dass ich auch die Warn Meldungen erhalte (cannot read property, valiue undefined) in anderem Issue behandelt.

Ich habe definitiv Interesse beide Strings im ioBroker zu sehen.  Entschuldige, bin halt eher der Endanwender und schaue zu wenig ins Git...
Im Attachement was ich bei meinem Symo 15.0-3-M  mit Datalogger Card als Resultat von  /components/inverter  erhalte:
[Inverter-result.txt](https://github.com/iobroker-community-adapters/ioBroker.fronius/files/5321424/Inverter-result.txt)

Sieht mir jedoch nicht danach aus als ob die einzelnen Daten der Strings gezeigt würden :( 

Entschlüssel konnte ich: aktuelle Leitung "67109125" : 2845,
Leistung total heute "68419848" : 3870,
Plus noch Leistung im Jahr unt Leistung total

gibts weitere Abfragemöglichkeiten für an die String Daten zu kommen?

Dreistein19, 2020-10-03:
Ich nutze Modbus zum Auslesen der Werte die der Adapter nicht liefert.
Holding Register z.B. 40282_1_DCA
Infos dazu muss man auf der Fronius Seite suchen. 
Weiter Infos hier zu posten, würde den Rahmen sprengen.

Aare07, 2020-10-05:
Danke, werd ich machen, kann entsprechend geschlossen werden.

pacco81, 2021-03-14:
Nur zur Doku wenn hier nochmal jemand landet bzgl. dem Aufruf von components/Inverter
Bei meinem Symo 15 Baujahr 2020 bekomme ich leider keine Daten zu den einzelnen String sondern nur die üblichen Summendaten:

},
				"channels" : 
				{
					"1046" : 7,
					"13" : 1615711495,
					"1303" : 24000000,
					"22" : 1,
					"65558" : 1,
					"66582" : 0,
					"67109125" : 4046,
					"68419848" : 5271,
					"68485384" : 1477629.375,
					"68550920" : 2309990

nkleber78, 2023-06-04:
Close as not availlable as live data. Part of archive data the points are integrated

closed 2023-06-04 by nkleber78