iobroker-community-adapters/ioBroker.fronius #19: Warnmeldung TypeError: Assignment to constant variable seit update 1.0.3

mikewolf, 2018-12-18:
Hallo schreibe hier für nen guten Freund:
https://forum.iobroker.net/viewtopic.php?f=20&t=9704&p=205461&hilit=fronius#p205461
hallo habe ein problem:
ich habe beim Fronius Adapter ein update gemacht auf 1.0.3,
und jetzt habe ich immer die Warnmeldung TypeError: Assignment to constant variable.Ich brauche nur den wert wen ins netz eingespeist wird .
Ich bin drauf gekommen das das der wert P-Grid ist,
Doch bei netzbetzug ist der wert normal und ich kann in auslesen, und in eine SV schreiben
Doch wen eingespeist wird ist der wert minus, und ich kann ihn nicht mehr in eine SV schreiben.
Weis jemand wie mann das problem umgehen könnte.

DutchmanNL, 2018-12-18:
Was für eine Anlage ist das ?
Ich kennen fronius nur von Strom erzeugung Dan wundert mich der minus wert ?

mikewolf, 2018-12-18:
hallo ,
Fronius simo ,
primär geht es um den Fehler
Assignment to constant variable 

unter powerflow gibts 3 Werte p-Grid,Pload,ppv
- bei pload bei bezug vom stromlieferanten
+ bei pload bei ertrag 
+ pgrid bei bei bezug von stromlieferanten
- pgrid bei solarbetrag
habe auch einen s0 zaehler im fronius hinterlegt
 dieser beitrag ist aus 2017

tobintax, 2018-12-18:
Hallo, 

ich kann euch nicht folgen. Was ist das konkrete Problem und mit welchem Wechselrichter?
Wie kann ich es reproduzieren? 

Ich habe gerade ein Update auf 1.0.3 gemacht und es funktioniert bei mir einwandfrei.

tobintax, 2018-12-18:
Hier noch ein Ausschnitt aus der Fronius Doku: 

# mandatory field
#this value is null if no meter is enabled ( + generator , - consumer )
number P_Load;

P_Load kann also sehr wohl Minus sein.

commit 83a64537 (2018-12-20): This fixes #19 and fixes #20
closed 2018-12-20 by ldittmar81