iobroker-community-adapters/ioBroker.fronius #22: Think about to fix the issues found by adapter checker

ioBrokerChecker, 2019-02-03:
I am an automatic service that looks for possible errors in ioBroker and creates an issue for it. The link below leads directly to the test:

https://adapter-check.iobroker.in/?q=https://raw.githubusercontent.com/iobroker-community-adapters/ioBroker.fronius

- [ ] Title should not have word "adapter" in the name. It is clear, that this is adapter. Now: Fronius inverters adapter
- [ ] Bluefox is not in the collaborators!. Please add.

Thanks,
your automatic adapter checker.

ldittmar81, 2019-02-15:
fixed wuth commit https://github.com/iobroker-community-adapters/ioBroker.fronius/commit/2b7714ab4fe5de442b95911f7a0a73537ab5d37d

closed 2019-02-15 by ldittmar81

ioBrokerChecker, 2019-02-27:
Thanks, that all bugs have been fixed.