iobroker-community-adapters/ioBroker.fronius #24: Fronius Webinterface reboot

dietlman, 2019-03-07:
WEnn ich den fronius adapter installiere und die refresh-zeit auf weniger als 40 Sekunden einstelle dann startet der Wechselrichter ca. alle 3 Minuten neu. Gibt es da eine Möglichkeit das eventuell zu verbessern?
Vielen Dank!

ldittmar81, 2019-03-07:
Sorry, aber ich habe kein Fronius mehr und kann es also nicht testen. Ich denke aber es liegt am Wechselrichter. Er ist wahrscheinlich mit den Abfragen total überfordert. Dass er deswegen ein Reboot macht ist aber schon der Hammer. Das dürfte nicht sein und die Entwickler des Wechselrichters sollten das abfangen... naja... Vielleicht geht es per Modbus besser. Da müsste sich entweder einer ein neuen Adapter entwickeln, der über Modbus geht oder du müsstest dir den Modbus Adapter installieren und schauen ob es besser geht.

dietlman, 2019-03-07:
Danke für die rasche Antwort, ich denke auch dass ein Reboot nicht
passieren sollte, es sieht so aus als ob das Interface einfach abschmiert.
Modbus wäre auch mein nächste Ansatz gewesen, nur leider kenn ich mich
damit gar nicht aus. Modbus scheint ja sehr komplex zu sein...

Manfred DIETL
phone: +43 664 8497777
manfred@dietl.us
www.dietl.us

nkleber78, 2023-05-31:
Das Problem ist nicht reproduzierbar. Nur bei sehr hoher Abfragerate (im Bereich von <10s) kann es sein dass das WR interface nicht mehr reagiert -> FW issue bei z.b. Symo