iobroker-community-adapters/ioBroker.fronius #25: Think about to fix the issues found by adapter checker

ioBrokerChecker, 2019-03-12:
I am an automatic service that looks for possible errors in ioBroker and creates an issue for it. The link below leads directly to the test:

https://adapter-check.iobroker.in/?q=https://raw.githubusercontent.com/iobroker-community-adapters/ioBroker.fronius

- [ ] [E605] No actual year found in copyright. Please add "Copyright (c) 2019 [object Object]" or "Copyright (c) 20xx-2019 [object Object]" at the end of Readme
- [ ] [E701] No actual year found in LICENSE. Please add "Copyright (c) 2019 [object Object]" or "Copyright (c) 20xx-2019 [object Object]" at start of LICENSE

Thanks,
your automatic adapter checker.

ioBrokerChecker, 2020-01-28:
Do you need help fixing the bugs?

mcm1957, 2023-05-31:
obsolete - will be checked with 2.0.0 release building anyway

closed 2023-05-31 by mcm1957

ioBrokerChecker, 2023-11-30:
Thanks, that all bugs have been fixed.