iobroker-community-adapters/ioBroker.fronius #34: [FeatureRequest] SOC BYD Akku

Strahler85, 2019-12-05:
Ein BYD Akku der an einen Fronius Hybrid Wechselrichter angeschlossen ist kann nicht ausgelesen werden.

Mit /solar_api/v1/GetPowerFlowRealtimeData.fcgi kann man aber den SOC Zustand der Batterie auslesen. Leider wird diese Information nicht im IOBROKER angezeigt.

{
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "Battery_Mode" : "suspended",
               "DT" : 99,
               "E_Day" : 5269,
               "E_Total" : 311282.90625,
               "E_Year" : 311282.1875,
               "P" : 0,
               "SOC" : 4.1999998092651367
            }
         },
         "Site" : {
            "BatteryStandby" : true,
            "E_Day" : 5269,
            "E_Total" : 311282.90000000002,
            "E_Year" : 311282.20000000001,
            "Meter_Location" : "grid",
            "Mode" : "bidirectional",
            "P_Akku" : null,
            "P_Grid" : 526.20000000000005,
            "P_Load" : -526.20000000000005,
            "P_PV" : null,
            "rel_Autonomy" : 0,
            "rel_SelfConsumption" : null
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2019-12-05T08:11:24+01:00"
   }
}

ldittmar81, 2019-12-05:
Hi... Danke für die Info. Ich habe leider kein Fronius mehr und gerade ziehe ich um, aber ich kann es bei Gelegenheit (wahrscheinlich erst Januar 2020) einbauen und zum Testen freigeben. Sollte jemand anderes schneller sein und mir ein entsprechenden PR stellt, schaue ich es mir gerne an und gebe es frei :-)

nkleber78, 2020-11-26:
Das ist gelöst mit der V1.1 im latest channel. Bitte testen und schließen wenn ok

mcm1957, 2024-07-24:
no (negative) feedback - so lets close here

closed 2024-07-24 by mcm1957