iobroker-community-adapters/ioBroker.fronius #35: [FeatureRequest]  Werte der MPP auslesen. #enhancement

Dreistein19, 2020-01-25:
Hallo,

Ich habe gelesen das du keinen Fronius mehr hast. Das macht die Entwicklung natürlich nicht einfacher.
Vielleicht ist aber doch möglich,  den Code so anzupassen das es möglich wird, 
die einzelnen "Strings" der Solaranlage auslesen zu können.
Bei uns ist es ein Primo 4.01 und wir haben "hinten" weniger Panels als "vorne".

nkleber78, 2020-11-26:
Hallo @Dreistein19 
kannst du mir sagen von welchen Datenpunkten auf der SolarAPI du sprichst? Wenn die Daten auf der API verfügbar sind dann bitte hier die notwendigen Datenpunkte beschreiben. Dnke

Dreistein19, 2020-11-27:
Hallo, leider weiß ich nicht ob die API die Werte überhaupt liefert. Ich frage sie per Modbus ab.

modbus.0.holdingRegisters.40282_1_DCA liefert z.b den Strom vom MPP 1.
modbus.0.holdingRegisters.40302_2_DCA z.B den Strom vom MPP2.

Damit kann man z.b. sehen wie bei einer Ost West Anlage die Stromerzeugung sich mit dem Sonnenstand ändert.
Ich finde das sehr Interessant.

nkleber78, 2020-11-27:
Diese Datenpunkte solltest du über IDC und IDC_2 sowie UDC, UDC_2 bekommen und sind in der aktuellen Version enthalten...

Dreistein19, 2020-11-28:
Leider liefert der Adapter diese Datenpunkte nicht. Ich habe einen Fronius Primo (Nur Wechselrichter).
Ich habe die Adapter Datenpunkte gelöscht und gehofft das die fehlenden Datenpunkte neu angelegt werden.
Leider nicht. 
Auch ein "händisches" Anlegen der Datenpunkte brachte keine Werte.
Mir ist auch aufgefallen, das bei einigen Datenpunkten keine "Units" (Maßeinheiten) angelegt werden.
Was ist z.b. der Punkt "DT" unter fronius.0.powerflow.inverter1.DT ?
MfG

nkleber78, 2020-11-29:
Was der Primo kann, weiß ich nicht. Aber schau dir mal die Doku der solar API an. Dort werden die API calls sowie auch die Bedeutung der Datenpunkte beschrieben. https://www.fronius.com/~/downloads/Solar%20Energy/Operating%20Instructions/42%2C0410%2C2012.pdf Falls die datenpunkte tatsächlich verfügbar sind bei dir, dann schick mir deinen Output von http://deineIP/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DataCollection=CommonInverterData und schau ob du im Log warnungen erhälst. Aber ich denke dass dein Inverter das einfach nicht liefert. Wie gesagt, zuerst bitte schauen was bei dir über die API verfügbar ist...
Bzgl. Units: Jeden datenpunkt manuell anzulegen ist viel Aufwand, daher werden diese Automatisch erstellt, allerdings liefert die API die meisten units nicht. Diese müssen dann manuell nachgepflegt werden.

Dreistein19, 2020-12-05:
Ich habe eben "mit Gewalt"  die Version 1.1.2 Installiert.  Die wurde mir irgendwie nicht als update von 1.1.1 angezeigt.
Ich habe dann eine Installation über die "Katze" durchgeführt.
Die Werte für die MPP 1&2 werden nun angezeigt.
Vielen Dank für deine Arbeit.

pacco81, 2021-03-14:
Hi ! 
Bei mir läuft auch V1.1.2 vom Adapter und liest den Symo15 aus. Läuft super.

Die Archivdaten und insb. die Daten zu MPP 1&2 kommen bei mir aber leider nicht an.
@Dreistein19 : Wie sehen die Daten bei dir aus, wenn sie im iobroker ankommen ? Meine Objektliste sieht so aus:
[image]

Folgender Aufruf gibt die String-MPPT-Daten aus: `http://10.11.12.92/solar_api/v1/GetArchiveData.cgi?Scope=System&StartDate=13.3.2021&EndDate=13.3.2021&Channel=Current_DC_String_1&Channel=Current_DC_String_2`

{
	"Body" : 
	{
		"Data" : 
		{
			"inverter/1" : 
			{
				"Data" : 
				{
					"Current_DC_String_1" : 
					{
						"Unit" : "A",
						"Values" : 
						{
							"0" : 0,
							"10200" : 0,
							"10500" : 0,
							"10800" : 0,
							"11100" : 0,
							"11400" : 0,
							"11700" : 0,
							"1200" : 0,
							"12000" : 0,
							"12300" : 0,
							"12600" : 0,
							"12900" : 0,
							"13200" : 0,
							"13500" : 0,
							"13800" : 0,
							"14100" : 0,
							"14400" : 0,
							"14700" : 0,
							"1500" : 0,
							"15000" : 0,
							"15300" : 0,
							"15600" : 0,
							"15900" : 0,
							"16200" : 0,
							"16500" : 0,
							"16800" : 0,
							"17100" : 0,
							"17400" : 0,
							"17700" : 0,
							"1800" : 0,
							"18000" : 0,
							"18300" : 0,
							"18600" : 0,
							"18900" : 0,
							"19200" : 0,
							"19500" : 0,
							"19800" : 0,
							"20100" : 0,
							"20400" : 0,
							"20700" : 0,
							"2100" : 0,
							"21000" : 0,
							"21300" : 0,
							"21600" : 0,
							"21900" : 0,
							"22200" : 0,
							"22500" : 0,
[... 551 lines cut ...]
		{
			"Channel" : 
			[
				"Current_DC_String_1",
				"Current_DC_String_2"
			],
			"EndDate" : "2021-03-13T23:59:59+01:00",
			"HumanReadable" : "True",
			"Scope" : "System",
			"SeriesType" : "Detail",
			"StartDate" : "2021-03-13T00:00:00+01:00"
		},
		"Status" : 
		{
			"Code" : 0,
			"ErrorDetail" : 
			{
				"Nodes" : []
			},
			"Reason" : "",
			"UserMessage" : ""
		},
		"Timestamp" : "2021-03-14T10:04:31+01:00"
	}
}

pacco81, 2021-03-29:
Leider kommen nach Update auf 1.1.3 bei mir immer noch keine MPPT-Daten.

Beim Stöbern im Quellcode scheint es eine Funktion "GetArchiveData" zu geben. Im Log sehe ich keine Fehlermeldungen.
Was kann ich noch tun ? Da ich ein Norddach an einem eigenen String habe, würde ich gerne die Leistung dieses Strings beobachten im Vergleich zum Süddach. Leider geht das aktuell nur im FroniusSolarWeb.

nkleber78, 2021-03-30:
Was hast du für ein Intervall für Archiv eingestellt?

pacco81, 2021-03-30:
Hi !

Ich habe das 120s default gelassen.

P.S. Ich habe jetzt testweise die Objekte manuell angelegt. Jetzt werden dir Objekte vom Adapter mit Daten befüllt ! Also scheint es bei mir ein Problem bei der automatischen Anlage der Archiv-Objekte zu geben. Wenn ich logs o.Ä. zur Analyse beisteuern kann, bitte Bescheid sagen.

Auf jeden Fall Vielen Dank für die Adapterentwicklung! Bald kann ich meinen kostenpflichtigrn SolarEeb Account einsparen.

nkleber78, 2021-03-30:
Das Problem habe ich bereits erkannt und gelöst in meinem Repository. Ist dann beim offiziellen Release gelöst

pacco81, 2021-04-02:
Super. Dann werden die Objekte in Zukunft automatisch angelegt. Besten Dank !

mcm1957, 2023-05-31:
Ist da noch was offen?
Ansonsten bitte label fixed setzen und comment "fxied with release 2.0.0 (oder was auch immer)

nkleber78, 2023-05-31:
Implementiert in der Version 2.0.0

closed 2023-05-31 by nkleber78