iobroker-community-adapters/ioBroker.fronius #37: Power consumed from grid today/ power send to grid

m4ding, 2020-05-10:
Hi,

I‘m using v1.05 of the adapter. I could not find a value for the following two:
Power consumed from grid today (until now) 
Power send to grid today (until now)

Example shown on fronius website (left to grid, right from grid)

[image]

Thanks!

m4ding, 2020-05-19:
I have asked the Fronius Support. According to them, the value of power to/ from grid (P_Grid: http://froniusIP/solar_api/v1/GetPowerFlowRealtimeData.fcgi) must be saved each at midnight and upon next midnight like an electricity meter (which counts the electricity taken retrieved from the grid over the entire day) in order to subtract the two values.

m4ding, 2020-12-04:
@schweigel wäre es möglich dass du dir das mal ansiehst? Das wäre eine super Erweiterung. Man könnte über iobroker dann sehen, wie viel Strom am Tag vom Netz bezogen wird.

schweigel, 2020-12-04:
Hallo,

also in den PowerFlowRealtimeData konnte ich da nichts nützliches finden. Da wären zwar die aktuellen Werte wie P_Grid, aber die sind immer nur eine Momentaufnahme. Da diese während der Zeit schwanken können, wäre das nur sehr ungenau, wenn man das mit der Zeit zwischen den Messwerten multiplizieren würde.

Aber ich habe in den ArchivData die Werte EnergyReal_WAC_Minus_Absolute und EnergyReal_WAC_Plus_Absolute gefunden.
Diese geben bereits einen absoluten Verbrauch an. Wenn man nun jeweils den Tagesstartwert abzieht bekommt man den Tages Verbrauch.

An dieser Stelle müssen wir dann aber die genauen Grenzen für diesen Adapter abstecken.
Wir haben zwei Optionen:

1. Nur Werte auslesen
* die Daten EnergyReal_WAC_Minus_Absolute & EnergyReal_WAC_Plus_Absolute  werden wie die anderen Archivdaten zyklisch ausgelesen
* eine Verarbeitung der Zählerwerte zu Tages-, Wochen- und Monatswerten muss dann auf anderem Wege in ioBroker erfolgen

2. Werte auslesen und Statistiken berechnen
* die Daten EnergyReal_WAC_Minus_Absolute & EnergyReal_WAC_Plus_Absolute  werden wie die anderen Archivdaten zyklisch ausgelesen
* die ausgelesenen Daten werden statistisch verarbeitet und fallen als All-in-one-Lösung aus diesem Adapter raus

@ldittmar81 was meinst du dazu?

schweigel, 2020-12-06:
Ich habe gesehen das die Werte *EnergyReal_WAC_Minus_Absolute* & *EnergyReal_WAC_Plus_Absolute* bereits als normale Messwerte enthalten sind. Daher ist die umständliche Abfrage über die Archivdaten nicht nötig.

Die Statistik der z.B. Tagesdaten kann dann via JavaScript erfolgen oder zum Beispiel über den Adapter [statistics](https://github.com/foxthefox/ioBroker.statistics).

m4ding, 2021-03-13:
Hi,

ich habe es nun mit Blockly gelöst. Voraussetzung ist, dass das SmartMeter den kompletten Hausverbrsuch erfasst (an Position 0?). 

Zuerst die Datenpunkte als Zahl anlegen:

[image]

`<xml xmlns="https://developers.google.com/blockly/xml">
  <variables>
    <variable id="-8PF#`5kx1eM55XS+2mq">NetzbezugAktuell</variable>
    <variable id="-`O;EzYvACeVPS+3KjnH">NetzbezugUmMitternacht</variable>
    <variable id="fy2pJDC?f|0HOfEOajCT">Netzbezug</variable>
    <variable id="M{G_i}a#(9?p=/5|b{?E">PVProduktion</variable>
    <variable id="bni6KKZ7T,g}raa~5ghd">EinspeisBezug</variable>
    <variable id="]Pj#~jR[#Bap^@K8j:*7">Hausverbrauch</variable>
    <variable id="J{1{_L@ZJTMUe#mdd8M3">AutarkieGerundet</variable>
  </variables>
  <block type="comment" id="jtKnUrHo!zjyyO{cA?0K" x="-2587" y="-3812">
    <field name="COMMENT">nachts Bezug setzen um Mitternacht</field>
    <next>
      <block type="schedule" id="=9h%5`.|%}F:(Qq(|LAr">
        <field name="SCHEDULE">{"time":{"exactTime":true,"start":"23:59"},"period":{"days":1}}</field>
        <statement name="STATEMENT">
          <block type="timeouts_wait" id=",}P;:%!iqmBDZyuwXwx{">
            <field name="DELAY">59</field>
            <field name="UNIT">sec</field>
            <next>
              <block type="variables_set" id="kKZf$t4vj*ta_+hnF-.H">
                <field name="VAR" id="-`O;EzYvACeVPS+3KjnH">NetzbezugUmMitternacht</field>
                <value name="VALUE">
                  <block type="get_value" id="^m]m#}2V|mO=1,Dd;O?E">
                    <field name="ATTR">val</field>
                    <field name="OID">fronius.0.meter.0.EnergyReal_WAC_Plus_Absolute</field>
                  </block>
                </value>
                <next>
                  <block type="update" id="BU#?h5G19IBr]Ufh#6Ps">
                    <mutation xmlns="http://www.w3.org/1999/xhtml" delay_input="false"></mutation>
                    <field name="OID">javascript.0.scriptEnabled.EigeneDatenpunkte.Fronius.NetzbezugMitternacht</field>
                    <field name="WITH_DELAY">FALSE</field>
                    <value name="VALUE">
                      <block type="variables_get" id="c{v?vqDv=OJX,L+)RHDK">
                        <field name="VAR" id="-`O;EzYvACeVPS+3KjnH">NetzbezugUmMitternacht</field>
                      </block>
                    </value>
                    <next>
                      <block type="telegram" id="_`tL^`b+_*Er2|0-`?vs">
                        <field name="INSTANCE"></field>
                        <field name="LOG"></field>
                        <field name="SILENT">FALSE</field>
                        <field name="PARSEMODE">default</field>
                        <value name="MESSAGE">
                          <shadow type="text" id="{!@;7U`/K%kM^PaP[H:!">
                            <field name="TEXT">text</field>
                          </shadow>
                          <block type="text_join" id="O;kS=DDo,/ySbLP!A^@X">
                            <mutation items="2"></mutation>
                            <value name="ADD0">
                              <block type="text" id="mjQCtBluWIqxo|UIIZ,Y">
                                <field name="TEXT">Netzbezug um Mitternacht: </field>
                              </block>
                            </value>
                            <value name="ADD1">
                              <block type="variables_get" id="[Cv0~U6)W{:0.ITud=98">
                                <field name="VAR" id="-`O;EzYvACeVPS+3KjnH">NetzbezugUmMitternacht</field>
                              </block>
                            </value>
                          </block>
                        </value>
[... 564 lines cut ...]
                                        </next>
                                      </block>
                                    </next>
                                  </block>
                                </next>
                              </block>
                            </next>
                          </block>
                        </next>
                      </block>
                    </next>
                  </block>
                </next>
              </block>
            </next>
          </block>
        </statement>
      </block>
    </next>
  </block>
</xml> `

NetzbezugJeTag liefert den Wert in Wh und kann dann über vis/ Grafana etc. angezeigt werden. 

Bei mir läuft es gut.

m4ding, 2021-03-13:
sorry die Fode Formatierung ist nicht wirklich gut.

mcm1957, 2023-05-31:
Der Adapter soll(te) jene Werte auslesen die das Device bietet,
Für statistische Abfragen, Archivierungen etc. gibts eigen Adapter.

Ich schließe daher hier mal.

closed 2023-05-31 by mcm1957