iobroker-community-adapters/ioBroker.fronius #38: (754) TypeError: Cannot read property 'Value' of undefined

juamiso, 2020-05-21:
Adapter v1.0.5

I am trying to get rid of all warnings in the log, any clue how to get this of this one? it happens continuosly...

Reading from a ethernet connected fronius symo 7

```
fronius.0 | 2020-05-21 18:22:53.780 | warn | (754) TypeError: Cannot read property 'Value' of undefined
-- | -- | -- | --
fronius.0 | 2020-05-21 16:51:24.113 | warn | (754) TypeError: Cannot read property 'Value' of undefined
fronius.0 | 2020-05-21 16:32:54.250 | warn | (754) TypeError: Cannot read property 'Value' of undefined
fronius.0 | 2020-05-21 14:29:24.006 | warn | (754) TypeError: Cannot read property 'Value' of undefined
fronius.0 | 2020-05-21 12:40:23.641 | warn | (754) TypeError: Cannot read property 'Value' of undefined
```

Aare07, 2020-08-10:
Hi
Also on my Adapter V 1.05 this pops up in the logs:

`
fronius.0 | 2020-08-10 14:10:56.468 | warn | (1445) TypeError: Cannot read property 'Value' of undefined
fronius.0 | 2020-08-10 14:07:27.146 | warn | (1445) TypeError: Cannot read property 'Value' of undefined
fronius.0 | 2020-08-10 13:50:56.578 | warn | (1445) TypeError: Cannot read property 'Value' of undefined
fronius.0 | 2020-08-10 13:48:26.412 | warn | (1445) TypeError: Cannot read property 'Value' of undefined
`

Thanks for the work on this adapter!

fladdie, 2020-10-02:
same here with Adapter V 1.05

fronius.0 | 2020-10-02 09:03:02.017 | warn | (7031) TypeError: Cannot read property 'Value' of undefined
-- | -- | -- | --

nkleber78, 2020-11-19:
Hi all, this is because some of the parameters previously defined in the solarAPI are no longer available or has a changed name. The newest version which is currently build as 1.0.6 should get rid of this except the case that you have a Battery in combination with the GEN24 because the Battery currently is not supported

nkleber78, 2020-11-26:
Fixed with V1.1. Please try out and if fixed pleas close the issue

mcm1957, 2024-07-24:
no (negative) feedback so lets close here.

If the problem still occures please open a new issue.

closed 2024-07-24 by mcm1957