iobroker-community-adapters/ioBroker.fronius #39: Adapter nicht verbunden mit Fronius

BenjaminH1988, 2020-07-12:
Ich habe das Problem, dass der Adapter nach dem Umzug auf einen Intel NUC mit Proxmox sich nicht verbinden lässt.

Ich habe die IP-Adresse neu eingegeben aber es kommt immer
Verbunden mit Host: Wahr
Lebenszeichen: Wahr
Verbunden mit Fronius: falsch

Tommmek, 2020-08-18:
Hallo ich habe genau das gleiche Problem

Triple-M77, 2022-01-21:
Ich habe heute den Fronius Adapter zum ersten Mal installiert (V1.1.6) und bekomme auch keine Verbindung zu meinem Gen24. Wie oben schon geschrieben sind die ersten beiden Punkte grün aber der letzte rot und der Adapter steht somit auf gelb. Was kann man machen?

HemanMaster, 2022-12-02:
Hallo ich hab das selbe Problem, hat jemand von euch eine Lösung? Ich gebe doch im Adapter nur die IP:Adresse ein oder? Brauch ich noch einen Port? Danke

Mesok77, 2024-01-11:
Bei mir lag das Problem darin, dass die Solar_Api im Wechselrichter deaktiviert war.
Diese kann man im WR unter Kommunikation-->Solar API aktivieren.

mcm1957, 2024-07-24:
Sollte das Problem mit der aktuellen Release noch immer auftreten, bitte ein neues Issue öffnen.

closed 2024-07-24 by mcm1957