iobroker-community-adapters/ioBroker.fronius #49: Adapter seems to have issues with SYMO GEN24 device

nkleber78, 2020-09-18:
I connected a SYMO GEN24 device and setup the adapter. Unfortunately i do not get values from the device, at least for the most of the values, only for very few powerflow values like P_Grid, P_Load where i get correct values. I just get following warnings
(641) TypeError: Cannot read property 'Value' of undefined
(641) TypeError: Cannot read property 'HWVersion' of undefined

---

**System data:**

Platform: linux
Distribution: Debian GNU/Linux
Architecture: x64
Model: Intel(R) Atom(TM) CPU D2701   @ 2.13GHz
RAM: 3.86 GB
Node.js: v12.18.3
NPM: 6.14.6
JS-Controller (1): 3.1.4
Admin: 4.0.10
Fronius inverters adapter: 1.0.5

> *created by ioBroker.info*

nkleber78, 2020-09-25:
After some investigation with Fronius support if found out, that not everything returned by the API is according to specification. Furthermore there are many changes made on the API even the Version is still V1. So it is hard to detect the changes. The idea which is partly implemented is to generate the Objects only if existing and generate all not predefined objects in a general way.

JBSullivan, 2020-11-19:
Hallo nkleber78 - Sorry wenn ich dich hier von der Seite anquatsche ;) Es geht um den Adapter für den Fronius GEN24 Wechselrichter - offensichtlich hast du da schon einiges erweitert, bzw. gerade gebogen. Wir haben im Forum einen relativ regen Thread, wo es genau um diesen Wechselrichter geht. Der User @mike97714 hat ein JavaSkript geschrieben, welches noch mehr Daten aus dem WR heraus holt ( https://forum.iobroker.net/topic/33809/fronius-adapter-entwicklung-eingestellt/203 ) Wäre cool, wenn das ggf. in den Fronius Adapter integrierbar wäre.  Leider haben die meisten von uns keine Ahnung vom Programmieren, aber du scheinst dich da auszukennen - hast du ggf. Lust dort weiter zu machen? Der Adapter Entwickler Idittmar würde dir bei Bedarf sogar das ganze Fronius Projekt übergeben, da er selber keinen Fronius WR mehr besitzt.

nkleber78, 2020-11-26:
Issues fixed with V1.1

closed 2020-11-26 by nkleber78