iobroker-community-adapters/ioBroker.fronius #52: Enhancement: use more available Solar API calls

stan23, 2020-10-17:
The Solar API provides more APIs than this adapter is using.

These API calls are answered with data on my **Fronius Symo 10.0-3-M**:

| used in this adapter | API example call |
| -------------------- | ---------------- |
| yes | http://fronius-ip/solar_api/GetAPIVersion.cgi |
| no  | http://fronius-ip/solar_api/v1/GetInverterRealtimeData.cgi?Scope=System |
| no  | http://fronius-ip/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DeviceId=1&DataCollection=CumulationInverterData |
| yes | http://fronius-ip/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DeviceId=1&DataCollection=CommonInverterData |
| no  | http://fronius-ip/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DeviceId=1&DataCollection=MinMaxInverterData |
| no  | http://fronius-ip/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DeviceId=1&DataCollection=3PInverterData |
| yes | http://fronius-ip/solar_api/v1/GetLoggerInfo.cgi |
| no  | http://fronius-ip/solar_api/v1/GetLoggerLEDInfo.cgi |
| no  | http://fronius-ip/solar_api/v1/GetInverterInfo.cgi |
| no  | http://fronius-ip/solar_api/v1/GetActiveDeviceInfo.cgi?DeviceClass=System |
| yes | http://fronius-ip/solar_api/v1/GetPowerFlowRealtimeData.fcgi?Scope=System |

These calls are not supported from my device and return no data, because I have neither a storage nor a meter:
| used in this adapter | API example call |
| -------------------- | ---------------- |
| yes | http://fronius-ip/solar_api/v1/GetSensorRealtimeData.cgi?Scope=System&DataCollection=NowSensorData |
| no  | http://fronius-ip/solar_api/v1/GetStringRealtimeData.cgi?Scope=System&DataCollection=NowStringControlData |
| yes | http://fronius-ip/solar_api/v1/GetStorageRealtimeData.cgi?Scope=System |
| yes | http://fronius-ip/solar_api/v1/GetMeterRealtimeData.cgi?Scope=System |

The internal temperature can only be read by using this trick:
http://fronius-ip/solar_api/v1/GetArchiveData.cgi?Scope=System&StartDate=17.10.2020&EndDate=17.10.2020&Channel=Temperature_Powerstage

Maybe the adapter can be enhanced with the API calls that return useful additional data.

nkleber78, 2020-11-26:
Please try out the newest Version V1.1. There the most datapoints are visible. If you need specific extensions, then just let us know the missing datapoints

nkleber78, 2023-06-04:
@stan23 Following requests are added. In general the calls are added to the README.md
* http://fronius-ip/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DeviceId=1&DataCollection=MinMaxInverterData
* http://fronius-ip/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DeviceId=1&DataCollection=3PInverterData
* http://fronius-ip/solar_api/v1/GetInverterInfo.cgi
* http://fronius-ip/solar_api/v1/GetActiveDeviceInfo.cgi?DeviceClass=System
* http://fronius-ip/solar_api/v1/GetStringRealtimeData.cgi?Scope=System&DataCollection=NowStringControlData