iobroker-community-adapters/ioBroker.fronius #64: Adapter kann ab Version 1.1.0 nicht gestartet werden

BlueScorpion001, 2020-12-07:
Seit dem Upgrade von 1.0.5 auf 1.1.0 und 1.1.2 kann ich in der Adapterkonfiguration zwar noch die IP meines Gen24 eintragen aber beim Klick auf die blaue Taste "ÜBERPRÜFEN SIE IHRE IP-ADRESSE" passiert nichts. Auch bleibt die Instanz rot.
In der Version 1.0.5 wird mein Gen24 aber gefunden und die Instanz geht auf grün.

ioBroker auf Raspi 3 b+
ioBroker-Pi
Typ: js-controller
Titel: JS controller
OS: linux
Verfügbar: 3.1.6
Installiert: 1.4.2

nkleber78, 2020-12-07:
@BlueScorpion001 also bei mir funktioniert das problemlos. Habe gerade mit einer neuen Instanz getestet. Hast du ggf. https eingestellt anstelle http für den request type? 
[image]

BlueScorpion001, 2020-12-07:
Hallo Norbert,
danke für deine schnelle Antwort!

Ich habe jetzt doch ein Upgrade des JS Controller auf 3.6.1 gemacht und
jetzt ist die Instanz auf grün gegangen. Mit dem alten Controller ging
nicht.

Auf jeden Fall super dass ihr den Fronius Adapter jetzt weiter entwickelt!
Herzlichen Dank dafür!

Kommt auch noch die Einbindung des BYD Akkus? Da hab ich aktuell noch keine
Werte angezeigt.

Liebe Grüße
Christian

nkleber78, 2020-12-07:
Alles klar. Bzgl. BYD. Welche daten meinst du? die API liefert da keine Daten soweit mir bekannt ist...

BlueScorpion001, 2020-12-07:
Aha,
BYD Statusdaten erwarten wir von dem Fronius Firmware Update.

Danke für deine Info!!! Hast du den Adapter jetzt weiter entwickelt?
Hoffe nach dem UpDate von Fronius können wir noch mehr im ioBroker
auswerten.

glG und nochmals herzlichen Dank!!!
Christian

Freundliche Grüße
Christian Kremser

mcm1957, 2023-05-31:
Falls diese Problem noch aktuell ist bitte einen neuen isseu unter Angabe der derzeit verwendeten SW Versionen erstellen.
Danke

closed 2023-05-31 by mcm1957