iobroker-community-adapters/ioBroker.fronius #76: Warning in LOG: (23514) TypeError: Cannot read property 'Value' of undefined

BigStephan1978, 2021-01-08:
In version 1.0.5 and also in version 1.1.2
Sometimes after 3 minutes, sometimes after 20 minutes.
as i can see the values which commin from my fronius Symo 3.7-3-M

nkleber78, 2021-01-09:
@BigStephan1978 can you might give as more information about this issue. When does this happen? Usually during startup/shutdown of inverter or also during normal operation where enough power to feed in is available. Does this just happen a few time a day or all the time?

BigStephan1978, 2021-01-10:
@nkleber78 It's happen durring "normal" duty of the adapter. The inverter is connected already (not in start or so).

Here is a log (I've switched to debug-level log at 8 o'clock pm!):

2021-01-10 18:27:26.807 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 18:29:56.550 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 18:32:26.521 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 18:34:56.347 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 18:37:26.881 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 18:39:56.444 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 18:42:26.474 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 18:44:56.475 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 18:47:26.642 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 18:49:56.784 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 18:52:26.537 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 18:54:56.710 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 18:57:27.036 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 18:59:57.027 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:02:26.819 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:04:56.628 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:07:26.703 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:09:56.869 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:12:26.592 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:14:56.864 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:17:26.595 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:19:56.686 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:22:26.545 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:24:57.466 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:27:26.998 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:29:56.899 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:32:26.392 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:34:56.525 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:37:26.747 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:39:57.018 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:42:26.885 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:44:56.565 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:47:26.797 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:49:56.959 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:52:26.807 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:54:56.577 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:57:26.881 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 19:59:56.844 - warn: fronius.0 (18032) TypeError: Cannot read property 'Values' of undefined
2021-01-10 20:00:22.773 - info: host.iobrokervm stopInstance system.adapter.fronius.0 (force=false, process=true)
2021-01-10 20:00:22.787 - info: fronius.0 (18032) Got terminate signal TERMINATE_YOURSELF
2021-01-10 20:00:22.782 - info: host.iobrokervm stopInstance system.adapter.fronius.0 send kill signal
2021-01-10 20:00:23.290 - info: fronius.0 (18032) terminating
2021-01-10 20:00:23.292 - info: fronius.0 (18032) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
2021-01-10 20:00:23.784 - info: host.iobrokervm stopInstance system.adapter.fronius.0 killing pid 18032
2021-01-10 20:00:23.833 - info: host.iobrokervm instance system.adapter.fronius.0 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
2021-01-10 20:00:25.825 - info: host.iobrokervm instance system.adapter.fronius.0 started with pid 31165
2021-01-10 20:00:30.613 - debug: fronius.0 (31165) Redis Objects: Use Redis connection: 127.0.0.1:9001
2021-01-10 20:00:30.661 - debug: fronius.0 (31165) Objects client ready ... initialize now
2021-01-10 20:00:30.668 - debug: fronius.0 (31165) Objects create System PubSub Client
2021-01-10 20:00:30.670 - debug: fronius.0 (31165) Objects create User PubSub Client
2021-01-10 20:00:30.671 - debug: fronius.0 (31165) Objects client initialize lua scripts
[... cut ...]
2021-01-10 20:01:31.827 - debug: fronius.0 (31165) System command: /bin/ping -n -w2 -c1 192.168.1.163
2021-01-10 20:01:41.829 - debug: fronius.0 (31165) System command: /bin/ping -n -w2 -c1 192.168.1.163
2021-01-10 20:01:51.830 - debug: fronius.0 (31165) System command: /bin/ping -n -w2 -c1 192.168.1.163
2021-01-10 20:02:01.830 - debug: fronius.0 (31165) System command: /bin/ping -n -w2 -c1 192.168.1.163
2021-01-10 20:02:11.831 - debug: fronius.0 (31165) System command: /bin/ping -n -w2 -c1 192.168.1.163

nkleber78, 2021-01-10:
@BigStephan1978 What i meant with startup was start/stop of feeding from PV to the grid. According to the log it is as i thought just during the time where the Inverter does not feed in. Could you might deliver a log file of a whole day in debug mode for fronius.0 to analyse which parameters could cause this errors?

BigStephan1978, 2021-01-11:
@nkleber78 as mentiones in my last post it's not only when its driven by PV (feeded) but not. So yes...independed on sunshine or not.

Hansi1989, 2021-01-13:
I have had the exact same problem as of today. But I also have to say that I made a firmware update to 3.16.7-1 on the inverter today.

weindler, 2021-01-13:
auch das gleiche

  | (992) TypeError: Cannot read property 'Value' of undefined
-- | --
hatte ich aber auch schon in den letzten versionen, kommt aber sporadisch und ist schlecht einzugrenzen woran das liegen kann.

nkleber78, 2021-01-13:
If someone of you could install the latest version from Github for testing, then we would at least able to check which API request causes this error because i added this information to the log output. Then hopefully by inspecting the code in detail for this API call should help to identify the root cause. As @weindler mentioned it is not easy to catch as this is not a permanent error. I am also running the latest FW on my Symo 20 without such messages, so it seems not FW related...
Thanks

weindler, 2021-01-15:
ich installier mal die git hub version dann schaun wir mal.

weindler, 2021-01-16:
Folgende Warnmeldung habe ich heute erhalten:

fronius.0 | 2021-01-16 07:34:12.004 | warn | (873) GetArchiveData: TypeError: Cannot read property 'Values' of undefined
-- | -- | -- | --
fronius.0 | 2021-01-16 07:31:41.626 | warn | (873) GetArchiveData: TypeError: Cannot read property 'Values' of undefined
fronius.0 | 2021-01-16 07:29:11.493 | warn | (873) GetArchiveData: TypeError: Cannot read property 'Data' of undefined

Ansonsten sehe ich keine weiteren Probleme

weindler, 2021-01-17:
Die Warnmeldungen von gestern auf heute:

fronius.0 | 2021-01-17 07:29:11.443 | warn | (873) GetArchiveData: TypeError: Cannot read property 'Values' of undefined
-- | -- | -- | --
fronius.0 | 2021-01-17 07:26:41.736 | warn | (873) GetArchiveData: TypeError: Cannot read property 'Values' of undefined
fronius.0 | 2021-01-17 07:24:11.833 | warn | (873) GetArchiveData: TypeError: Cannot read property 'Data' of undefined
fronius.0 | 2021-01-17 03:11:40.696 | warn | (873) GetArchiveData: TypeError: Cannot read property 'Data' of undefined

weindler, 2021-01-22:
Moin, mir ist gerade aufgefallen daß die folgende warnmeldung kommt wenn der wechselrichter von standby zu running wechselt

fronius.0 | 2021-01-22 07:36:42.939 | warn | (873) GetArchiveData: TypeError: Cannot read property 'Values' of undefined
-- | -- | -- | --

hoffe das hilft

nkleber78, 2021-01-22:
@weindler Ich denke das problem sollte mit dem fix gelöst sein. Mal schauen wie schnell das in den aktuellen Branch gemerged wird damit du das testen kannst

weindler, 2021-01-23:
@nkleber78 habe mir den Adapter über git aktualisiert, schauen wir mal, update lief ohne Fehler durch Adapter läuft. Halte dich auf dem laufenden, danke dafür.

weindler, 2021-01-24:
@nkleber78 sollte ich die neuste version draufhaben, dann sind die warnungen immer noch da und zwar zum start des wechselrichters wenn er von standby auf running wechselt.

fronius.0 | 2021-01-24 07:42:59.526 | warn | (905) GetArchiveData: TypeError: Cannot read property 'Values' of undefined
-- | -- | -- | --

ansonsten habe ich gestern während der Arbeit des Wechselrichters nichts außergewöhnliches feststellen können, auch keine Warnung oder Error

nkleber78, 2021-01-24:
@weindler hast du von meinem Repository (https://github.com/nkleber78/ioBroker.fronius) installiert oder vom globalen. Aktuell ist derzeit nur meines, das andere muss noch gemerged werden damit meine Änderungen dort auch integriert sind.

weindler, 2021-01-25:
@nkleber78 ok habs jetzt von deinem repository geladen, dann laß uns mal schauen, halte dich auf dem laufenden.

weindler, 2021-01-27:
@nkleber78 folgende warnmeldung wurde ausgespuckt, es sind jetzt definitiv weniger warnmeldungen.

fronius.0 | 2021-01-27 07:33:16.096 | warn | (735) GetArchiveData: TypeError: Cannot read property 'hasOwnProperty' of undefined
-- | -- | -- | --

nkleber78, 2021-01-27:
@weindler hab nochmals eine prüfung eingefügt da in diesem Fall offensichtlch das ganze objekt nicht existiert. Bitte nochmals von meinem Repository updaten. Danke für die Unterstützung

weindler, 2021-01-27:
ok, upgedatet, ich halte dich auf dem laufenden.

weindler, 2021-01-28:
Selbe wie gestern @nkleber78 

fronius.0 | 2021-01-28 07:34:12.495 | warn | (21471) GetArchiveData: TypeError: Cannot read property 'hasOwnProperty' of undefined
-- | -- | -- | --
fronius.0 | 2021-01-28 07:31:41.820 | warn | (21471) GetArchiveData: TypeError: Cannot read property 'hasOwnProperty' of undefined

nkleber78, 2021-01-28:
@weindler Bitte versuch es nochmals von meinem Repository. Habe die Prüfung nochmals erweitert auch wenn ich der Meinung war dass es passen sollte

weindler, 2021-01-31:
@nkleber78 moin, bis jetzt keine Fehlermeldungen im Log. Schaut gut aus wenn es so gewollt ist.

nkleber78, 2023-06-04:
Issue scheint gefixt zu sein

closed 2023-06-04 by nkleber78