iobroker-community-adapters/ioBroker.fronius #78: Keine Werte unter Powerflow E_Day E_Total E_Year bei Symo 8.2-3M

jonnyb2, 2021-01-11:
Bei meinem Symo 8.2-3M zeigt es mit keine Tageswerte an, alle Werte 0
Genutzt wird Symo mit smartmeter
Ist-Werte (aktuelle Verbrauch / Bezug) funktionieren.

nkleber78, 2021-01-11:
@jonnyb2 Was liefert die API? Bitte poste mal den output von http://ip_of_inverter/solar_api/v1/GetPowerFlowRealtimeData.fcgi 
Bei meinem System 4 * Symo 20kw funktioniert das...

weindler, 2021-01-13:
Hi jonny, das kann sein, wenn der fronius in den standby geht dann sind hier die werte auf 0 mußt du mal beobachten, wenn wieder sonne scheint dann steht da auch was drin, zumindestens bei mir, habe den selben WR. Schau unter inverter/1/Day Ernergy hier findest du die erbrachte Tagesleistung und weiter unten dann die Jahresleistung für das aktuelle Jahr und die Gesamtleistung.

m4ding, 2021-02-01:
Man kann den „Nachtmodus“ ausschalten

aDabbelju, 2021-05-22:
> Man kann den „Nachtmodus“ ausschalten

Wo findet man das denn genau? Ich habe einen Symo Gen24 und kann in der Technikerebene leider nichts finden

weindler, 2021-05-23:
das kannst du direkt am wechselrichter im menü auswählen oder eben nicht, bei mir geht er auf standby um auch einfach strom zu sparen, habe hiermit keine probleme, kommt sonne dann geht er auf running.