iobroker-community-adapters/ioBroker.fronius #86: Gehäuse Temperatur Anzeige hinzufügen

JBSullivan, 2021-02-03:
Ist kein Fehler, aber ich weiß nicht wie man hier Verbesserungsvorschläge einreichen kann.

Wie schon im Forum geschrieben, steht auch die Gehäuse Temperatur des WR via http Request zur Verfügung. Vielleicht kann
man den Datenpunkt noch mit aufnehmen. Ich denke der ist für viele von Interesse.

http://[inverter_ip]/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DataCollection=3PInverterData&DeviceId=0

`{
   "Body" : {
      "Data" : {
         "IAC_L1" : {
            "Unit" : "A",
            "Value" : 1.0751680135726929
         },
         "IAC_L2" : {
            "Unit" : "A",
            "Value" : 1.0783138275146484
         },
         "IAC_L3" : {
            "Unit" : "A",
            "Value" : 1.0795806646347046
         },
         "T_AMBIENT" : {
            "Unit" : "C",
            "Value" : 39.195526123046875 <-------------------------------Temperatur
         },
         "UAC_L1" : {
            "Unit" : "V",
            "Value" : 232.28556823730469
         },
         "UAC_L2" : {
            "Unit" : "V",
            "Value" : 234.69015502929688
         }
      }
   },
   "Head" : {
      "RequestArguments" : {
         "DataCollection" : "3PInverterData",
         "DeviceClass" : "Inverter",
         "DeviceId" : "0",
         "Scope" : "Device"
      },
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2021-02-03T13:50:00+00:00"
   }
}`

JBSullivan, 2021-03-16:
Habe heute die neueste Version aus dem latest installiert - Schade das die Temperatur nicht mit hinzugefügt wurde.

nkleber78, 2021-03-16:
Hallo @JBSullivan
dieser part der API ist nicht bei allen typen vorhanden (nur bei 3phasigen typen). Und derzeit ist dieser teil der API nicht implementiert. Wenn es hier mehr Interesse and diesem teil der API gibt, dann kann ich mir so eine Implementierung schon vorstellen

JBSullivan, 2021-03-16:
Vielen Dank für deine Antwort - ich hatte angenommen, das die API identisch mit den ModBus Adressen ist und somit "nur" auf eine Enduser "freundliche" Weise die Werte zur Verfügung stellt.

Die Gehäuse Temperatur ist, zumindest für den GEN24, über ModBus Register auslesbar. Das ist für mich der Grund, warum ich den ModBus Adapter überhaupt betreibe - wegen eben diesem einen Temperatur Datenpunkt. Alles andere läuft über den Fronius Adapter.

Ich persönlich hätte mich halt gefreut wenn auch die Temperatur via API übergeben worden wäre. Dann hätte ich den ModBus Adapter still legen können.

[image]

Aber OK, warten wir mal ab, was sie Fronius Software Abteilung noch so an Firmware Updates bereit hält. Vielleicht ist die Temperatur Geschichte ja für zukünftige Versionen dann wieder ein Thema für die breite Mehrheit.

In diesem Sinne - Vielen Dank für deine bis hierher geleistete Arbeit an den Adapter.

nkleber78, 2021-03-16:
@JBSullivan was ich eigentlich meinte ist der teil der API ist im Adapter nicht eingebaut. Bei den 3Phasigen Wechselrichtern müsste es aber schon überall drin sein. D.h. es wäre über den Adapter natürlich möglich das auszulesen...