iobroker-community-adapters/ioBroker.fronius #87: mit 1.1.0 PAC keine 0 Watt bei 0 Erzeugung

Negalein, 2021-02-13:
Mit der Version 1.1.0 komme ich auch bei Dunkelheit nicht mehr auf 0 Watt Erzeugung.

Der Inverter-Datenpunkt fronius.0.inverter.1.PAC zeigt bei mir dann immer noch zB 33 Watt an.

Wir haben einen Symo 10.0-3-M

Erst mit 1.0.5 wird fronius.0.inverter.1.PAC bei keiner Erzeugung (zB bei Nacht) wieder 0 Watt angezeigt.

nkleber78, 2021-02-13:
Ich nehme an dass dein Inverter in den sleep mode geht. Wenn das geschieht wenn der vorherige Datenpunkt noch eine Leistung hatte, dann kann es sein dass der letzte Wert erhalten bleibt. Was hast du für eine Abfrageinterval eingestellt?

Negalein, 2021-02-13:
Nein, geht nicht in den Sleep Mode.

Abfrageintervall sind 30 Sek.

[image]

nkleber78, 2021-02-13:
Aber hier ist der Datenpunkt ja 0... Bitte schau dir mal an in welchem Zustand das genau passiert. Und wenn das auftritt. Dann schau mal was bei dir die API hier liefert "http://ip_des_inverters/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DeviceId=1&DataCollection=CommonInverterData"

Negalein, 2021-02-13:
Ja, weil es mit 1.0.5 (läuft aktuell bei mir) funktioniert, aber mit 1.1.0 nicht.

Bin nicht allein mit dem Phänomen.
https://forum.iobroker.net/topic/33809/fronius-adapter-entwicklung-eingestellt/238?_=1613081208863

nkleber78, 2021-02-13:
Alles klar, aber das Forum kann nur helfen wenn auch details über das Problem bekannt sind. Das kann ein Verhalten eines spezifischen WR sein usw. Somit bitte um details damit das Problem eingegrenzt werden kann. Bei mir taucht das Problem nicht auf beim Symo Gen24...

Negalein, 2021-02-13:
haben den Symo 10.0-3-M

welche Daten kann ich dir noch schicken?

nkleber78, 2021-02-13:
Bitte schick mir die Daten von der API wie im vorigen Post geschrieben. Vielleicht hilft das weiter... Weiters wäre noch interessant wie der status von Fronius.0.info.connection ist. Ist dieser status true oder false? Nicht zu vergessen bitte mal den Adapter auf debug schalten und schauen was er da so alles ausspuckt. Wir versuchen möglichst sinnvolle debug messages einzubauen damit wir fehler lokalisieren können

Negalein, 2021-02-13:
fronius.0.info.connection = true

Debug schick ich morgen. Da tut sich etwas mehr als in der Nacht! ;)

API:
{
   "Body" : {
      "Data" : {
         "DAY_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 1391
         },
         "DeviceStatus" : {
            "ErrorCode" : 0,
            "LEDColor" : 3,
            "LEDState" : 0,
            "MgmtTimerRemainingTime" : -1,
            "StateToReset" : false,
            "StatusCode" : 2
         },
         "TOTAL_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 50745600
         },
         "YEAR_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 87658
         }
      }
   },
   "Head" : {
      "RequestArguments" : {
         "DataCollection" : "CommonInverterData",
         "DeviceClass" : "Inverter",
         "DeviceId" : "1",
         "Scope" : "Device"
      },
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2021-02-13T22:15:51+01:00"
   }
}

nkleber78, 2021-02-14:
Hallo @Negalein 
was man hier sieht ist dass der Inverter den Datenpunkt nicht mehr liefert. Somit wird der Wert nicht mehr verändert. Da sich die Struktur verändert hat, damit es auch mit den unterschiedlichen Inverter Typen funktioniert, wird die Struktur dynamisch gefüllt und nur die Daten die auch auf der API verfügbar sind werden in der Struktur upgedated. Das ist mit einfachen Mitteln nicht sinnvol lösbar. Gibt es bei dir ein Problem wenn der letzte wert stehen bleibt?

Negalein, 2021-02-14:
Ok, dann bleibt mir nur übrig auf der 1.0.5 zu bleiben.

>Gibt es bei dir ein Problem wenn der letzte wert stehen bleibt?

Ja, da der Wert dann bis zur Änderung in Grafana auch so bleibt. Oder in Sourceanalytic, usw.

Negalein, 2021-02-14:
Hallo

Der Debug-Log sagt nicht viel aus.
Nur die normalen Ping
>04:27:56.650  - [34mdebug[39m: fronius.0 (6825) System command: /bin/ping -n -w 2 -c 1 10.0.1.90

und 3x war dieses
>04:28:01.270  - [33mwarn[39m: fronius.0 (6825) TypeError: Cannot read property 'Value' of undefined

Soll ich den Debug mal mit 1.1.1 machen?

nkleber78, 2021-02-14:
Installier mal die Variante von Github aus meinem Repository und gib mir dann Bescheid ob deine Probleme damit gelöst sind...
https://github.com/nkleber78/ioBroker.fronius

Diese Variante dann mal im Debug modus laufen lassen. Dann solltest du auch entsprechende log Einträge sehen...

Negalein, 2021-02-15:
>Installier mal die Variante von Github aus meinem Repository und gib mir dann Bescheid ob deine Probleme damit gelöst sind...

Ja, mit 1.1.2 zeigt es wieder 0 an! 
Danke

closed 2021-02-15 by Negalein