iobroker-community-adapters/ioBroker.fronius #96: Adapterinstanz liefert Fehlermeldungen nach js-controller - Update auf Version 3.2.16

matthiasisrael, 2021-03-01:
Nach dem Update des js-controllers auf die Version 3.2.16 läuft meine erste Instanz tadellos. Die zweite Instanz liefert folgende Fehlermeldungen im Log:

`

host.raspberrypi4 | 2021-03-01 08:36:56.369 | error | instance system.adapter.fronius.1 terminated with code 6 (UNCAUGHT_EXCEPTION)
-- | -- | -- | --
fronius.1 | 2021-03-01 08:36:55.647 | warn | (3658) Terminated (UNCAUGHT_EXCEPTION): Without reason
fronius.1 | 2021-03-01 08:36:55.644 | info | (3658) terminating
fronius.1 | 2021-03-01 08:36:55.126 | error | (3658) The id "inverter." is invalid. Ids are not allowed to end in "."
fronius.1 | 2021-03-01 08:36:55.126 | error | (3658) Error: The id "inverter." is invalid. Ids are not allowed to end in "." at validateId (/opt/iobroker/node_modules/iobroker.js-controller/lib/adapter.js:532:23) at Adapter.setObjectNotEx
fronius.1 | 2021-03-01 08:36:55.122 | error | (3658) unhandled promise rejection: The id "inverter." is invalid. Ids are not allowed to end in "."
fronius.1 | 2021-03-01 08:36:55.122 | error | (3658) Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch().
fronius.1 | 2021-03-01 08:36:55.120 | warn | (3658) TypeError: Cannot read property 'Value' of undefined
fronius.1 | 2021-03-01 08:36:55.102 | warn | (3658) TypeError: Cannot read property 'HWVersion' of undefined

`
Dieser Fehler tritt sowohl bei der Nodeversion v10.18.0 als auch nach bei vorherigem Upgrade auf v12.21.0 auf.
Adapterversion des Fronius ist aktuell (1.0.5)

Ich hoffe ihr könnt helfen :-)

nkleber78, 2021-03-01:
Bitte mal auf die version von GIT updaten. Sollte eigentlich bereits gelöst sein.

matthiasisrael, 2021-03-01:
Hat mit dem Git-Update funktioniert. Danke!!!

nkleber78, 2021-03-12:
@ldittmar81 Kannst du uns bitte den gefallen tun und ein Release vorbereiten? Es scheint dass wir doch einige Probleme gelöst haben. Wenn ich unterstützten kann, dann gib mir bitte Bescheid. Danke
gruß Norbert

ldittmar81, 2021-03-12:
Klar... Mache ich morgen

ldittmar81, 2021-03-15:
Hi Norbert - Version 1.1.3 ist online. Wir können demnächst mal eine Stable Version raus bringen, wenn nichts größeres mehr dazwischen kommt.

mcm1957, 2023-05-31:
falls noch was offen, bitte neues issue anlgen.
Scheint lt. comments mit 1.1.3 gelöst zu seein.

closed 2023-05-31 by nkleber78