iobroker-community-adapters/ioBroker.fronius #97: Undefined fronius.0.inverter.1.Power_DC_String_2  leads to warning

homecineplexx, 2021-03-17:
Hallo
Es kommt folgende Warn-Meldung:
`(10060) State "fronius.0.inverter.1.Power_DC_String_2" has no existing object, this might lead to an error in future versions`

Diese ist aber ganz leicht zu beseitigen. Du hast im Code sicher nur ein Copy-Paste Fehler gemacht. Nämlich in dem unteren Code-Block. 
`if(resp.hasOwnProperty('Voltage_DC_String_2')){
                        var values = inverter.Data.Voltage_DC_String_2.Values;
                        var keys = Object.keys(values);
                        var s2voltage = values[keys[keys.length - 1]];
                        adapter.setState("inverter." + id + ".Voltage_DC_String_2", { val: s2voltage, ack: true });
                        if (typeof s2current !== 'undefined') {
                            adapter.setState("inverter." + id + ".Power_DC_String_2", { val: s2voltage * s2current, ack: true });
                        }
                    }
                    adapter.setState("inverter." + id + ".Power_DC_String_2", { val: s2voltage * s2current, ack: true });`

Du brauchst nur das zweite setzen für Power_DC_String_2 weg schmeissen, da es ausserhalb der Abfrage ist.

Lg chris

pacco81, 2021-07-03:
@nkleber78 
Vielen Dank ! Damit ist der "undefined" Fehler bei mir auch weg (nach dem 2.Adapterstart)

superfloh247, 2021-07-22:
State "fronius.0.inverter.1.Power_DC_String_2" has no existing object, this might lead to an error in future versions

Fronius 1.1.6
Node v12.22.2 
iobroker version 3.3.14

Franzformator, 2023-01-27:
Hallo,

ich bekomme auch eine solche Warnung: State "fronius.0.inverter.1.Power_DC_String_2" has no existing object, this might lead to an error in future versions

Mein WR hat zudem keinen zweiten String.

mcm1957, 2023-05-31:
see https://github.com/iobroker-community-adapters/ioBroker.fronius/issues/206

closed 2023-05-31 by mcm1957