iobroker-community-adapters/ioBroker.fronius #115: Wrong typ nach Umstellung auf js-controller 3.3.4

JBSullivan, 2021-05-01:
lt. Community sollen Meldungen die sich aus der Umstellung des js-controller auf die Version 3.3.4 ergeben bei den entsprechenden Adaptern gemeldet werden.

Der Adapter funktioniert Augenscheinlich normal, jedoch wird das LOG mit WARN Meldungen übersäät.

`

fronius.0 | 2021-05-01 21:19:58.018 | warn | (3488) State value to set for "fronius.0.powerflow.BackupMode" has wrong type "boolean" but has to be "number"
-- | -- | -- | --

`

nkleber78, 2021-05-01:
@JBSullivan Kannst du bitte bei dir verifizieren dass der datenpunkt wirklich über den Adapter mit bool angelegt wird. Ich kann im code nichts finden wo dieser datenpunkt angelegt wird. Alle nicht definierten Datenpunkte werden lt. code als "mixed" angelegt. Meine Vermutung wäre dass dieser Datenpunkt entweder schon vorher mit diesem Typ angelegt wurde oder der typ nachträglich verändert wurde. Falls möglich mal diesen Datenpunkt löschen und dann den Adapter neu starten.'
Danke

JBSullivan, 2021-05-01:
Hier die RAW Daten des DP

`{
  "type": "state",
  "common": {
    "name": "BackupMode",
    "type": "number",
    "role": "value",
    "unit": "",
    "read": true,
    "write": false,
    "desc": "BackupMode"
  },
  "native": {},
  "from": "system.adapter.fronius.0",
  "user": "system.user.admin",
  "ts": 1605803112680,
  "_id": "fronius.0.powerflow.BackupMode",
  "acl": {
    "object": 1636,
    "state": 1636,
    "owner": "system.user.admin",
    "ownerGroup": "system.group.administrator"
  }
}`

Ich habe den DP dann gelöscht und via Instanz Restart neu erzeugen lassen. Siehe da, er hat dieses mal einen anderen Typen bekommen.

`{
  "type": "state",
  "common": {
    "name": "BackupMode",
    "type": "mixed",
    "role": "value",
    "unit": "",
    "read": true,
    "write": false,
    "desc": "BackupMode"
  },
  "native": {},
  "from": "system.adapter.fronius.0",
  "user": "system.user.admin",
  "ts": 1619900626174,
  "_id": "fronius.0.powerflow.BackupMode",
  "acl": {
    "object": 1636,
    "state": 1636,
    "owner": "system.user.admin",
    "ownerGroup": "system.group.administrator"
  }
}`

mcm1957, 2024-07-24:
Should be fixed with current versions.
If Problem still occures please open a new issue.

closed 2024-07-24 by mcm1957