iobroker-community-adapters/ioBroker.fronius #116: (24449) TypeError: Cannot read property 'Value' of undefined

Starter4711, 2021-05-02:
**Describe the bug**  
Fehlermeldung: 

fronius.0 | 2021-05-02 14:00:21.249 | warn | (24449) TypeError: Cannot read property 'Value' of undefined
-- | -- | -- | --

Alles Stable, aktuell.

nkleber78, 2021-05-02:
Duplikat von #76. Bitte testen und Ticket schließen wenn das so passt

Starter4711, 2021-05-02:
Hallo, sehe nicht was ich testen soll?

nkleber78, 2021-05-02:
Installation der letzen version von meinem Repository oder warten bis die Version offiziell released wird. Da leider keine Versionsinformationen von deiner Seite vorhanden sind wissen wir auch nicht was du für einen Stand hast
Ich denke es wird bald ein neues Stable release geben...

Starter4711, 2021-05-02:
Version 1.0.5 (stable)

mcm1957, 2023-05-31:
Duplikat von https://github.com/iobroker-community-adapters/ioBroker.fronius/issues/76.

closed 2023-05-31 by mcm1957