iobroker-community-adapters/ioBroker.fronius #119: Compatibility check and testing for Node.js 14 and 16

ioBrokerChecker, 2021-05-05:
Dear Adapter develop,

Node.js 14 is now available for a year and Node.js 16 was release just some days ago and will become LTS by October 2021. We plan to update the ioBroker Node.js recommendation (currently 12.x) to 14.x later this year.

Please check your adapter with Node.js 14 especially, and ideally also directly with Node.js 16

Please add both versions to the adapter testing which is executed on commits.

If your adapter requires a certain minimum version of Node.js please set the 'engine' setting in package.json accordingly! Please also do this if the adapter is not able to work in certain Node.js versions, so that ioBroker can prevent users from installing te adapter if not compatible!

On questions please talk to us at https://github.com/ioBroker/ioBroker.js-controller/issues/1138 

Please close the issue after you checked it.

Thank you very much for your support!

nkleber78, 2021-05-10:
@ldittmar81 Wie kann ich das machen? Habe leider keine Ahnung was ich da tun muss. Kannst du mir ggf. da helfen?

Apollon77, 2021-05-10:
Naja, erst einmal testing erweitern bedeutet in https://github.com/iobroker-community-adapters/ioBroker.fronius/blob/master/.github/workflows/test-and-release.yml#L53 die 16.x noch zu ergänzen.

Ansonsten ggf dein lokales Dev Syste auch nodejs mässig zu aktualisieren und zu testen

mcm1957, 2023-05-31:
obsolete

closed 2023-05-31 by mcm1957