iobroker-community-adapters/ioBroker.fronius #142: Keine Verbindung mehr zum GEN24  - IP ist aber OK

JBSullivan, 2021-08-15:
Da ich überhaupt nicht mehr weiter komme, hier nochmal die Beschreibung des Problems, wie ich es auch schon im Forum beschrieben habe.

---------------------------------------------------------------
Mahlzeit zusammen,
hatte von Euch schonmal jemand über den ioB Adapter von Fronius folgendes Problem?

```
fronius.0	2021-08-15 15:56:06.829	error	(13152) IP is not a Fronis inverter
```

BTW -> es muss FRONIUS und nicht FRONIS inverter heißen. Das ist aber Kleinkram

```
fronius.0	2021-08-15 15:52:34.376	debug	(13152) Unable to read data from inverters solarAPI
fronius.0	2021-08-15 15:52:34.376	debug	(13152) Exception thrown in check API: SyntaxError: Unexpected token < in JSON at position 0
fronius.0	2021-08-15 15:52:34.359	debug	(13152) Unable to read data from inverters solarAPI
fronius.0	2021-08-15 15:52:34.358	debug	(13152) Exception thrown in check API: SyntaxError: Unexpected token < in JSON at position 0
fronius.0	2021-08-15 15:52:34.222	debug	(13152) System command: C:\WINDOWS\system32\cmd.exe /s /c " C:\WINDOWS\system32\ping.exe -n 1 -w 2000 10.122.60.57 "
fronius.0	2021-08-15 15:52:34.214	debug	(13152) System command: C:\WINDOWS\system32\cmd.exe /s /c " C:\WINDOWS\system32\ping.exe -n 1 -w 2000 10.122.60.57 "
fronius.0	2021-08-15 15:52:34.188	info	(13152) starting. Version 1.1.3 in C:/iobroker/GLT/node_modules/iobroker.fronius, node: v14.16.0, js-controller: 3.3.15
fronius.0	2021-08-15 15:52:33.680	debug	(13152) statesDB connected
fronius.0	2021-08-15 15:52:33.679	debug	(13152) States connected to redis: 127.0.0.1:9000
fronius.0	2021-08-15 15:52:33.655	debug	(13152) States create User PubSub Client
fronius.0	2021-08-15 15:52:33.654	debug	(13152) States create System PubSub Client
fronius.0	2021-08-15 15:52:33.636	debug	(13152) Redis States: Use Redis connection: 127.0.0.1:9000
fronius.0	2021-08-15 15:52:33.633	debug	(13152) objectDB connected
fronius.0	2021-08-15 15:52:33.620	debug	(13152) Objects connected to redis: 127.0.0.1:9001
fronius.0	2021-08-15 15:52:33.591	debug	(13152) Objects client initialize lua scripts
fronius.0	2021-08-15 15:52:33.590	debug	(13152) Objects create User PubSub Client
fronius.0	2021-08-15 15:52:33.588	debug	(13152) Objects create System PubSub Client
fronius.0	2021-08-15 15:52:33.586	debug	(13152) Objects client ready ... initialize now
fronius.0	2021-08-15 15:52:33.522	debug	(13152) Redis Objects: Use Redis connection: 127.0.0.1:9001
host.GLT(GLT)	2021-08-15 15:52:31.276	info	instance system.adapter.fronius.0 started with pid 13152
host.GLT(GLT)	2021-08-15 15:52:29.375	info	instance system.adapter.fronius.0 terminated with code null ()
host.GLT(GLT)	2021-08-15 15:52:29.375	warn	instance system.adapter.fronius.0 terminated due to SIGTERM
host.GLT(GLT)	2021-08-15 15:52:29.242	info	stopInstance system.adapter.fronius.0 killing pid 15140
```

Seit Monaten läuft alles was mit Fronius zu tun hat 1A. Unter der im Adapter eingetragenen IP erreiche ich den WR direkt. Auch solarweb funktioniert einwandfrei und über den Fritzbox Router ist der WR ebenfalls unter der IP erreichbar.

Ich habe den Adapter, als auch ioB selber schon neu gestartet, aber der Fronius Adpter bleibt mit der oben genannten Fehlermeldung gelb und baut auch keine Verbindung mehr zum WR auf.

Das Phänomen trat irgendwann heute Vormittag auf, OHNE das ich in den letzten Tagen etwas verändert hätte - quasi von jetzt auf gleich.

Hat jemand eine Idee dazu?

EDIT: Ich habe in der Zwischenzeit nochmal alles mögliche ausprobiert, was mir so in den Sinn gekommen ist. Interessant daran ist, das der ModBus Adapter der auf die gleiche Adresse des Gerätes zugreift,  nach wie vor z.B. die Gehäuse Temperatur und bei Bedarf auch andere Werte abholt. 

Ich finde diese beiden Meldungen interessant - was ist mit dem Invalid Token gemeint`? Ist das etwas was ggf. nicht lokal läuft, sondern auf den Fronius Servern? Sind das ggf. anzeichen für das lange angekündigte Firmware Update incl. API 2.x?

```
fronius.0	2021-08-15 15:52:34.376	debug	(13152) Unable to read data from inverters solarAPI
fronius.0	2021-08-15 15:52:34.376	debug	(13152) Exception thrown in check API: SyntaxError: Unexpected token < in JSON at position 0
```

EDIT Teil 2: Hmmmm - es wird immer lustiger - jetzt habe ich den Adapter und die Objekte gelöscht und von GIT die Version 1.1.6 neu installiert.

Jetzt bekomme ich die folgende Meldung, obwohl der Adapter incl. IP konfiguriert ist ?!?!?!

```
fronius.0	2021-08-15 17:36:54.817	error	(11492) Please configure the Fronius adapter
fronius.0	2021-08-15 17:36:54.804	info	(11492) starting. Version 1.1.3 in C:/iobroker/GLT/node_modules/iobroker.fronius, node: v14.16.0, js-controller: 3.3.15
```

AHH - ich glaube jetzt weiß ich was er  will. Durch die Neuinstallation ist der Ausleseintervall verloren gegangen. Da die Prüfung der IP Adresse fehlt schlägt, kann man auch keinen Aktualisierungsintervall eingeben. Die IP ist aber 100%ig korrekt - verstehe es gerade nicht, warum er sich nicht mehr mit dem WR verbinden mag :(

nkleber78, 2021-08-15:
Hallo @JBSullivan 
kannst du mal versuchen die Daten direkt im Objekt system.adapter.fronius.0 einzustellen und dann den Adapter neu zu starten? Hast du kürzlich ein Update gemacht? Bzgl. Admin 5 ist es zumindest bei mir so (Linux mit Docker) dass der Adapter läuft. Aber ob es hier ggf. mit Windows Ping ein Problem gibt kann ich nicht beurteilen. Was vielleicht noch hilft ist ein API test... Was ist der Output von http://10.122.60.57/solar_api/GetAPIVersion.cgi bei dir? Denn es gibt hier einen Fehler den es eigentlich so nicht geben sollte...

JBSullivan, 2021-08-16:
Auf die URL bekomme ich die folgende Anzeige.

`{
   "APIVersion" : 1,
   "BaseURL" : "/solar_api/v1/",
   "CompatibilityRange" : "1.7-1"
}`

In der Zwischenzeit haben sich ein paar Neuigkeiten ergeben. Offensichtlich ist gestern zur gleichen Zeit meinen openWB Wallbox abgestürzt und softwaremäßig eingefroren. Das habe ich aber erst Spät am Abend gemerkt. Die openWB kommuniziert ja ebenfalls über das sogenannte PV Modul mit dem Wechselrichter.

Was ich allerdings am interessantesten fand, ist die Tatsache das nach einem Backup Restore vom Vortag ( Da war Admin noch auf 5.1.24) der Adapter wieder lief, JEDOCH unter den Objekten eine neue Baumstruktur eröffnet hat. Vorher wurden die Daten alle in .... inverter.0 geschrieben. Nach dem Restore, wurde in den Fronius Objekten ein ..... inverter.1 angelegt und dort alle Inverterdaten rein geschrieben. .... inverter.0 enthielt die alten Inhalte, welche aber nicht mehr aktualisiert wurden. Es werden jetzt nur noch die Daten in ....inverter.1 aktualisiert.

Ist natürlich, sau blöd, da ich ganz viel skripte, VIS Verknüpfungen und Influxdb/Grafana Variablen auf inverter.0 verknüpft hatte.

Ich habe gestern lediglich VOR dem besagten Fehler auf Admin 5.1.25 geupdatet. Mehr habe ich gestern nicht gemacht. Das ausbleiben der PV Werte ist mir erst gegen Mittag aufgefallen. Aktuell bin ich wieder bei Admin 5.1.24 und der Adapter läuft auch wieder. Meine ganzen Verknüpfungen habe ich auf ...... inverter.1 geändert.

nkleber78, 2021-08-16:
Ich hatte bei mir Admin 5.1.23 drauf... Sollte es wirklich mit dem letzten update von Admin zu tun haben dann müsstest du dort noch ein Ticket aufmachen. Wobei ich vermute das es ein unglücklicher Zufall mit deiner Wallbox war... Bzgl. Inverter ID: Leider liefert der WR das nicht korrekt über die API sodass ich ein Fallback programmiert habe. Dieser War in der ersten version fälschlicherweise 0 für den GEN24 obwohl er eigentlich die ID 1 hat. Somit habe ich das nochmals angepasst. Die Änderung wird aber nur angewendet wenn erneut "Check IP" im Admin interface gedrückt wird... Das war bei dir natürlich der Fall :(

nkleber78, 2021-08-16:
Habe gerade bei mir das update auf 5.1.25 gemacht, läuft ohne Probleme...

JBSullivan, 2021-08-16:
OK - dann wäre das mit der ID schonmal geklärt - war also ein kleiner BUG in einer der ersten Versionen. Aber meter.0 bleibt meter.0 - oder? Weil da habe ich auch noch einen Ordner - mit der Bezeichnung meter.0.

Admin 5.1.25 habe ich jetzt auch nochmal riskiert - und läuft auch bei mir unter Windows.

Das Absturz LOG der Wallbox ist gerade beim Hersteller zur analyse - vielleicht kommt dabei ja etwas heraus. Angeblich greift das Wallbox Modul für die Wechselrichter Daten beim GEN24 auf eine REST API zurück.  Für mein Verständnis ist das die API die Fronius zur Verfügung stellt - oder?

Das sich ioBroker, bzw. der Fronius Adapter mit dem Zugriff der Wallbox in die Quere gekommen sind? Die Wallbox schickt ja auch einen nicht gerade unerheblichen Teil ihrer Datenpunkte via MQTT an ioBroker.

nkleber78, 2021-08-16:
Meter bleibt 0 wie er soll, das passt so. Schön das es bei dir nun wieder läuft, hoffe dass das mit der Wallbox kein Dauerzustand ist...

laguna120, 2021-09-02:
Ich habe seit heute das gleiche Problem wie @JBSullivan . Der Adapter verbindet sich einfach nicht mehr mit meinem Symo, obwohl ich sicher die richtige IP eingetragen habe.
Über den Browser funktioniert der Zugriff. Als API-Version wird
{ "APIVersion" : 1, "BaseURL" : "/solar_api/v1/", "CompatibilityRange" : "1.6-3" }
geliefert.

nkleber78, 2021-09-02:
@laguna120 gibt es andere services die auf den Wechselricher zugreifen. Wie z.b. bei @JBSullivan  die Wallbox. Ich bin mir relativ sicher dass es bei ihm ein Blocker durch die Wallbox war... Was hast du auf dem WR für eine Firmware drauf?

laguna120, 2021-09-02:
Ich habe jetzt eine Weile überlegt. Mir ist nichts eingefallen, was lokal auf den WR zugreift. Ich habe durch die Probleme heute die Firmware vom WR auf 3.18.7-1 geupdatet.
Den Adapter habe ich jetzt auf 1.1.6 hochgezogen - hat alles nichts gebracht.

Hier die Logs:
`
<html><body>

fronius.0 | 2021-09-02 23:36:56.574 | error | Please configure the Fronius adapter
-- | -- | -- | --
fronius.0 | 2021-09-02 23:36:56.562 | info | starting. Version 1.1.6 in /opt/iobroker/node_modules/iobroker.fronius, node: v14.15.4, js-controller: 3.3.10
fronius.0 | 2021-09-02 23:36:56.423 | debug | statesDB connected
fronius.0 | 2021-09-02 23:36:56.422 | debug | States connected to redis: 127.0.0.1:9000
fronius.0 | 2021-09-02 23:36:56.378 | debug | States create User PubSub Client
fronius.0 | 2021-09-02 23:36:56.377 | debug | States create System PubSub Client
fronius.0 | 2021-09-02 23:36:56.367 | debug | Redis States: Use Redis connection: 127.0.0.1:9000
fronius.0 | 2021-09-02 23:36:56.365 | debug | objectDB connected
fronius.0 | 2021-09-02 23:36:56.362 | debug | Objects connected to redis: 127.0.0.1:9001
fronius.0 | 2021-09-02 23:36:56.351 | debug | Objects client initialize lua scripts
fronius.0 | 2021-09-02 23:36:56.351 | debug | Objects create User PubSub Client
fronius.0 | 2021-09-02 23:36:56.350 | debug | Objects create System PubSub Client
fronius.0 | 2021-09-02 23:36:56.349 | debug | Objects client ready ... initialize now
fronius.0 | 2021-09-02 23:36:56.334 | debug | Redis Objects: Use Redis connection: 127.0.0.1:9001
fronius.0 | 2021-09-02 23:36:53.385 | info | Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
fronius.0 | 2021-09-02 23:36:53.385 | info | terminating
fronius.0 | 2021-09-02 23:36:52.884 | info | Got terminate signal TERMINATE_YOURSELF

</body>
</html>`

laguna120, 2021-09-02:
OK, läuft wieder.
Neue Instanz angelegt mit gleicher IP -> lief.
Beide Instanzen gelöscht und neue fronius.0 angelegt -> läuft

Jetzt muss ich nur noch schauen, ob die Datenpunkte alle noch passen.

MauntyA1, 2022-08-03:
Muss mal diesen Thread nochmals hochholen. Bekomme keine Verbindung zu meinem GEN24 zusammen.
IP Adresse stimmt 100%, Neustart des WR durchgeführt, Instanz neu hinzugefügt aber ohne Erfolg.
Liegt es eventuell an der Fronius Software. Habe die neueste oben.

Auf den parallel installierten Symo Fronius WR komme ich ohne Prob rauf.

fronius.1 | 11365 | 2022-08-03 10:52:10.170 | error | IP is not a Fronis inverter

Rapptor33, 2023-01-31:
Bei mir besteht auch das selbe Problem.
Neustart WR, weitere Instanz hinzufügen hat leider nicht geholfen.

`| error | IP is not a Fronis inverter`

j0scha1, 2023-03-07:
Ich hatte das gleiche Problem.
Ausgangslage war eine Migration vom Pi zu LXC Container in Proxmox.
iobroker im lxc container ist als Multihost eingerichtet. Der Pi hängt noch dran wegen des ZWave Shields.
Nachdem ich den Adapter wieder auf den Pi geschoben habe, funktionierte alles wieder.

lindejo1979, 2023-05-30:
Hallo

Nachdem ich nun 1Jahr auf meine Anlage gewartet war mein Entsetzen groß, dass ich den Adpater nicht zum laufen bekomme. Ich habe schon so einige Tricks die hier gepostet wurden ausprobiert. Aber ohne Erfolg. Ich bekomme keien Daten aus dem Adapter. Verbindung ist Gelb.

Die Taste IP prüfen liefert folgende Fehler:
fronius.0 | 2023-05-30 19:29:39.127 | error | IP is not a Fronis inverter

Bin ich zu doof? Wo kann ich noch ansetzen. Es handelt sich um einen Symo Gen24.

Gruß

Jochen

nkleber78, 2023-05-30:
@lindejo1979 Welche Version des Adapters verwendest du? Versuch doch mal die 1.2.0 https://github.com/nkleber78/ioBroker.fronius Diese läuft schon bei ziemlich vielen ohne Probleme. Bei mir ebenfalls mit Symo und Symo GEN24. Wichtig ist dass der IP check gemacht wird wenn der WR wirklich in Betrieb ist (d.h. am Tag) In der Nacht kann es probleme geben falls der WR in den Standby mode geht...

lindejo1979, 2023-05-30:
@nkleber78: Ich trau mich es gar nicht zu schreiben. Ein echtes DAU Problem. Ich habe vergessen die API Schnittstelle zu aktivieren. So kann man einen Tag auch verdampfen. Danke für die schnelle Rückmeldung.

MauntyA1, 2023-05-30:
@nkleber78 : Habe noch nie von hier einen Adapter installiert. Wie mache ich das ?
Kann ich ein neues repository eintragen ?

mcm1957, 2023-05-30:
@nkleber78
Da du offensichtlich den Fronius Adapter weiterentwickelst / entwickelt hast spricht aus deiner Sicht etwas dagegen hier wieder mal saubere Releases zu machen?

Da du offensichtlich keinen vollen Zugang zu den Community Adapters hats, könnte ich einen aktuellen PR mergen und ne neue Release veranlassen. Ich hab (zumidnest derzeit) keine Froniusanlage und kann daher nichts testen, mit der Infrastruktur von ioB komm ich aber zurecht :-).

Ergo wenn di mithelfen willst, den Adapter auf saubere Füße zu stellen, meld dich bitte.
hier oder
mail [mcm57@gmx.at](mailto:mcm57@gmx.at)
telegram @mcm1957

nkleber78, 2023-06-04:
Es scheint kein echtes problem zu sein. Kann auch in der aktuellen Version 2.0.0 nicht reproduziert werden

closed 2023-06-04 by nkleber78