iobroker-community-adapters/ioBroker.fronius #144: Dokumentation Berechnung Werte (Eigenverbrauch, Hausverbrauch, Netzbezug etc) - Wiki?

m4ding, 2021-08-24:
Hi,

ich habe einige Blockly Skripte zur Berechnung der Daten geschrieben. Plant ihr die Berechnung ggf. in den Adapter zu integrieren oder macht es Sinn z.B. auf einer Wiki Page die Skripte etc zur Verfügung zu stellen? Ich stelle sie gern zur Verfügung. Beim Klick auf Wiki wird man auf Issues zurück geleitet.

VG
mading

empty0123, 2021-11-12:
ich wäre sehr interessiert an den Skripten

mcm1957, 2024-07-24:
Grundsätzlich sollten Adapter primär Daten die ein Gerät liefert 1:1 zur Verfügung stellen. Zusätzliche Berechnungen machen nur in wenigen Fällen direkt einen Sinn da sie oft vom konkreten Verwendungszweck abhängen.

Derzeit sind keine Erweiterungen diesbezüglich geplant.

closed 2024-07-24 by mcm1957