iobroker-community-adapters/ioBroker.fronius #145: Präventiver Hinweis Fronius SYMO Update

JBSullivan, 2021-08-24:
Ich greife hier mal die Info aus dem Photovoltaik Forum auf, das es offensichtlich mit dem Datenmanager Update auf die Version 3.18.7-1 zu Inkompatibilitäten mit den json abfragen (API?) kommen kann.

Ich weiß nicht, ob das für den Adapter hier überhaupt von Relevanz ist, aber man kann es ja mal erwähnen.

Hier das Posting aus dem PV-Forum

`Guten Morgen,

nach dem gestrigen FW Update auf die Version 3.18.7-1 funktionierte der JSON-Abruf nicht mehr. Die manuelle Abfrage ergab: Der Fehler steckt(e) im Detail, lediglich "DeviceID musste zu DeviceId geänderrt werden.

Die manuelle Abfrage brachte den Fehler :

Code
CGI-Args: Invalid parameter '' for attribute 'DeviceId' (must be numeric)
Geändert werden musste entsprechend:

Code
"http://192.168.178.60/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DeviceID=1&DataCollection=CommonInverterData"
zu

Code
"http://192.168.178.60/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DeviceId=1&DataCollection=CommonInverterData"
Vielleicht hilft es dem ein oder anderen!

Liebe Grüße`

nkleber78, 2021-08-24:
Hallo @JBSullivan 
Danke für den Input. Glücklicherweise haben wir im Adapter schon die Kleinschreibung verwendet welche bisher auch funktioniert hat. In der neuen Version funktioniert offensichtlich wirklich nur noch die kleingeschriebene Variante...

Gruß Norbert

nkleber78, 2023-05-31:
@JBSullivan Bitte dieses issue schließen. Danke

JBSullivan, 2023-05-31:
Ich glaube nicht das ICH das schließen kann - es ist ja dein GIT Konto. Wenn doch, möge man mir einen kurzen Hinweis geben wo ich das machen kann.

closed 2023-05-31 by mcm1957