iobroker-community-adapters/ioBroker.fronius #152: Sensorbox Daten werden nicht angezeigt

doemb33, 2021-09-28:
Adapter liefert keine Werte der Sensorbox
[image]

[image]

Hallo miteinander,

ich bin neu hier und ersuche um Hilfe.
mache ich was falsch, dass ich die Werte der Sensorbox nicht bekomme oder ist meine Vermutung richtig, dass diese Werte noch keiner gebraucht hat.

[image]

LG Johannes

nkleber78, 2021-09-28:
@doemb33 Du hast leider recht, die Sensordaten hat bisher keiner gebraucht und somit sind zwar die Strukturen eingefügt, aber nicht die entsprechenden Datenpunkte. Wenn du uns eine Beschreibung liefern kannst welche daten du genau benötigst, dann kann ich mir das mal in nächster Zeit anschauen wie hoch der  Aufwand dafür ist. Wenn du mit automatisch generierten Datenpunkten zufrieden bist, dann geht das ggf. relativ einfach. Allerdings sind dann alle daten vom typ "mixed", und der Name und die Beschreibung entsprechen dem Text der auf der API geliefert wird und Einheit gibt es nicht. Das hab ich bereits für den Meter so implementiert als fallback...

nkleber78, 2021-09-28:
Btw. Da ich keine Sensorbox habe, bitte den response von folgenden API calls hier als file anhängen 
GetSensorRealtimeData.cgi?Scope=Device&DeviceId=1&DataCollection=NowSensorData bzw. 
GetSensorRealtimeData.cgi?Scope=Device&DeviceId=1&DataCollection=MinMaySensorData

doemb33, 2021-09-28:
[Bedienungsanleitung Fronius Solar API V1a.pdf](https://github.com/iobroker-community-adapters/ioBroker.fronius/files/7245757/Bedienungsanleitung.Fronius.Solar.API.V1a.pdf)
[GetSensorRealtimeData.cgi.json.txt](https://github.com/iobroker-community-adapters/ioBroker.fronius/files/7245802/GetSensorRealtimeData.cgi.json.txt)
[GetSensorRealtimeDataMinMax.cgi.json.txt](https://github.com/iobroker-community-adapters/ioBroker.fronius/files/7245806/GetSensorRealtimeDataMinMax.cgi.json.txt)
Hallo Norbert,

anbei wie gewünscht; die automatisch generierten Datenpunkte sollten reichen.

Danke im voraus

LG Johannes

nkleber78, 2023-06-04:
Implementierung ist in V2.0.0

closed 2023-06-04 by mcm1957