iobroker-community-adapters/ioBroker.fronius #168: Fronius Adapter Warning solange keine Energie geliefert wird

Hard-n-Soft, 2021-11-06:
Hallo,

nachdem ich den ioBroker neu aufgesetzt habe und auch das Backup per FTP funktioniert ist nun auch der Fronius Adapter für den Gen24 Wechselrichter in Betrieb und liefert munter Daten solange die Sonne scheint. Abfrageinterval 10 Sekunden.

Außerhalb des Zeitraumes liefert der Adapter doch folgende Meldung und überschwemmt das Log

```
fronius.0 | 209780 | 2021-11-06 07:43:50.096 | warn | TypeError: Cannot read property 'Value' of undefined
fronius.0 | 209780 | 2021-11-06 07:43:40.110 | warn | TypeError: Cannot read property 'Value' of undefined
fronius.0 | 209780 | 2021-11-06 07:43:30.093 | warn | TypeError: Cannot read property 'Value' of undefined
fronius.0 | 209780 | 2021-11-06 07:43:20.095 | warn | TypeError: Cannot read property 'Value' of undefined
fronius.0 | 209780 | 2021-11-06 07:43:10.093 | warn | TypeError: Cannot read property 'Value' of undefined
fronius.0 | 209780 | 2021-11-06 07:43:00.094 | warn | TypeError: Cannot read property 'Value' of undefined
fronius.0 | 209780 | 2021-11-06 07:42:50.107 | warn | TypeError: Cannot read property 'Value' of undefined
fronius.0 | 209780 | 2021-11-06 07:42:40.111 | warn | TypeError: Cannot read property 'Value' of undefined

```
Ich denke sobald die PV Anlage Strom liefert und der Wechselrichter aktiv arbeitet kommt keine Fehlermeldung. Lässt sich da was machen?

Gruß Hard-n-Soft

nkleber78, 2021-11-06:
Liegt vmtl. daran dass nicht die letzte Version installiert ist. Bei der 1.1.6 sollte das nicht auftreten, zumindest tritt es bei mir nicht auf

Hard-n-Soft, 2021-11-06:
Hallo,
danke für die schnelle RM. Bezieht sich die 1.1.6 auf die Adapter Version? Bei mir steht als aktuelle die 1.0.5 zur Auswahl.

[image]

nkleber78, 2021-11-06:
Das ist leider nicht im Stable Channel. Du musst also auf den Beta wechseln und dann Updaten… oder direkt von GitHub

mcm1957, 2023-05-31:
Dürfte in 2.0.0 behoben sein.
zu testen ,..

mcm1957, 2024-07-24:
Kein (negatives) Feedback - lets close here.

closed 2024-07-24 by mcm1957