iobroker-community-adapters/ioBroker.fronius #169: Adapter doesn´t work without ping util

smolo-de, 2021-11-28:
I have installed the adapter on a virtual system and a container on a new system and the adapter doesn´t work. The problem is the missing  ping util!

Resolution for the problem:
> Docker-Counter > apt-get install iputils-ping

**Versions:**  
 - Adapter version: 1.1.6
 - JS-Controller version: 3.3.18
 - Node version: V12.22.7
 - Operating system: Docker

nkleber78, 2021-11-29:
@smolo-de This is a good point, but that should be handled on the docker container. So I would suggest that you open a ticket on the docker container to add the iputils-ping to the container. Which container are you using?

smolo-de, 2021-11-29:
@nkleber78 I use the official docker container from buanet https://github.com/buanet/ioBroker.docker

I don't think it's a good idea to assume this function. I have the identical problem with an Ubuntu container that was set up manually.

When the adapter starts it hangs with the ping command but applies some of the variables correctly that has been more than confusing at this point that it does something but not everything ;-)

nkleber78, 2021-11-29:
Using ping is from my point of view the correct way as that simply checks if the IP is accessible and is not dependent on any specific API implementation. That ping is not available per default on docker containers is another thing. An idea would be to provide a proper error message that tells that ping is not supported and must be installed. But still, for the official docker container for ioBroker from buanet it makes sense to include the iputils in the distribution

smolo-de, 2021-11-29:
I share your opinion and I can of course request this at Buanet but on the other hand it is so that so many adapters exist and this is the first for which I had to install something in addition.

Regardless, an error message is never wrong :-)

coolcow, 2022-10-08:
@smolo-de 
> > Docker-Counter > apt-get install iputils-ping

Thx very much, this solved the problem for me too ! Had to do a `apt-get update` before though.

The problem with ping is that crafting ICMP packages requires root permissions. That's why the usual way is to use a wrapper that calls the ping command.

But... @nkleber78 
...is ping really the only possibility to check if the inverter is alive? Couldn't the adapter just call an URL and check the response status code? Maybe to be considered...

mcm1957, 2023-05-31:
@nkleber78 
If I read the PR correctly, ping has been removed for upcomi ng 2.0.0?
Please somment.

nkleber78, 2023-05-31:
@mcm1957 Yes it is true ping is no longer used in the current implementation. Therefore issue is resolved in the upcomming 2.0.0 release

closed 2023-05-31 by nkleber78