iobroker-community-adapters/ioBroker.fronius #176: State "fronius.0.inverter.1.FAC" has no existing object, this might lead to an error in future versions

firex2, 2021-12-14:
Ich habe den Fronius Symo GEN24 10.0 Plus und bekomme sehr oft die Meldung `State "fronius.0.inverter.1.FAC" has no existing object, this might lead to an error in future versions`. Habe den Adapter schon neu gestartet, hat nicht geholfen. Was kann ich da tun?

[image]

maloross, 2022-02-01:
Bei mir hat dieser DP und noch ein paar andere ebenfalls gefehlt. Ich habe die fehlenden DP selbst erstellt ( im Expertenmodus, DP als Zahl definiert). Sofort wurden auch Werte eingetragen
`{
  "common": {
    "name": "FAC",
    "desc": "Manuell erzeugt",
    "role": "state",
    "type": "number",
    "read": true,
    "write": true,
    "def": ""
  },
  "type": "state",
  "_id": "fronius.0.inverter.1.FAC",
  "acl": {
    "object": 1636,
    "state": 1636,
    "owner": "system.user.admin",
    "ownerGroup": "system.group.administrator"
  },
  "from": "system.adapter.admin.0",
  "user": "system.user.admin",
  "ts": 1643716695085
}`

Bis jetzt alles ok, werde aber nach demnächst mal schauen, ob nach Systemneustart (oder Adapter-Update) die selbst erstellt DP auch bleiben ;)

mcm1957, 2023-05-31:
see https://github.com/iobroker-community-adapters/ioBroker.fronius/issues/206

closed 2023-06-04 by nkleber78