iobroker-community-adapters/ioBroker.fronius #179: Weitere Informationen zur Batterie integrieren?

marlan99, 2021-12-23:
Wenn man eine Batterie am Symo Hybrid oder Gen24 Wechselrichter angeschlossen hat, kann man zusätzliche Daten abrufen.
Kann man diese Funktion bitte integrieren?

Beispiel:
http://192.168.1.71/solar_api/v1/GetStorageRealtimeData.cgi?Scope=System

gibt ein Beispielresultat von:
`
{
   "Body" : {
      "Data" : {
         "0" : {
            "Controller" : {
               "Capacity_Maximum" : 7200,
               "Current_DC" : 0,
               "DesignedCapacity" : 7200,
               "Details" : {
                  "Manufacturer" : "Fronius",
                  "Model" : "Fronius Solar Battery",
                  "Serial" : "27443123"
               },
               "Enable" : 1,
               "StateOfCharge_Relative" : 18,
               "Temperature_Cell" : 18.25,
               "TimeStamp" : 1640277702,
               "Voltage_DC" : 309.39999999999998,
               "Voltage_DC_Maximum_Cell" : 3.2360000000000002,
               "Voltage_DC_Minimum_Cell" : 3.2109999999999999
            },
            "Modules" : [
               {
                  "Capacity_Maximum" : 1200,
                  "Current_DC" : -0.02,
                  "CycleCount_BatteryCell" : 932,
                  "DesignedCapacity" : 1200,
                  "Details" : {
                     "Manufacturer" : "Sony",
                     "Model" : "unknown",
                     "Serial" : "S013016737 "
                  },
                  "Enable" : 1,
                  "StateOfCharge_Relative" : 23,
                  "Status_BatteryCell" : 53,
                  "Temperature_Cell" : 18.850000000000023,
                  "Temperature_Cell_Maximum" : 19.450000000000045,
                  "Temperature_Cell_Minimum" : 18.550000000000011,
                  "TimeStamp" : 1640277692,
                  "Voltage_DC" : 51.621000000000002,
                  "Voltage_DC_Maximum_Cell" : 3.2320000000000002,
                  "Voltage_DC_Minimum_Cell" : 3.2200000000000002
               },
               {
                  "Capacity_Maximum" : 1200,
                  "Current_DC" : -0.02,
                  "CycleCount_BatteryCell" : 932,
                  "DesignedCapacity" : 1200,
                  "Details" : {
                     "Manufacturer" : "Sony",
                     "Model" : "unknown",
                     "Serial" : "S013016995 "
                  },
                  "Enable" : 1,
                  "StateOfCharge_Relative" : 22,
                  "Status_BatteryCell" : 53,
                  "Temperature_Cell" : 18.850000000000023,
                  "Temperature_Cell_Maximum" : 19.350000000000023,
                  "Temperature_Cell_Minimum" : 18.550000000000011,
                  "TimeStamp" : 1640277692,
                  "Voltage_DC" : 51.552999999999997,
                  "Voltage_DC_Maximum_Cell" : 3.2269999999999999,
                  "Voltage_DC_Minimum_Cell" : 3.2160000000000002
               },
               {
                  "Capacity_Maximum" : 1200,
                  "Current_DC" : -0.02,
                  "CycleCount_BatteryCell" : 936,
                  "DesignedCapacity" : 1200,
                  "Details" : {
                     "Manufacturer" : "Sony",
                     "Model" : "unknown",
                     "Serial" : "S013016997 "
                  },
                  "Enable" : 1,
                  "StateOfCharge_Relative" : 18,
                  "Status_BatteryCell" : 53,
                  "Temperature_Cell" : 18.550000000000011,
                  "Temperature_Cell_Maximum" : 19.25,
                  "Temperature_Cell_Minimum" : 18.350000000000023,
                  "TimeStamp" : 1640277692,
                  "Voltage_DC" : 51.584000000000003,
                  "Voltage_DC_Maximum_Cell" : 3.2360000000000002,
                  "Voltage_DC_Minimum_Cell" : 3.2109999999999999
               },
               {
                  "Capacity_Maximum" : 1200,
                  "Current_DC" : -0.01,
                  "CycleCount_BatteryCell" : 928,
                  "DesignedCapacity" : 1200,
                  "Details" : {
                     "Manufacturer" : "Sony",
                     "Model" : "unknown",
                     "Serial" : "S013017000 "
                  },
                  "Enable" : 1,
                  "StateOfCharge_Relative" : 22,
                  "Status_BatteryCell" : 53,
                  "Temperature_Cell" : 18.350000000000023,
                  "Temperature_Cell_Maximum" : 19.050000000000011,
                  "Temperature_Cell_Minimum" : 18.050000000000011,
                  "TimeStamp" : 1640277692,
                  "Voltage_DC" : 51.576000000000001,
                  "Voltage_DC_Maximum_Cell" : 3.2309999999999999,
                  "Voltage_DC_Minimum_Cell" : 3.2170000000000001
               },
               {
                  "Capacity_Maximum" : 1200,
                  "Current_DC" : -0.02,
                  "CycleCount_BatteryCell" : 931,
                  "DesignedCapacity" : 1200,
                  "Details" : {
                     "Manufacturer" : "Sony",
                     "Model" : "unknown",
                     "Serial" : "S013016785 "
                  },
                  "Enable" : 1,
                  "StateOfCharge_Relative" : 22,
                  "Status_BatteryCell" : 53,
                  "Temperature_Cell" : 17.950000000000045,
                  "Temperature_Cell_Maximum" : 18.550000000000011,
                  "Temperature_Cell_Minimum" : 17.550000000000011,
                  "TimeStamp" : 1640277692,
                  "Voltage_DC" : 51.609999999999999,
                  "Voltage_DC_Maximum_Cell" : 3.2330000000000001,
                  "Voltage_DC_Minimum_Cell" : 3.218
               },
               {
                  "Capacity_Maximum" : 1200,
                  "Current_DC" : -0.02,
                  "CycleCount_BatteryCell" : 935,
                  "DesignedCapacity" : 1200,
                  "Details" : {
                     "Manufacturer" : "Sony",
                     "Model" : "unknown",
                     "Serial" : "S013016869 "
                  },
                  "Enable" : 1,
                  "StateOfCharge_Relative" : 24,
                  "Status_BatteryCell" : 53,
                  "Temperature_Cell" : 17.450000000000045,
                  "Temperature_Cell_Maximum" : 18.050000000000011,
                  "Temperature_Cell_Minimum" : 17.25,
                  "TimeStamp" : 1640277692,
                  "Voltage_DC" : 51.552,
                  "Voltage_DC_Maximum_Cell" : 3.2280000000000002,
                  "Voltage_DC_Minimum_Cell" : 3.214
               }
            ]
         }
      }
   },
   "Head" : {
      "RequestArguments" : {
         "DeviceClass" : "Storage",
         "Scope" : "System"
      },
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2021-12-23T17:41:42+01:00"
   }
}
`

IceBoosteR, 2022-01-16:
Super Idee, dem schließe ich mich an. Leider kann ich bei mir nicht ordentlich testen, weshalb ich vom rumgebastele in einem weiteren Branch absehe, und das den Profis überlasse. Deswegen teile ich mal mein Outcome aus dem API call, damit mehrere Beispiele vorliegen. Läuft an einem SymoGen 24

`<html><body>
<div id="content"><div class="tabs "><nav class="tabs-navigation"><ul class="tabs-menu" role="tablist"><li class="tabs-menu-item json is-active" role="presentation"><span class="devtools-tab-line"></span></li></ul></nav></div></div><span class="devtools-tab-line"></span><span class="devtools-tab-line"></span><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"></div></div></div></div><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"><div class="devtools-separator"></div><input class="searchBox devtools-filterinput" placeholder="JSON durchsuchen"></div><div class="panelContent">

  |  
-- | --
Body |  
Data |  
0 |  
Controller |  
Capacity_Maximum | 10240
Current_DC | 0
DesignedCapacity | 10240
Details |  
Manufacturer | "BYD"
Model | "BYD Battery-Box Premium HV"
Serial | "P030Txxxxxxxxx     "
Enable | 1
StateOfCharge_Relative | 20.8
Status_BatteryCell | 3
Temperature_Cell | 14
TimeStamp | 1642338059
Voltage_DC | 415.5
Modules | []
Head |  
RequestArguments |  
Scope | "System"
Status |  
Code | 0
Reason | ""
UserMessage | ""
Timestamp | "2022-01-16T13:01:00+00:00"

</div></div></div></div>
</body>
</html>`

nkleber78, 2022-01-17:
@IceBoosteR, @marlan99 Diese Daten sind bereits integriert in der Version 1.1.6. bitte schaut mal ob bei euch eine Meldung im Debug mode bzgl. Storage kommt. Ggf. auch mal in die Adapter settings gehen und auf checkIP drücken um sicherzustellen dass Storage auch richtig erkannt wird. Falls alles nichts hilft könnt ihr auch die Version von meinem Repo (1.2.0) installieren dort sind noch diverse fixes/verbesserungen drin. Bin gerade dabei die Objekte dynamisch aufzubauen, was bedeutet dass die Struktur neu wäre. Vielleicht könnt ihr mir da auch feedback geben...
Alte struktur: [fronius.json.txt](https://github.com/iobroker-community-adapters/ioBroker.fronius/files/7884202/fronius.json.txt)
Neue struktur: [javascript.0.PV_Schwarzach_test.json.txt](https://github.com/iobroker-community-adapters/ioBroker.fronius/files/7884201/javascript.0.PV_Schwarzach_test.json.txt)

Was haltet ihr davon? Ich finde es wäre besser aufgeräumt, allerdings sind Objekte die derzeit aufgezeichnet werden an einer anderen stelle -> aufwand bei der Umstellung :(

marlan99, 2022-01-19:
nkleber78 Danke, Ich habe es soeben getestet.
Mit der Version 1.1.6 hatte ich keine Datenobjekte für "storage".
Mit deiner Version im git V2.0.0 wurde dann auch der Baum für storage erstellt mit einigen globalen informationen.

Ich denke die Daten von @IceBoosteR aufgelistet sind darin enthalten.
Mein Beispiel enthält aber auch die Detailinformationen aller 2-8 Akkumodule des Fronius Akku.
Diese werden auch mit der V2.0.0 nicht als Datenobjekt angelegt.

nkleber78, 2022-01-19:
@marlan99 Das wäre genau die Idee mit der neuen Struktur wo das JSON dynamisch geparst wird. Bisher müssen die Objekte programmiert werden da diese Art der Struktur speziell ist. Andrerseits wird dieser Akku auch nicht mehr Verkauft da Fronius dein eigenen komplett abgekündigt hat und nun auf BYD geht...

marlan99, 2022-01-19:
Die im V2 generierten Objekte werden wunderbar angelegt.
Bekomme initial einfach für jedes Objekt eine Warnung:

`
fronius.0 | 2022-01-19 13:17:10.390 | warn | State "fronius.0.inverter.1.Power_DC_String_2" has no existing object, this might lead to an error in future versions
-- | -- | -- | --
fronius.0 | 2022-01-19 13:17:10.389 | warn | State "fronius.0.inverter.1.Voltage_DC_String_2" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-01-19 13:17:10.375 | warn | State "fronius.0.inverter.1.Power_DC_String_1" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-01-19 13:17:10.373 | warn | State "fronius.0.inverter.1.Voltage_DC_String_1" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-01-19 13:17:10.372 | warn | State "fronius.0.inverter.1.Current_DC_String_2" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-01-19 13:17:10.370 | warn | State "fronius.0.inverter.1.Current_DC_String_1" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-01-19 13:16:55.568 | warn | State "fronius.0.inverterinfo.1.UniqueID" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-01-19 13:16:55.567 | warn | State "fronius.0.inverterinfo.1.StatusCode" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-01-19 13:16:55.565 | warn | State "fronius.0.inverterinfo.1.Show" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-01-19 13:16:55.563 | warn | State "fronius.0.inverterinfo.1.PVPower" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-01-19 13:16:55.561 | warn | State "fronius.0.inverterinfo.1.ErrorCode" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-01-19 13:16:55.559 | warn | State "fronius.0.inverterinfo.1.DTString" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-01-19 13:16:55.558 | warn | State "fronius.0.inverterinfo.1.DT" has no existing object, this might lead to an error in future versions
`

Dass die der Fronius Akku abgekündigt ist, ist korrekt. Ich hoffe trotzdem, dass meine noch Jahre funktioniert, zumal die Garantie erst diesen Sommer ausläuft :-)

marlan99, 2022-01-19:
Wenn ich es korrekt sehe, dann wird wohl der Abschnitt controller dynamisch angelegt, jedoch der zweite Baum mit Namen "modules" nicht.
Evtl. müsste man nur noch dies berücksichtigen?

nkleber78, 2022-01-19:
@marlan99 Bitte verwende mal folgendes script. Dann siehst du die neue Struktur und aus meiner Sicht müssten deine Daten auch korrekt geparst werden. Bitte prüfe die settings im script und schau dir an ob alle objekte die du heute hast auch in der neuen Struktur (wenn auch an anderer stelle) wieder findest
[Parse structure.js.txt](https://github.com/iobroker-community-adapters/ioBroker.fronius/files/7900002/Parse.structure.js.txt)

marlan99, 2022-01-20:
Hierbei müsstest du mir behilflich sein, was ich genau mit dieser Datei machen muss.
Ich kann sie inhaltlich lesen, weiss aber nicht, wie ich sie verwenden soll.

nkleber78, 2022-01-20:
@marlan99 dafür müsstest du den Javascript adapter verwenden.... https://github.com/ioBroker/ioBroker.javascript und den Inhalt dort in ein script kopieren. Wichtig ist auch dass axios in den npm modulen in den adapter settings drin ist
[image]
[image]

marlan99, 2022-01-20:
klappt soweit gut.
Beim erstmaligen ausführen wurde die Struktur angelegt (es gab diverse Warnings, aber nur das erste Mal)
Bei jedem weiteren Ausführen des Scripts wurde die Objekte befüllt.
[image]

marlan99, 2022-01-20:
Jetzt habe ich noch eine Fehlermeldung erhalten, kann sie aber nicht zuordnen:

javascript.0 | 2022-01-20 10:26:02.684 | error | at processTicksAndRejections (internal/process/task_queues.js:95:5)
-- | -- | -- | --
javascript.0 | 2022-01-20 10:26:02.683 | error | at runMicrotasks (<anonymous>)
javascript.0 | 2022-01-20 10:26:02.682 | error | at script.js.common.Fronius_Test:82:22
javascript.0 | 2022-01-20 10:26:02.680 | error | script.js.common.Fronius_Test: TypeError: Cannot read property 'url' of undefined
javascript.0 | 2022-01-20 10:26:02.676 | error | script.js.common.Fronius_Test: Cannot read property 'Data' of undefined
javascript.0 | 2022-01-20 10:26:02.674 | info | script.js.common.Fronius_Test: http://192.168.1.71/solar_api/v1/GetArchiveData.cgi?Scope=System&StartDate=20.1.2022&EndDate=20.1.2022&Channel=Current_DC_String_1&Channel=Current_DC_String_2&Channel=Temperature_Powerstage&Channel=Voltage_DC_String_1&Channel=Voltage_DC_String_2
javascript.0 | 2022-01-20 10:26:02.673 | info | script.js.common.Fronius_Test: GetArchiveData
javascript.0 | 2022-01-20 10:25:47.903 | info | Stop script script.js.common.Fronius_Test
javascript.0 | 2022-01-20 10:25:44.797 | info | script.js.common.Fronius_Test: GetLoggerInfo
javascript.0 | 2022-01-20 10:25:44.297 | info | script.js.common.Fronius_Test: GetInverterInfo
javascript.0 | 2022-01-20 10:25:44.094 | info | script.js.common.Fronius_Test: GetPowerflowRealtimeData
javascript.0 | 2022-01-20 10:25:44.034 | info | script.js.common.Fronius_Test: GetSensorRealtimeData
javascript.0 | 2022-01-20 10:25:43.970 | info | script.js.common.Fronius_Test: GetMeterRealtimeData
javascript.0 | 2022-01-20 10:25:43.705 | info | script.js.common.Fronius_Test: GetInverterRealtimeData_common
javascript.0 | 2022-01-20 10:25:43.512 | info | script.js.common.Fronius_Test: GetInverterRealtimeData with id=1
javascript.0 | 2022-01-20 10:25:43.423 | info | script.js.common.Fronius_Test: GetStorageRealtimeData
javascript.0 | 2022-01-20 10:25:42.939 | info | script.js.common.Fronius_Test: registered 0 subscriptions and 0 schedules

nkleber78, 2022-01-20:
@marlan99 Den Fehler kannst du ignorieren da du keine Archivdaten hast in deinem WR. Hast du alle elemente wie bisher auch in der neuen Struktur? Dass du zuerst warnungen bekommst und beim 2. run gefüllt werden ist bekannt und klar

marlan99, 2022-01-20:
Es hat einige Unterschiede, wobei ich vermute dass ein paar Objekte vom Adapter und nicht von Fronius kommen wie:
fronius.0.info.connection
fronius.0.info.lastsync
fronius.0.info.lastsyncarchive

Inverters.1.CustomName ist etwas unleserliches: &#77;&#117;&#101;&#108;&#105;&#115;&#116;&#101;&#105;&#103;
Vom jetzigen Adapter hate er diesen Inhalt: Muelisteig 

anbei die beiden Bäume vom Adapter und vom Script
[fronius.0.json.txt](https://github.com/iobroker-community-adapters/ioBroker.fronius/files/7904710/fronius.0.json.txt)
[javascript.0.PV_test.json.txt](https://github.com/iobroker-community-adapters/ioBroker.fronius/files/7904711/javascript.0.PV_test.json.txt)

nkleber78, 2022-01-20:
Das klingt perfekt, bist du mit der neuen Struktur glücklich oder hättest du da Änderungswünsche? Bin mir noch nicht sicher ob dieser change bei allen gut ankommt, wäre aber für die zukünftige Weiterentwicklung viel einfacher/stabiler.

marlan99, 2022-01-20:
Für mich passt dies super gut.
Die Umstellung wird mich ca. 1h kosten, da ich einige Objekte in einer InfluxDB speichere und die vis Visu anpassen muss (habe bewusst auf Alias verzichtet).

IceBoosteR, 2022-01-24:
@nkleber78 
Moin, danke erstmal für deine flotte Antwort. Ein Problem auf meinem IoBroker scheint zu sein, dass er keine aktuellere Version als 1.1.3 findet. Was natürlich eine Problem ist, da 1.1.6 schon länger draußen ist. Da muss ich gleich mal schauen wie ich das manuell update.
Dann habe ich mir die beiden Strukturen angeschaut. Ja es wäre eine Migration notwendig, aber es hält sich alles im Rahmen. Solange das Konstrukt soweit dynamisch ist, dass auch mit zusätzlichen Datenfeldern, die ggf. mal hinzukommen, die Struktur im Kern so bleibt, sehe ich da absolut kein Problem. Und das war ja auch die Intention dahinter. Zusätzlich fällt dir damit weniger Arbeit an, sodass du dich auch eher auf Bugs o.Ä. stürzen kannst, statt Objekte nachzupflegen.

nkleber78, 2022-01-25:
@IceBoosteR Das mit den Releases ist leider so eine Sache. Bin da vom Owner des repository abhängig um releases zu erstellen. Ich selber kann das nicht und der Owner hat wenig Interesse :(

IceBoosteR, 2022-02-05:
Perfekt. Sowohl die Version 1.1.6 tut seine Dienste ohne Probleme, wie auch die direkte Installation aus deinem Repo @nkleber78 von Version 1.2.0
Mir werden nun auch die Batterie-Infos angezeigt. Bin Happy :)

Ak11rA, 2023-03-08:
Was in Zeiten dynamischer Stromtarife noch cool wäre: Den Schalter "Charge battery from grid" steuern. Aber das gibt die Fronius API gar nicht her, oder?

simply-inventive, 2023-05-24:
Wenn man iobroker mit node-red verwendet könnte man auch einfach einen http-request machen und den Output nach json formatieren. So mache ich es.

mcm1957, 2023-05-24:
Wenn eine INfo direkt aus dem html output extrahiert werden kann, dann könnte auch der parser Adapter helfen:
https://github.com/ioBroker/ioBroker.parser

nkleber78, 2023-06-04:
Die Batterie infos sind integriert in der V2.0.0 und eine Steuerung über die API ist generell nicht vorgesehen/möglich

closed 2023-06-04 by nkleber78