iobroker-community-adapters/ioBroker.fronius #185: Fronius Gen24 10 keine Verbindung

Fischotter, 2022-01-01:
Hallo zusammen,
vielleicht kann mir jemand helfen.
Ich habe den Adapter in meinem IOBroker installiert (VM Linux Proxmox)
Verbindungsaufbau mit HOST und Lebenszeichen i.o.
Verbindung mit dem Gerät bleibt rot.
Im LOG ist kein Fehler zu sehen nur der debug:
fronius.0 | 2022-01-01 16:12:56.836 | debug | System command: /bin/ping -n -w2 -c1 192.168.178.68

Datenquelle:
pilot-05e Ver.1.14.110-1 SYMO GEN24

Adapter Ver. 1.1.3 
nodeJS Vers. 14.17.1
JS controller 3.3.21

Gibt es noch was, das ich kontrollieren oder einstellen kann?

Gruß fischotter

Fischotter, 2022-01-02:
Hallo zusammen,
habe die Löung in einem anderen issue gefunden.
An der Konsole im Proxmox den Befehl:
sudo chmod u+s `which ping` eingegeben. 
Jetzt läufts

closed 2022-01-02 by Fischotter

fpfisterer, 2023-03-21:
Hatte dasselbe Problem im Proxmox, der genannte Befehlt hat auch bei mir geholfen. 
Danke!