iobroker-community-adapters/ioBroker.fronius #188: Nach SW Update Symo 8.2-3-M im DataManager auf die 3.20.6-1 kein PV-Total mehr

weindler, 2022-01-30:
Hi,

habe gestern das Update gemacht, auf die 3.20.6-1, doch leider sind nun die Pv-Total Daten (gesamtertrag) weg.  Vielleicht haben sich hier Datenpunkte geändert oder sind welche dazugekommen. Könntest du vielleicht mal bitte einen Blick drauf werfen.

Adapter Version 1.1.3

Danke

weindler, 2022-01-31:
Nachricht von Fronius:

vielen Dank für Ihr Vertrauen in Fronius und unsere Produkte.

Es handelt sich bei dem von Ihnen beschriebenen Verhalten um einen Bug.
Dieser soll mit dem nächsten Update des Datamanagers (Ende Februar 2022) behoben werden. 

also sollte es dann wieder behoben werden, ansonsten gibt es noch einen Datenpunkt der die Gesamtleistung ausliest.

und zwar in fronius/inverter/1/TOTAL_ENERGY

Gruß

nkleber78, 2023-06-04:
Bug von Fronius, daher nichts zu ändern

closed 2023-06-04 by nkleber78