iobroker-community-adapters/ioBroker.fronius #206: State "fronius.0.inverter.1.Power_DC_String_2" has no existing object, this might lead to an error in future versions

homecineplexx, 2022-02-15:
v1.1.3

`State "fronius.0.inverter.1.Power_DC_String_2" has no existing object, this might lead to an error in future versions`

homecineplexx, 2022-02-16:
with version v.1.1.6 there are some more:
`

fronius.0 | 2022-02-16 09:19:54.985 | warn | State "fronius.0.inverter.1.UAC_L3" has no existing object, this might lead to an error in future versions
-- | -- | -- | --
fronius.0 | 2022-02-16 09:19:54.983 | warn | State "fronius.0.inverter.1.UAC_L2" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:19:54.982 | warn | State "fronius.0.inverter.1.UAC_L1" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:19:54.982 | warn | State "fronius.0.inverter.1.IAC_L3" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:19:54.981 | warn | State "fronius.0.inverter.1.IAC_L2" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:19:54.939 | warn | State "fronius.0.inverter.1.IAC_L1" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:19:26.662 | warn | State "fronius.0.inverter.1.UAC_L3" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:19:26.662 | warn | State "fronius.0.inverter.1.UAC_L2" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:19:26.661 | warn | State "fronius.0.inverter.1.UAC_L1" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:19:26.660 | warn | State "fronius.0.inverter.1.IAC_L3" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:19:26.658 | warn | State "fronius.0.inverter.1.IAC_L2" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:19:26.657 | warn | State "fronius.0.inverter.1.IAC_L1" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:18:55.991 | warn | State "fronius.0.inverter.1.UAC_L3" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:18:55.990 | warn | State "fronius.0.inverter.1.UAC_L2" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:18:55.990 | warn | State "fronius.0.inverter.1.UAC_L1" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:18:55.990 | warn | State "fronius.0.inverter.1.IAC_L3" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:18:55.989 | warn | State "fronius.0.inverter.1.IAC_L2" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:18:55.988 | warn | State "fronius.0.inverter.1.IAC_L1" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:18:27.681 | warn | State "fronius.0.inverter.1.UAC_L3" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:18:27.681 | warn | State "fronius.0.inverter.1.UAC_L2" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:18:27.680 | warn | State "fronius.0.inverter.1.UAC_L1" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:18:27.679 | warn | State "fronius.0.inverter.1.IAC_L3" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:18:27.679 | warn | State "fronius.0.inverter.1.IAC_L2" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-02-16 09:18:27.678 | warn | State "fronius.0.inverter.1.IAC_L1" has no existing object, this might lead to an error in future versions

`

nkleber78, 2022-02-16:
Are the messages "self healing"? A restart should definitely fix it... Anyhow the target is to fix it mainly in version 2.0

homecineplexx, 2022-02-16:
no there is no self healing and i did  some restarts but isn't better.

so, you plan a version 2.0??

SteveWi-291, 2022-05-19:
Hallo nkleber (norbert), weißt du schon, wann Version 2 erscheinen soll (Stable)? Ich bekomme auch diese Fehler...

<html>
<body>

fronius.0 | 2022-05-19 11:08:25.553 | warn | State "fronius.0.inverter.1.Power_DC_String_2" has no existing object, this might lead to an error in future versions
-- | -- | -- | --
fronius.0 | 2022-05-19 11:08:25.552 | warn | State "fronius.0.inverter.1.Power_DC_String_2" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-05-19 11:08:25.552 | warn | State "fronius.0.inverter.1.Voltage_DC_String_2" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-05-19 11:08:25.551 | warn | State "fronius.0.inverter.1.Power_DC_String_1" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-05-19 11:08:25.551 | warn | State "fronius.0.inverter.1.Voltage_DC_String_1" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-05-19 11:08:25.550 | warn | State "fronius.0.inverter.1.Temperature_Powerstage" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-05-19 11:08:25.549 | warn | State "fronius.0.inverter.1.Current_DC_String_2" has no existing object, this might lead to an error in future versions
fronius.0 | 2022-05-19 11:08:25.548 | warn | State "fronius.0.inverter.1.Current_DC_String_1" has no existing object, this might lead to an error in future versions

</body>
</html>

nkleber78, 2022-05-19:
Leider bekomme ich keinen Support vom Repository maintainer. Somit ist das Thema leider on Hold. Bei mir persönlich läuft die neue Version (habe sie aber noch nicht 2.0 genannt) 
Wer möchte kann diese Version auf eigenes Risiko von meinem Branch installieren. Aber bitte gebt acht bzgl. der neuen Datenstruktur!...
https://github.com/nkleber78/ioBroker.fronius/tarball/DynamicParsing

SteveWi-291, 2022-05-20:
Hab die Version mal drübergebügelt. Er verbindet sich garnicht mehr mit dem Dienst

nkleber78, 2023-06-04:
Probleme sollten alle in der V2.0.0 gelöst sein. Jedenfalls bei mir weder im Test noch im Live system reproduzierbar

closed 2023-06-04 by mcm1957