iobroker-community-adapters/ioBroker.fronius #217: Adapter gibt Fehler aus und stürzt ab: error | adapter.config.inverter.split is not a function

riconr1, 2022-03-17:
Ich erhalte folgende Fehlermeldung in IOBroker:

**<html>
<body>

2022-03-17 16:51:30.222 | warn | Terminated (UNCAUGHT_EXCEPTION): Without reason
-- | -- | --
fronius.0 | 2022-03-17 16:51:30.219 | info | terminating
fronius.0 | 2022-03-17 16:51:29.697 | error | adapter.config.inverter.split is not a function
fronius.0 | 2022-03-17 16:51:29.696 | error | TypeError: adapter.config.inverter.split is not a function at Request._callback (/opt/iobroker/node_modules/iobroker.fronius/main.js:616:53) at Request.self.callback (/opt/iobroker/node_modules/request/request.js:185:22) at Request.emit (events.js:400:28) at Request.<anonymous> (/opt/iobroker/node_modules/request/request.js:1154:10) at Request.emit (events.js:400:28) at IncomingMessage.<anonymous> (/opt/iobroker/node_modules/request/request.js:1076:12) at Object.onceWrapper (events.js:519:28) at IncomingMessage.emit (events.js:412:35) at endReadableNT (internal/streams/readable.js:1334:12) at processTicksAndRejections (internal/process/task_queues.js:82:21)
fronius.0 | 2022-03-17 16:51:29.694 | error | uncaught exception: adapter.config.inverter.split is not a function

</body>
</html> the bug**  
A clear and concise description of what the bug is.

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '....'
4. See error

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Screenshots & Logfiles**  
If applicable, add screenshots and logfiles to help explain your problem.

**Versions:**  
 - Adapter version: <adapter-version>
 - JS-Controller version: <js-controller-version> 
 - Node version: <node-version> 
 - Operating system: <os-name>

**Additional context**  
Add any other context about the problem here.

Kann jemand erkennen, woran das liegen mag?

LG Rico

nkleber78, 2022-03-19:
Welche Version des Adapters hast du? Hast du schon mal in der Adapterconfig auf "checkIP" gedrückt? Ich vermute da ist ggf. was in der config nicht richtig...

riconr1, 2022-03-20:
Ich habe die letzte stable version, die ich IOBroker laden kann: die v1.1.3
Konfigurieren kann man ja lediglich die IP und die ist garantiert richtig. Jetzt wo kein Strom mehr produziert wird. Lässt sich der Adapter starten, ich gucke mir das morgen Früh bei Sonnenschein noch einmal an, wenn beide Inverter Strom liefern.

nkleber78, 2022-03-21:
Bitte verwende die latest (1.1.6) Und es ist schon wichtig dass wenn ein weiterer WR im System ist dass nach dem starten des adapters auch checkIP gedrückt wird. Denn dahinter verbirgt sich mehr als nur ein check... dabei wird z.b. auch die liste and Inverter aktualisiert

masterkey2000, 2022-03-30:
Ich hab das gleiche Problem. Hab nur einen Wechselrichter. Wie kann ich denn auf 1.1.6 updaten. Mir wird in iobroker nur die 1.1.3 angezeigt.

nkleber78, 2022-03-30:
Was steht denn bei euch in der Adapter config drin?
Diese findet ihr in system.adapter.fronius.X. Relevant wäre der native part welcher so ähnlich aussehen sollte...
 "native": {
    "ip": "192.168.1.x",
    "apiversion": "1",
    "baseurl": "/solar_api/v1/",
    "inverter": "1",
    "sensorCard": "1",
    "stringControl": "",
    "meter": "0",
    "storage": "0",
    "poll": "30",
    "requestType": "http://",
    "pollarchive": "150"
  },

flyinghuman, 2022-06-21:
ich hatte den gleichen Fehler. Man muss noch den Ping-Adapter installieren, danach den Fronius Adapter wieder entfernen und nochmal installieren. Ich habe auch vor dem neuinstallieren des Fronius-Adapters noch diesen Fix angewendet: https://github.com/iobroker-community-adapters/ioBroker.fronius/issues/126 
Scheint also nur bei Neuinstallationen ohne PING Adapter aufzutreten.

nkleber78, 2023-06-04:
Problem mit ping adapter ist gelöst in V2.0.0 da PING nicht mehr verwendet wird.

closed 2023-06-04 by nkleber78