iobroker-community-adapters/ioBroker.fronius #242: Nach Austausch von neuem Smartmeter keine Werte mehr im Objektbaum "meter"

nikmaier, 2022-07-20:
Hallo,
gestern ist mein altes Fronius Smartmeter gegen ein neues ausgetauscht worden, beim alten hatte ich immer diese Datenpunkte:

[image]

Das neue Smartmeter wird anscheinend nicht erkannt, hier habe ich unter dem Objektbaum "meter" keine Daten:

[image]

Gibt es hier eine Lösung?

Danke im Voraus.

nkleber78, 2023-06-04:
@nikmaier Existiert das Problem noch? Falls ja bitte mehr details um dies zu reproduzieren. Bitte auch den API output anhängen. Entweder ist die Adresse anders oder es kommen generell keine Daten...

nikmaier, 2023-06-04:
Hallo, ja das ist immer noch nach wie vor vorhanden, kommen keinerlei Daten vom neuen Smartmeter.
Was genau brauchst du? Was meinst du mit API output?

nkleber78, 2023-06-04:
Adapter version?
Api output von http://192.168.0.1/solar_api/v1/GetMeterRealtimeData.cgi?Scope=Device&DeviceId=0
Object data von diesem Objekt system.adapter.fronius.0 wobei die ID am schluß mit deiner Instanz zusammenpassen muss

Danke

nikmaier, 2023-06-04:
Adapter: 1.1.6:
[image]

Aber wo finde ich die richtige Device-ID?

nkleber78, 2023-06-04:
http://192.168.0.1/solar_api/v1/GetActiveDeviceInfo.cgi?DeviceClass=System Da findest du deine device informationen. Bitte schick mir auch davon den response

nikmaier, 2023-06-04:
Also hier der erste response, hab mal die S/N weg:

`{
   "Body" : {
      "Data" : {
         "Inverter" : {
            "1" : {
               "DT" : 1,
               "Serial" : "3310xx"
            }
         },
         "Meter" : {
            "1" : {
               "DT" : -1,
               "Serial" : "271xx"
            }
         },
         "Ohmpilot" : {},
         "Storage" : {
            "0" : {
               "DT" : -1,
               "Serial" : "P030T020Z22xxx     "
            }
         }
      }
   },
   "Head" : {
      "RequestArguments" : {
         "DeviceClass" : "System"
      },
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2023-06-04T10:11:21+00:00"
   }
}`

Und hier dann der zweite, aber hier kommt nichts, ich muss doch als ID die Seriennummer vom Meter nehmen oder:

`{
   "Body" : {
      "Data" : {}
   },
   "Head" : {
      "RequestArguments" : {
         "DeviceId" : "2717xx",
         "Scope" : "Device"
      },
      "Status" : {
         "Code" : 12,
         "Reason" : "Meter '2717xx' does not exist.",
         "UserMessage" : ""
      },
      "Timestamp" : "2023-06-04T10:12:44+00:00"
   }
}`

nkleber78, 2023-06-04:
Für den 2. response musst du hier die ID  des  meters nehmen und nicht die Sn. D.h. in deinem fall 1. bitte auch noch die Adapter config. Ich vermute dass in deiner adapter config die ID 0 drin steht und nicht 1... Offensichtlich hat sich hier die ID auf der API geändert mit dem neuen Smartmeter. Hast du schon mal checkIP gedrückt in der Instanz Konfiguration? 
[image]

nikmaier, 2023-06-04:
Ah wahnsinn, ich hab mal Check Ip gedrückt und dann nochmal gespeichert.
Jetzt hat er mir tatsächlich unter meter einen neuen Ordner mit der Nummer 1, also quasi die ID1 gemacht. 
Und dort ist jetzt alles drin.
Es kann so einfach sein *lol*

Vielen lieben Dank für deine Hilfe und diesen Tip!

nkleber78, 2023-06-04:
Sehr gut. 
Problem behoben, wird dann in der V2.0.0 dann auch zyklisch gemacht (Prüfung ob neue Geräte am Bus). Daher wird das ticket geschlossen

closed 2023-06-04 by nkleber78