iobroker-community-adapters/ioBroker.fronius #255: Bool-Werte als common-type boolean speichern

Steff42, 2022-10-19:
Hallo,

aktuell werden bool-Werte als common-type mixed gespeichert. Wäre es möglich diese als common-type boolean zu speichern?
Des betrifft aktuell die z.B. die beiden Objekte:

`fronius.0.powerflow.BackupMode`
`fronius.0.powerflow.BatteryStandby`

Gruß Steffen

nkleber78, 2022-10-20:
Das Thema ist, dass in vielen Fällen der Typ nicht erkannt werden kann und manchmal ist es sogar so dass sich die "Werte" ändern. In diesem fall sind die Parameter nicht gepflegt und daher werden sie default als mixed angelegt all fallback option. Es sollte aber problemlos sein wenn du den typ nachträglich auf Bool änderst...

mcm1957, 2023-05-31:
Mein Input:
States sollten immer einen passenden Datentyp und soweit wie möglich eine passende role haben.
Dieser issue sollte noch mal genauer analysiert werden.

Steff42, 2023-07-16:
Passt in Version 2.0.2

Gruß Steffen

closed 2023-07-17 by nkleber78