iobroker-community-adapters/ioBroker.fronius #256: Keine Daten auslesbar, trotz erkannter IP

Ulle80, 2022-10-25:
Hi zusammen, seit geraumer Zeit habe ich das Problem, dass keine Daten mehr vom Fronius Symo WR zum iobroker übertragen werden. Die IP wird erkannt, eine Verbindung zum Gerät erfolgt nicht. Habe über den WR traceroute und Ping gecheckt, das war auch okay. so langsam gehen mir die Ideen aus, was ich noch checken kann. Firmware des WR ist die 3.24.2-1, Adapter v1.1.6.
Ich habe mittlerweile eine Feste IP, DHCP IP, LAN und WLAN getestet, ohne Erfolg.

[image]
[image]

VG Ulle

mayli01, 2022-10-29:
Hallo!
Leider habe ich das gleiche Problem seit Anfang an. 
Wechselrichter: Symo GEN24 10 plus
Software Version: 1.22.2-1
Wie @Ulle80  beschrieben auch schon alles versucht, ohne Erfolg.
Lg mayli

nkleber78, 2022-10-29:
Was sagt der Debug mode dazu? Habt ihr sichergestellt dass das Protokoll passt? (http vs https). Wenn der Ping geht dass ist das recht warscheinlich. Weiters auch mal schauen ob die API auch entsprechend antwortet. http://WechselrichterIP/solar_api/GetAPIVersion.cgi -> 
{
   "APIVersion" : 1,
   "BaseURL" : "/solar_api/v1/",
   "CompatibilityRange" : "1.7-7"
}

mayli01, 2022-10-29:
API antwortet mit:
{
   "APIVersion" : 1,
   "BaseURL" : "/solar_api/v1/",
   "CompatibilityRange" : "1.7-7"
}

Log:

2022-10-29 10:09:10.383 - info: host.IoBroker "system.adapter.fronius.0" disabled
2022-10-29 10:09:10.384 - info: host.IoBroker stopInstance system.adapter.fronius.0 (force=false, process=true)
2022-10-29 10:09:10.385 - info: fronius.0 (59081) Got terminate signal TERMINATE_YOURSELF
2022-10-29 10:09:10.449 - info: host.IoBroker stopInstance system.adapter.fronius.0 send kill signal
2022-10-29 10:09:10.886 - info: fronius.0 (59081) terminating
2022-10-29 10:09:10.887 - info: fronius.0 (59081) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
2022-10-29 10:09:11.437 - info: host.IoBroker instance system.adapter.fronius.0 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
2022-10-29 10:09:13.134 - info: host.IoBroker "system.adapter.fronius.0" enabled
2022-10-29 10:09:13.150 - info: host.IoBroker instance system.adapter.fronius.0 started with pid 59105
2022-10-29 10:09:13.374 - debug: fronius.0 (59105) Redis Objects: Use Redis connection: 127.0.0.1:9001
2022-10-29 10:09:13.401 - debug: fronius.0 (59105) Objects client ready ... initialize now
2022-10-29 10:09:13.401 - debug: fronius.0 (59105) Objects create System PubSub Client
2022-10-29 10:09:13.401 - debug: fronius.0 (59105) Objects create User PubSub Client
2022-10-29 10:09:13.476 - debug: fronius.0 (59105) Objects client initialize lua scripts
2022-10-29 10:09:13.477 - debug: fronius.0 (59105) Objects connected to redis: 127.0.0.1:9001
2022-10-29 10:09:13.503 - debug: fronius.0 (59105) Redis States: Use Redis connection: 127.0.0.1:9000
2022-10-29 10:09:13.521 - debug: fronius.0 (59105) States create System PubSub Client
2022-10-29 10:09:13.521 - debug: fronius.0 (59105) States create User PubSub Client
2022-10-29 10:09:13.594 - debug: fronius.0 (59105) States connected to redis: 127.0.0.1:9000
2022-10-29 10:09:13.734 - info: fronius.0 (59105) starting. Version 1.1.3 in /opt/iobroker/node_modules/iobroker.fronius, node: v19.0.0, js-controller: 4.0.23
2022-10-29 10:09:13.745 - debug: fronius.0 (59105) System command: /bin/ping -n -w2 -c1 192.168.1.71
2022-10-29 10:09:13.749 - debug: fronius.0 (59105) System command: /bin/ping -n -w2 -c1 192.168.1.71

nkleber78, 2022-10-29:
Und was sagt der Requesttype? Sollte dann http bei dir sein
[image]

mayli01, 2022-10-29:
},
  "native": {
    "ip": "192.168.1.71",
    "apiversion": "1",
    "baseurl": "/solar_api/v1/",
    "inverter": "1",
    "sensorCard": "",
    "stringControl": "",
    "meter": "",
    "storage": "",
    "poll": "30",
    "requestType": "http://",
    "pollarchive": "150"
  },

Hab auch schon "https" probiert...

Ulle80, 2022-10-29:
> Was sagt der Debug mode dazu? Habt ihr sichergestellt dass das Protokoll passt? (http vs https). Wenn der Ping geht dass ist das recht warscheinlich. Weiters auch mal schauen ob die API auch [...]

So sieht es bei mir aus: 
{
   "APIVersion" : 1,
   "BaseURL" : "/solar_api/v1/",
   "CompatibilityRange" : "1.8-1"
}

Wo komme ich in die Einstellungen des Request-Types?

Ulle80, 2022-10-29:
Hatte über einen anderen Beitrag noch die Abfrage rausgefunden:

[http://192.168.178.10/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DeviceId=1&DataCollection=CommonInverterData]

Liefert auch Werte: 

> {
   "Body" : {
      "Data" : {
         "DAY_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 2467.0999999999999
         },
         "DeviceStatus" : {
            "ErrorCode" : 0,
            "LEDColor" : 3,
            "LEDState" : 0,
            "MgmtTimerRemainingTime" : -1,
            "StateToReset" : false,
            "StatusCode" : 2
         },
         "TOTAL_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 9728420
         },
         "YEAR_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 2504795.25
         }
      }
   },
   "Head" : {
      "RequestArguments" : {
         "DataCollection" : "CommonInverterData",
         "DeviceClass" : "Inverter",
         "DeviceId" : "1",
         "Scope" : "Device"
      },
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2022-10-29T17:42:53+02:00"
   }
}

nkleber78, 2022-10-30:
Schon mal versucht eine neue Instanz anzulegen? Habe derzeit keine Idee wo da problem liegen könnte. Gerne auch mal von meinem Repository den Branch "DynamicParsing" über "install from custom URL" installieren. Die Objektstruktur ist dabei geändert, läuft bei mir seit Monaten sehr stabil...
https://github.com/nkleber78/ioBroker.fronius/tarball/DynamicParsing

mayli01, 2022-10-30:
Super danke!
@nkleber78 Deine hat problemlos funktioniert.
Installiert, IP eingetragen und gestartet. Sofort waren die Datenpunkt da!
Kein Ahnung woran es bei der offiziellen Version liegt...

Ulle80, 2022-10-30:
> Schon mal versucht eine neue Instanz anzulegen? Habe derzeit keine Idee wo da problem liegen könnte. Gerne auch mal von meinem Repository den Branch "DynamicParsing" über "install from custom URL" [...]

@nkleber78: DANKE! Mit dem "DynamicParsing" funktionierte es auf Schlag! Andere Instanz nicht.

closed 2022-10-30 by Ulle80