iobroker-community-adapters/ioBroker.fronius #285: Zweiter Smart Meter nicht sichtbar

dako0815, 2023-03-05:
Hallo,
ich habe an meinem Fronius Symo Hybrid einen zweiten Smart Meter angeschlossen der die Erzeugung eines weiteren Wechselrichters misst. Diesen sehe ich auch im Solarweb und die Erzeugung wird mir auch korrekt für beide angezeigt:
[image]
[image]

Über den Adapter bekomme ich aber nur den ersten SmartMeter übermittelt:
[image]

Warum wird hier der zweite Fronius Smart Meter nicht mit angezeigt?

Grüße,
Daniel

nkleber78, 2023-06-04:
@dako0815 Welche Version hast du installiert. Das sollte nun in der V2.0.0 gefixt sein. Wenn du ein Testsystem hast, dann gerne mal die Version 2.0.0 Alpha von GIT installieren und verifizieren

nkleber78, 2023-06-28:
It is working on V2.0.1 on my test environment

closed 2023-06-28 by nkleber78