iobroker-community-adapters/ioBroker.fronius #297: Tracking v2.0.0 / merging nKleber enhencments

mcm1957, 2023-05-31:
I will use this issue solely to track an upcomin 2.0.0 release of thgis adapter.
Thanks to @nkleber78 who has spent much effort (and time) working an this adapter.

Please **do not** add problemreports or feature requests to this issue but comment on existing or open new issues for this.

**ATTENTION:** 
This upgrade contains many changes added during time. Strong upward compatibility cannot be guaranted. So do NOT try to install the github version at a production system and ensure to have a working backup for all installations

Tasklist:
- [ ] update ioBroker development tools (McM1957)
- [ ] cleanup testing environment (McM1957)
- [ ] merge PR from nKleber78 (McM1957)
- [ ] testinstall (github version) (McM1957)
- [ ] update releasenotes (McM1957)
- [ ] add installation / upgrade warning (McM1957)
- [ ] functiona tests (by @nKleber78) 
- [ ] release 2.0.0-alpha

mcm1957, 2023-06-01:
@nkleber78 

Schönen Abend
Ich hab jetzt 
- deinen PR gemerged
- request package entfernt und axios und ne hinzugefügt
- autofixable eslint warnungen gefixed
- die Revisionliste im Readme angefangen
- die Installationsmessage in io-package.json eingefügt
- eine 2.0.0-alpha.3 erstellt

Achtung: derzeit werden die Releases noch nicht nach npm deployed. Das möchte ich erst machen, wenn du kurz getestet hast das der Code zumindest halbwegs funktioniert :-)

Bitte - sobald du Zeit hast - installier mal die 2.0.0-alpha.3 (= head) von Github (https://github.com/iobroker-community-adapters/ioBroker.fronius) und schau was (noch) geht. Ich hab den Adapter zwar installiert - mangels Fronius kann ich aber nur testen dass sich die COnfig öffnet :-).

eslint bemängelt noch einige Dinge. Ich hab dir issues gegeben. Es scheint mir sinnvoll, dass du dir die ansiehst - ich möchet ungern blind in den dir bekannten Code eingreifen. Wenn es dir zu viel ist, kann ich aber zumindest die "eh klar" Dinge auch korrigieren (z.B. const statt var, ...).

Ich lass den Code jetzt mal in Ruhe und wart auf eine Rückmeldung v dir - wann imemr du dazu kommst.
Falls es einen neuen PR gibt kann ich den kurzfristig reinmergen - außer Samstag, da bin ich total offline.

Schöne Grüße
Und Danke für deine Arbeit !

nkleber78, 2023-06-01:
Danke dir, habe die Änderungen soweit mal gemacht. Bzgl. lokalem eslint komme ich leider nicht klar. Es gibt auch noch das problem dass das json file im test ordner dazu führt dass die Tests nicht funktionieren. Wie soll ich das am besten lösen? Im aktuellen PR sollte soweit jetzt eigentlich alles passen...

mcm1957, 2023-06-02:
@nkleber78 
2.0.0-alpha.4 erstellt. Bereit zum Test wenn du Zeit hast.
Release gibts aber nur auf Github zur Zeit. npm deploy folgt sobald ein erster Test der GH ok ist.

Sorry für das Delay - mein Monatsbackup (Totalbackup) hat meinen PC durch Snapshootting ein wenig abgeschossen und zu Hängern geführt...

closed 2023-06-28 by mcm1957