iobroker-community-adapters/ioBroker.fronius #313: "uncaught exception: Cannot convert undefined or null to object" at fronius restart

mcm1957, 2023-06-07:
Adapter crashes at restart of fronius system:

```
2023-06-06 08:55:12.079 - error: fronius.0 (26028) uncaught exception: Cannot convert undefined or null to object
2023-06-06 08:55:12.079 - error: fronius.0 (26028) TypeError: Cannot convert undefined or null to object
  | at hasOwnProperty ()
  | at createStates (/opt/iobroker/node_modules/iobroker.fronius/lib/devObjects.js:474:41)
  | at Timeout._onTimeout (/opt/iobroker/node_modules/iobroker.fronius/lib/devObjects.js:127:17)
  | at listOnTimeout (node:internal/timers:561:11)
  | at processTimers (node:internal/timers:502:7)
```

An exception stating problems would be OK, a crash must be avoided.

Detail log see comments at forum:

- https://forum.iobroker.net/topic/65948/test-adapter-fronius-2-0-x/12
- https://forum.iobroker.net/topic/65948/test-adapter-fronius-2-0-x/17

nkleber78, 2023-06-07:
There i need to think how to handle cases where somewhere in the middle of the process the object is not delivering the expected values.

nkleber78, 2023-06-10:
Part of Release 2.0.2

closed 2023-06-10 by nkleber78