iobroker-community-adapters/ioBroker.fronius #314: Datenpunkte site P_Grid P_Load und P_PV

woehrer12, 2023-06-07:
Diese Datenpunkte sollten eigentlich anzeigen:

P_Grid: Leistung von bzw ins Netz
Dieser Stimmt

P_Load: Leistung die verbraucht wird P_PV + P_Grid (Hier steht Plus da P_Grid negativ ist bei Netzeinspeisung)
Zeigt invertiert P_Grid an

P_PV: Leistung aller PV Anlagen
Zeigt nur die Leistung vom "Master"

nkleber78, 2023-06-07:
@woehrer12 Was hast du für ein Setup bei dir? Bei meinem Setup 4*Symo 20kW, 1* Master, 3* slave zeigt P_PV z.b. aktuell 50028W an. Das stimmt bei mir mit der Leistung
Bzgl. P_Load: Das scheint bei mir auch plausibel, ist aber negativ wenn Netzbezug, auch wenn das bei mir derzeit so ist dass ich ein nicht gemessenen Erzeuger auf der Lastseite habe. Daher ist P_Load bei mir derzeit auch positiv: 
[image]
Resultat: P_Load = -(P_PV+P_Grid) Ich denke das müsste auch passen. Es ist auch so dass keine Berechnungen gemacht werden, d.h. die Daten werden 1:1 von der API auf die Objekte übertragen. Kannst du das bei dir nochmals verifizieren?

nkleber78, 2023-06-28:
@woehrer12 ist da noch was offen? Falls nicht würde ich gerne das Ticket schließen

closed 2023-06-28 by nkleber78