iobroker-community-adapters/ioBroker.fronius #315: fronius.0.site.P_PV geht nicht auf 0W

Negalein, 2023-06-07:
mir ist soeben aufgefallen, dass `fronius.0.site.P_PV` am Abend nicht auf 0 geht.
Steht bei mir gerade um 23 Uhr auf 38 W.
Der Wert ist seit 20:59 stehen geblieben.

Vers.: 2.0.1

Mit Vers. 1.x funktionierte es.

Im WebIF wird aktuell auch keine Erzeugung angezeigt.

[image]

```
fronius.0
2023-06-07 23:16:40.577	debug	API Objekt rel_SelfConsumption is null, no object created!

fronius.0
2023-06-07 23:16:40.577	debug	API Objekt P_PV is null, no object created!

fronius.0
2023-06-07 23:16:40.577	debug	API Objekt P_Akku is null, no object created!

fronius.0
2023-06-07 23:16:39.576	debug	Property rel_SelfConsumption is null. No object created!

fronius.0
2023-06-07 23:16:39.576	debug	Property P_PV is null. No object created!

fronius.0
2023-06-07 23:16:39.576	debug	Property P_Akku is null. No object created!

fronius.0
2023-06-07 23:16:39.575	debug	Object {"DT":232,"E_Day":59030,"E_Total":72843704,"E_Year":3624414,"P":0} does not have a property SOC, therefore no State was created

fronius.0
2023-06-07 23:16:39.562	debug	Object {"Current_AC_Phase_1":1.063,"Current_AC_Phase_2":2.957,"Current_AC_Phase_3":2.804,"Details":{"Manufacturer":"Fronius","Model":"Smart Meter 63A","Serial":"17360313"},"Enable":1,"EnergyReactive_VArAC_Sum_Consumed":16072070,"EnergyReactive_VArAC_Sum_Produced":8029900,"EnergyReal_WAC_Minus_Absolute":3683573,"EnergyReal_WAC_Plus_Absolute":9180975,"EnergyReal_WAC_Sum_Consumed":9180975,"EnergyReal_WAC_Sum_Produced":3683573,"Frequency_Phase_Average":50,"Meter_Location_Current":0,"PowerApparent_S_Phase_1":247.78529999999998,"PowerApparent_S_Phase_2":689.2766999999999,"PowerApparent_S_Phase_3":654.4535999999999,"PowerApparent_S_Sum":1479,"PowerFactor_Phase_1":-0.99,"PowerFactor_Phase_2":0.99,"PowerFactor_Phase_3":-0.97,"PowerFactor_Sum":-0.99,"PowerReactive_Q_Phase_1":10.46,"PowerReactive_Q_Phase_2":-35.32,"PowerReactive_Q_Phase_3":126.49,"PowerReactive_Q_Sum":101.63,"PowerReal_P_Phase_1":209.73,"PowerReal_P_Phase_2":670.63,"PowerReal_P_Phase_3":595.94,"PowerReal_P_Sum":1476.3,"TimeStamp":1686172598,"Visible":1,"Voltage_AC_PhaseToPhase_12":403.7,"Voltage_AC_PhaseToPhase_23":404,"Voltage_AC_PhaseToPhase_31":404,"Voltage_AC_Phase_1":233.1,"Voltage_AC_Phase_2":233.1,"Voltage_AC_Phase_3":233.4} does not have a property Current_AC_Sum, therefore no State was created

fronius.0
2023-06-07 23:16:39.505	debug	Object {"DAY_PMAX":{"Unit":"W","Value":7264},"DAY_UACMAX":{"Unit":"V","Value":246},"DAY_UDCMAX":{"Unit":"V","Value":627.7},"TOTAL_PMAX":{"Unit":"W","Value":9888},"TOTAL_UACMAX":{"Unit":"V","Value":248.4},"TOTAL_UDCMAX":{"Unit":"V","Value":760.7},"YEAR_PMAX":{"Unit":"W","Value":9836},"YEAR_UACMAX":{"Unit":"V","Value":248.4},"YEAR_UDCMAX":{"Unit":"V","Value":755.6}} does not have a property T_AMBIENT, therefore no State was created

fronius.0
2023-06-07 23:16:39.505	debug	Object {"DAY_PMAX":{"Unit":"W","Value":7264},"DAY_UACMAX":{"Unit":"V","Value":246},"DAY_UDCMAX":{"Unit":"V","Value":627.7},"TOTAL_PMAX":{"Unit":"W","Value":9888},"TOTAL_UACMAX":{"Unit":"V","Value":248.4},"TOTAL_UDCMAX":{"Unit":"V","Value":760.7},"YEAR_PMAX":{"Unit":"W","Value":9836},"YEAR_UACMAX":{"Unit":"V","Value":248.4},"YEAR_UDCMAX":{"Unit":"V","Value":755.6}} does not have a property TOTAL_ENERGY, therefore no State was created

fronius.0
2023-06-07 23:16:39.505	debug	Object {"DAY_PMAX":{"Unit":"W","Value":7264},"DAY_UACMAX":{"Unit":"V","Value":246},"DAY_UDCMAX":{"Unit":"V","Value":627.7},"TOTAL_PMAX":{"Unit":"W","Value":9888},"TOTAL_UACMAX":{"Unit":"V","Value":248.4},"TOTAL_UDCMAX":{"Unit":"V","Value":760.7},"YEAR_PMAX":{"Unit":"W","Value":9836},"YEAR_UACMAX":{"Unit":"V","Value":248.4},"YEAR_UDCMAX":{"Unit":"V","Value":755.6}} does not have a property YEAR_ENERGY, therefore no State was created

fronius.0
2023-06-07 23:16:39.505	debug	Object {"DAY_PMAX":{"Unit":"W","Value":7264},"DAY_UACMAX":{"Unit":"V","Value":246},"DAY_UDCMAX":{"Unit":"V","Value":627.7},"TOTAL_PMAX":{"Unit":"W","Value":9888},"TOTAL_UACMAX":{"Unit":"V","Value":248.4},"TOTAL_UDCMAX":{"Unit":"V","Value":760.7},"YEAR_PMAX":{"Unit":"W","Value":9836},"YEAR_UACMAX":{"Unit":"V","Value":248.4},"YEAR_UDCMAX":{"Unit":"V","Value":755.6}} does not have a property DAY_ENERGY, therefore no State was created

fronius.0
2023-06-07 23:16:39.504	debug	Object {"DAY_PMAX":{"Unit":"W","Value":7264},"DAY_UACMAX":{"Unit":"V","Value":246},"DAY_UDCMAX":{"Unit":"V","Value":627.7},"TOTAL_PMAX":{"Unit":"W","Value":9888},"TOTAL_UACMAX":{"Unit":"V","Value":248.4},"TOTAL_UDCMAX":{"Unit":"V","Value":760.7},"YEAR_PMAX":{"Unit":"W","Value":9836},"YEAR_UACMAX":{"Unit":"V","Value":248.4},"YEAR_UDCMAX":{"Unit":"V","Value":755.6}} does not have a property IDC_2, therefore no State was created

fronius.0
2023-06-07 23:16:39.504	debug	Object {"DAY_PMAX":{"Unit":"W","Value":7264},"DAY_UACMAX":{"Unit":"V","Value":246},"DAY_UDCMAX":{"Unit":"V","Value":627.7},"TOTAL_PMAX":{"Unit":"W","Value":9888},"TOTAL_UACMAX":{"Unit":"V","Value":248.4},"TOTAL_UDCMAX":{"Unit":"V","Value":760.7},"YEAR_PMAX":{"Unit":"W","Value":9836},"YEAR_UACMAX":{"Unit":"V","Value":248.4},"YEAR_UDCMAX":{"Unit":"V","Value":755.6}} does not have a property UDC_2, therefore no State was created

fronius.0
2023-06-07 23:16:39.504	debug	Object {"DAY_PMAX":{"Unit":"W","Value":7264},"DAY_UACMAX":{"Unit":"V","Value":246},"DAY_UDCMAX":{"Unit":"V","Value":627.7},"TOTAL_PMAX":{"Unit":"W","Value":9888},"TOTAL_UACMAX":{"Unit":"V","Value":248.4},"TOTAL_UDCMAX":{"Unit":"V","Value":760.7},"YEAR_PMAX":{"Unit":"W","Value":9836},"YEAR_UACMAX":{"Unit":"V","Value":248.4},"YEAR_UDCMAX":{"Unit":"V","Value":755.6}} does not have a property IDC, therefore no State was created

fronius.0
2023-06-07 23:16:39.504	debug	Object {"DAY_PMAX":{"Unit":"W","Value":7264},"DAY_UACMAX":{"Unit":"V","Value":246},"DAY_UDCMAX":{"Unit":"V","Value":627.7},"TOTAL_PMAX":{"Unit":"W","Value":9888},"TOTAL_UACMAX":{"Unit":"V","Value":248.4},"TOTAL_UDCMAX":{"Unit":"V","Value":760.7},"YEAR_PMAX":{"Unit":"W","Value":9836},"YEAR_UACMAX":{"Unit":"V","Value":248.4},"YEAR_UDCMAX":{"Unit":"V","Value":755.6}} does not have a property UDC, therefore no State was created

fronius.0
2023-06-07 23:16:39.504	debug	Object {"DAY_PMAX":{"Unit":"W","Value":7264},"DAY_UACMAX":{"Unit":"V","Value":246},"DAY_UDCMAX":{"Unit":"V","Value":627.7},"TOTAL_PMAX":{"Unit":"W","Value":9888},"TOTAL_UACMAX":{"Unit":"V","Value":248.4},"TOTAL_UDCMAX":{"Unit":"V","Value":760.7},"YEAR_PMAX":{"Unit":"W","Value":9836},"YEAR_UACMAX":{"Unit":"V","Value":248.4},"YEAR_UDCMAX":{"Unit":"V","Value":755.6}} does not have a property IAC_L3, therefore no State was created

fronius.0
2023-06-07 23:16:39.504	debug	Object {"DAY_PMAX":{"Unit":"W","Value":7264},"DAY_UACMAX":{"Unit":"V","Value":246},"DAY_UDCMAX":{"Unit":"V","Value":627.7},"TOTAL_PMAX":{"Unit":"W","Value":9888},"TOTAL_UACMAX":{"Unit":"V","Value":248.4},"TOTAL_UDCMAX":{"Unit":"V","Value":760.7},"YEAR_PMAX":{"Unit":"W","Value":9836},"YEAR_UACMAX":{"Unit":"V","Value":248.4},"YEAR_UDCMAX":{"Unit":"V","Value":755.6}} does not have a property UAC_L3, therefore no State was created

fronius.0
2023-06-07 23:16:39.504	debug	Object {"DAY_PMAX":{"Unit":"W","Value":7264},"DAY_UACMAX":{"Unit":"V","Value":246},"DAY_UDCMAX":{"Unit":"V","Value":627.7},"TOTAL_PMAX":{"Unit":"W","Value":9888},"TOTAL_UACMAX":{"Unit":"V","Value":248.4},"TOTAL_UDCMAX":{"Unit":"V","Value":760.7},"YEAR_PMAX":{"Unit":"W","Value":9836},"YEAR_UACMAX":{"Unit":"V","Value":248.4},"YEAR_UDCMAX":{"Unit":"V","Value":755.6}} does not have a property IAC_L2, therefore no State was created

[... 149 lines cut ...]

fronius.0
2023-06-07 23:16:37.192	debug	Objects client initialize lua scripts

fronius.0
2023-06-07 23:16:37.060	debug	Objects create User PubSub Client

fronius.0
2023-06-07 23:16:37.059	debug	Objects create System PubSub Client

fronius.0
2023-06-07 23:16:37.059	debug	Objects client ready ... initialize now

fronius.0
2023-06-07 23:16:37.002	debug	Redis Objects: Use Redis connection: 0.0.0.0:9001

fronius.0
2023-06-07 23:16:34.063	info	Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason

fronius.0
2023-06-07 23:16:34.062	info	terminating

fronius.0
2023-06-07 23:16:33.561	info	Got terminate signal TERMINATE_YOURSELF
````

nkleber78, 2023-06-08:
@Negalein Da ich die Objekte nun alle dynamisch parse und erstelle weiß ich jetzt nie genau was alles vorhanden ist bzw. war. 

Was hälst du von der Idee die quality dafür zu verwenden und die quality lt. folgender Liste auf 0x42 zu setzen wenn die letzte aktualisierung des Datenpoints mehr als 2*aktualisierungsinterval ist. Z.b. wenn der Adapter alle 60s die daten abholt, würde nach 2min q auf 0x42 gesetzt. Das wäre dann ein generischer Ansatz. Einfach ohne zu wissen was es genau für ein Wert ist diesen auf 0 zu setzen ist glaub ich auch nicht so gut, denn dann hast du immer wieder im den Daten/Grafiken diese drops drin die eigentlich nicht existieren, sondern es fehlt einfach der wert...

q: Qualität – Nummer mit folgenden Zuständen:

0x00 - 00000000 - good (can be undefined or null)
0x01 - 00000001 - general bad, general problem
0x02 - 00000010 - no connection problem

0x10 - 00010000 - substitute value from controller
0x20 - 00100000 - substitute initial value
0x40 - 01000000 - substitute value from device or instance
0x80 - 10000000 - substitute value from sensor

0x11 - 01000001 - general problem by instance
0x41 - 01000001 - general problem by device
0x81 - 10000001 - general problem by sensor

0x12 - 00010010 - instance not connected
0x42 - 01000010 - device not connected
0x82 - 10000010 - sensor not connected

0x44 - 01000100 - device reports error
0x84 - 10000100 - sensor reports error

mcm1957, 2023-06-08:
Ich würde - sorry wenn ich das sage - die asuschließlich dynamische Bearbeitung der Datenpunkte in Frage stellen. Damit verlierst du die Möglichkeit sinnvolle ROLE Werte beim Objekt zu setzen. Es würde m.E. sinnvoll sein sehr wohl eine Liste der bekannten Datenpunkte im Adaptercode zu verwalten.

Liefert das Gerät neue so können diese natürlich  neu mit Role value angelegt werden. Der Umstand sollte dann gelogged werden damit man das in einer späteren Version erweitern kann.

Und für Datenpunkte die zwar bekannt sind aber nicht geliefert wurden braucht man aiuch keinen State anlegen - das wär schon OK.

ABER zur PV Frage:
Liefert der Fronius wirklich keinen 0 Wert sondern einfach "nichts" mehr? Kann ich mir irgendwie nicht vorstellen. 
Wie sehen denn die Kommunikationspakete / Rest Daten aus? Steht da kein PV=0, PV='', ...?
Gibts dazu was in der API Spec ?

State ist hie rnur zweite Wahl - weil der User erwartet dass er zu jedem Zeitpunkt eine PV Leistung lesen kann.

nkleber78, 2023-06-08:
@mcm1957 Wir haben schon einige datenpunkte die wir aktiv setzen. Das ist auch notwendig da nicht bei allen Werten über die API auch Units dabei sind. Diese Objekte werden nur angelegt wenn sie tatsächlich über die API geliefert werden. 
Bzgl. der Werte ist es wirklich so dass die API hier keine Werte mehr liefert. Das macht es leider auch nicht gerade einfacher :(

Im Debug / silly mode die ganzen Objekte zu liefern könnte man machen, führt aber zu extrem viel Log was ggf. auch unübersichtlich werden könnte. Aber du hast recht man könnte dann prüfen was tatsächlich geliefert wird

Negalein, 2023-06-08:
@nkleber78 
> Was hälst du von der Idee die quality dafür zu verwenden und die quality lt. folgender Liste auf 0x42 zu setzen wenn die letzte aktualisierung des Datenpoints mehr als 2*aktualisierungsinterval ist.

Hallo

Sorry, das versteh ich nicht. Hängt das mit meinem "0 Wert" zusammen?

@mcm1957 
> Liefert der Fronius wirklich keinen 0 Wert sondern einfach "nichts" mehr? Kann ich mir irgendwie nicht vorstellen.
> Wie sehen denn die Kommunikationspakete / Rest Daten aus? Steht da kein PV=0, PV='', ...?

Mit der letzten 1.x Version funktionierte es immer.
Sobald nichts mehr erzeugt wurde, stand im Wert eine 0.

Ich glaube, dass es dieses Problem aber auch mal in einer früheren V 1.x gab.

nkleber78, 2023-06-08:
@Negalein ja leider liefert die API den Wert nicht mehr wenn das Gerät den PV Betrieb einstellt. Dazu wurde im alten Code in einer der letzten Versionen ein spezieller Hack implementiert der den Wert, wenn nicht über die API geliefert, auf 0 gesetzt hat. Daher die Idee mit der quality…

Negalein, 2023-06-08:
@nkleber78 ah, ok. Jetzt verstehe ich.
Das könnte ich versuchen.

Aber wo stell ich das ein? Im Adapter gibts dort nichts.

nkleber78, 2023-06-08:
Das müsste ich implementieren. Dann könnte der consumer auf die Qualität, q beim State, filtern.  Ev. Könnte die Logik ja sogar kombiniert sein: wenn wert nicht verfügbar dann wird quality gesetzt, wenn das über einen definierten Zeitraum so ist, z.b. 5x Abfrageintervall dann auf 0 setzen. Oder koppeln an den Sonnenuntergang? Ich denke wir brauchen da eine klare Idee wie das sein sollte…

Negalein, 2023-06-08:
> Ev. Könnte die Logik ja sogar kombiniert sein: wenn wert nicht verfügbar dann wird quality gesetzt, wenn das über einen definierten Zeitraum so ist

hört sich gut an

> Oder koppeln an den Sonnenuntergang

Finde obiges besser

mcm1957, 2023-06-08:
> @Negalein ja leider liefert die API den Wert nicht mehr wenn das Gerät den PV Betrieb einstellt. Dazu wurde im alten Code in einer der letzten Versionen ein spezieller Hack implementiert der den Wert, wenn nicht über die API geliefert, auf 0 gesetzt hat. Daher die Idee mit der quality…

Der "Hack" erscheint mir sehr sinnvoll.
Wenn ich eine Abfrage stelle wo ich erwarte dass ein bestimmter Wert kommt und die Api liefert den nicht, sollte ich dieen Wert auch auf 0 (allenfalls null) setzen.

Negalein, 2023-06-08:
> Niemand hat Lust in Scripts auch die qualioty zu prüfen. Ob vis die quality nativ mit einbeziehen kann weiß ich nicht. Und was history und co da speichern / anzeigen ist die nächste Frage.

stimmt, so hab ich das nicht gesehen.
Da hast du mMn vollkommen recht.

Negalein, 2023-06-08:
> Nun werden parameter wenn sie 'null' liefern und das objekt existiert auf 0 gesetzt

kann man das Update schon installieren?

nkleber78, 2023-06-08:
@Negalein Wenn du von meinem Github repos nkleber78/ioBroker.fronius installieren willst dann ja. Tester sind immer willkommen :)

Negalein, 2023-06-08:
> Wenn du von meinem Github repos nkleber78/ioBroker.fronius installieren willst dann ja. Tester sind immer willkommen :)

werde dann berichten, wenns finster ist. :)

Negalein, 2023-06-08:
> Nun werden parameter wenn sie 'null' liefern und das objekt existiert auf 0 gesetzt.

funktioniert noch nicht.

Adapter zeigt noch 40 W
[image]

WebIF zeigt 0 W
[image]

Soll ich ein Issue auch auf nkleber78/ioBroker.fronius aufmachen, oder reicht es hier?

nkleber78, 2023-06-08:
@Negalein das passt hier. Ich verstehe es nur gerade nicht was da passiert. Hast du mir bitten den debug log? Dann kann ich versuchen das zu reproduzieren...

Negalein, 2023-06-08:
um 20:56 steht
`2023-06-08 20:56:20.601  - [34mdebug[39m: fronius.0 (1504358) API Objekt P_PV is null, object site.P_PV will be set to 0!` aber er macht es nicht.
Soll ich den DP mal löschen und vom Adapter neu anlegen lassen?

nkleber78, 2023-06-08:
DP neu anlegen hilft nicht. Aber die richtige instanz des adapters zu verwenden bei asynchronen prozessen hilft... Bitte nochmals meine Github version installieren. In meiner Simulation sieht es jetzt besser aus...

Negalein, 2023-06-08:
> In meiner Simulation sieht es jetzt besser aus...

YES, 0 W 😀 

[image]

nkleber78, 2023-06-09:
Issue geschlossen. Wird beim nächsten release integriert

closed 2023-06-09 by nkleber78

mcm1957, 2023-06-09:
planned for 2.0.2

Streifi89, 2023-10-08:
Ich hab heute das neue Update installiert, bei mir bleibt das Problem leider bestehen:
P_PV laut API liefert:

Site" : {
            "E_Day" : 8925,
            "E_Total" : 13737369,
            "E_Year" : 7773993.5,
            "Meter_Location" : "grid",
            "Mode" : "meter",
            "P_Akku" : null,
            "P_Grid" : 200.90000000000001,
            "P_Load" : -200.90000000000001,
            **"P_PV" : null,**
            "rel_Autonomy" : 0,
            "rel_SelfConsumption" : null

In den Objekten wird derzeit noch ein alter Wert angezeigt:
[image]

Restart des Adapters brachte keinen Erfolg. Ich update gerade aber noch meinen SYMO 10.0.3M auf den aktuellsten Stand - kann mir aber nicht vorstellen, dass das was bringen soll.

Objektbaum löschen hat auch nichts gebracht - ich habe mein Backup für Version 1.x wieder eingespielt :(

nkleber78, 2023-10-09:
@Streifi89 Bitte stell den Adapter mal auf debug mode (ohne Neustart) und lass das mal für 10min laufen. Dann kannst du mir bitte mal den log hochladen. Lt. code sollte dieser fall korrekt zu einem 0 Wert führen...

Streifi89, 2023-10-09:
Hab ich gemacht, ist jetzt blöd, weil mit der alten Adapter-Version der Wert schon auf 0 gesetzt war. Macht aber nichts, denn P_GRID wird ebenso nicht mehr mit 2.0.1 aktualisiert :( 

{
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 232,
               "E_Day" : 10674,
               "E_Total" : 13748039,
               "E_Year" : 7784669,
               "P" : 0
            }
         },
         "Site" : {
            "E_Day" : 10674,
            "E_Total" : 13748039,
            "E_Year" : 7784669,
            "Meter_Location" : "grid",
            "Mode" : "meter",
            "P_Akku" : null,
            "P_Grid" : 215.40000000000001,
            "P_Load" : -215.40000000000001,
            "P_PV" : null,
            "rel_Autonomy" : 0,
            "rel_SelfConsumption" : null
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2023-10-09T20:29:37+02:00"
   }
}

Es kommen auch neue Werte durch den Adapter:

Response to http://192.168.0.8/solar_api/v1/GetPowerFlowRealtimeData.fcgi: {"Body":{"Data":{"Inverters":{"1":{"DT":232,"E_Day":10674,"E_Total":13748039,"E_Year":7784669,"P":0}},"Site":{"E_Day":10674,"E_Total":13748039,"E_Year":7784669,"Meter_Location":"grid","Mode":"meter","P_Akku":null,"P_Grid":212.4,"P_Load":-212.4,"P_PV":null,"rel_Autonomy":0,"rel_SelfConsumption":null},"Version":"12"}},"Head":{"RequestArguments":{},"Status":{"Code":0,"Reason":"","UserMessage":""},"Timestamp":"2023-10-09T20:32:06+02:00"}}

Aber der Datenpunkt wird nicht aktualisiert. Und da vermute ich das Hauptproblem, dass die Datenpunkte nicht aktualisiert werden. 

20:22 habe ich das Update gemacht. Seitdem werden diverse Datenpunkte nicht mehr aktualisiert:

[image]

Das gilt auch für:
fronius.0.powerflow.rel_Autonomy
fronius.0.powerflow.rel_SelfConsumption
fronius.0.powerflow.P_PV
fronius.0.powerflow.P_Grid

[iobroker.2023-10-09.log.txt](https://github.com/iobroker-community-adapters/ioBroker.fronius/files/12849412/iobroker.2023-10-09.log.txt)

nkleber78, 2023-10-09:
Hallo @Streifi89 irgendwie glaub ich dass hier der Objektbaum bei dir nicht sauber ist. Powerflow gibt es im neuen Adapter 2.0 nicht mehr. Die Daten sind in der "site" und laut log werden die Daten auch zurückgesetzt...
2023-10-09 20:24:08.312  - [34mdebug[39m: fronius.0 (727) API Objekt P_Akku is null, object site.P_Akku will be set to 0!
2023-10-09 20:24:08.312  - [34mdebug[39m: fronius.0 (727) API Objekt P_PV is null, object site.P_PV will be set to 0!
2023-10-09 20:24:08.312  - [34mdebug[39m: fronius.0 (727) API Objekt rel_SelfConsumption is null, object site.rel_SelfConsumption will be set to 0!
kannst du das bitte nochmals prüfen? Bitte auch mit dem aktuellen Adapter arbeiten denn 2.0.1 hatte bugs in diesem Bereich die dann in der 2.0.2 gefixt wurden

Streifi89, 2023-10-10:
Du hast recht - ARGH! Sorry, nehme dann alles zurück.  Stand aber auch nicht explizit im Changelog :( Danke trotzdem!

nkleber78, 2023-10-10:
Alles klar. Bin auch froh wenn der Adapter funktioniert wie er soll :)