iobroker-community-adapters/ioBroker.fronius #374: Cannot convert undefined or null to object

geigl, 2023-10-05:
TypeError:  Cannot convert undefined or null to object    at hasOwnProperty  (<anonymous>)    at fillDataObject  (/opt/iobroker/node_modules/iobroker.fronius/main.js:1173:41)    at  Timeout._onTimeout  (/opt/iobroker/node_modules/iobroker.fronius/main.js:1231:13)    at  listOnTimeout (node:internal/timers:571:11)    at process.processTimers  (node:internal/timers:512:7)
--
error | uncaught exception: Cannot convert undefined or null to object

fronius.0 | 2230549

nkleber78, 2023-10-09:
@geigl kannst du hier etwas mehr info liefern? Wann tritt das auf (regelmäßig oder nur sporadisch z.b. beim abschalten des Inverters ...)

geigl, 2023-10-11:
Seit die Protoollierung auf Debug steht ist kein Fehler mehr aufgetreten.

geigl, 2023-12-14:
2023-12-14 19:33:50.449  - error: fronius.0 (84015) uncaught exception: Cannot convert undefined or null to object
--
2023-12-14 19:33:50.457  - error: fronius.0 (84015) TypeError: Cannot convert undefined or null to object
at hasOwnProperty ()
at fillDataObject (/opt/iobroker/node_modules/iobroker.fronius/main.js:1173:41)
at Timeout._onTimeout (/opt/iobroker/node_modules/iobroker.fronius/main.js:1231:13)
at listOnTimeout (node:internal/timers:571:11)
at process.processTimers (node:internal/timers:512:7)
2023-12-14 19:33:50.457  - error: fronius.0 (84015) Cannot convert undefined or null to object
2023-12-14 19:33:50.460  - error: fronius.0 (84015) uncaught exception: Cannot convert undefined or null to object
2023-12-14 19:33:50.460  - error: fronius.0 (84015) TypeError: Cannot convert undefined or null to object
at hasOwnProperty ()
at fillDataObject (/opt/iobroker/node_modules/iobroker.fronius/main.js:1173:41)
at Timeout._onTimeout (/opt/iobroker/node_modules/iobroker.fronius/main.js:1231:13)
at listOnTimeout (node:internal/timers:571:11)
at process.processTimers (node:internal/timers:512:7)
2023-12-14 19:33:50.461  - error: fronius.0 (84015) Cannot convert undefined or null to object
2023-12-14 19:33:50.966  - info: fronius.0 (84015) terminating
2023-12-14 19:33:50.967  - warn: fronius.0 (84015) Terminated (UNCAUGHT_EXCEPTION): Without reason
2023-12-14 19:33:51.607  - error: host.raspberrypi instance system.adapter.fronius.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
2023-12-14 19:33:51.609  - info: host.raspberrypi Restart adapter system.adapter.fronius.0 because enabled
2023-12-14 19:34:21.635  - info: host.raspberrypi instance system.adapter.fronius.0 started with pid 86569
2023-12-14 19:34:23.649  - info:  fronius.0 (86569) starting. Version 2.0.2 in  /opt/iobroker/node_modules/iobroker.fronius, node: v18.19.0,  js-controller: 5.0.17
2023-12-14 19:34:23.970  - info: fronius.0 (86569) The current system configuration is not up to date. Settings are updated and adapter restarted!
2023-12-14 19:34:23.987  - info: host.raspberrypi stopInstance system.adapter.fronius.0 (force=false, process=true)
2023-12-14 19:34:23.991  - info: fronius.0 (86569) Got terminate signal TERMINATE_YOURSELF
2023-12-14 19:34:24.039  - info: host.raspberrypi stopInstance system.adapter.fronius.0 send kill signal
2023-12-14 19:34:24.493  - info: fronius.0 (86569) terminating
2023-12-14 19:34:24.495  - info: fronius.0 (86569) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
2023-12-14 19:34:25.100  - info: host.raspberrypi instance system.adapter.fronius.0 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
2023-12-14 19:34:27.066  - info: host.raspberrypi instance system.adapter.fronius.0 started with pid 86585
2023-12-14 19:34:29.058  - info:  fronius.0 (86585) starting. Version 2.0.2 in  /opt/iobroker/node_modules/iobroker.fronius, node: v18.19.0,  js-controller: 5.0.17
2023-12-14 19:38:10.162  - info: admin.0 (76373) ==> Connected system.user.admin from ::ffff:192.168.25.244

geigl, 2024-03-15:
Der Fehler kommt zu unterschiedlichen Zeitpunkten, ob der Inverter dabei einen Neustart macht kann ich nicht beantworten, aktiv von mit wird keiner ausgelöst. Der Fehler tritt erst seit der Anhebung des Adapters auf Verion 2.x.x auf.

Wäre es möglich, den alten Adapter mit einem neuen Namen zu versehen, um beide Versionen zu installieren? Ich könnte dann den Symo Hybrid mit der alten Version betreiben und den Gen24 mit V.2.x.x!

nkleber78, 2024-03-15:
Hallo @geigl 
magst du mal die Version von meinem Repo installieren und testen ob das Problem jetzt weg ist?
https://github.com/nkleber78/ioBroker.fronius

geigl, 2024-03-16:
Hallo

Habe die Version von Github installiert. Ich melde mich bei dir in einigen Tagen ob der Fehler noch auftritt

 

LG

Thomas

 

https://github.com/nkleber78/ioBroker.fronius

geigl, 2024-03-19:
Hallo Norbert,
der Fehler tritt nicht mehr auf.
Vielen lieben Dank für deine Arbeit am Adapter

geigl, 2024-07-03:
> Hallo @geigl magst du mal die Version von meinem Repo installieren und testen ob das Problem jetzt weg ist? https://github.com/nkleber78/ioBroker.fronius

mcm1957, 2024-07-24:
> > Hallo @geigl magst du mal die Version von meinem Repo installieren und testen ob das Problem jetzt weg ist? https://github.com/nkleber78/ioBroker.fronius
> 
> Hallo Norbert, es gibt keine Fehlermeldungen mehr. Deine Version hat die selbe Versionsnummer wie die aktuelle Stable-Version. Kannst du deine mit einer neuen Nummer in das Stable-Repo überleiten? LG Thomas

Installationen von GitHub haben prinzipiell keine definierte Versionsnummer. Bei eine GitHub Installation ist die Releaseangabe irrelevant. Normalerweise wird eine die Versionsnummer erst im Zuge des Releases auf npm festgelegt.

Und zur Sache selbst:
Release 2.1.1 ist unterwegs ins LATEST Repo. Bitte testen und um Feedback. Codemäßig sollte sie idents zur Version von @nkleber78 sein. DANKE für den Fixe und den PR (der bei mir leider untergegangen ist - daher die Verzögerung).

mcm1957, 2024-07-24:
fixed with 2.1.1

geigl, 2024-07-25:
Teste ich gerne, ich gebe dann in ein paar Tagen eine Rückmeldung

geigl, 2024-07-29:
Ich habe in den letzten Tagen keine Fehlermeldungen mehr erhalten.
Danke für die tolle Arbeit.

mcm1957, 2024-07-29:
DANKE fürs Feedback.
Ok, dann schließe ich hier mal.
Bitte melden wenn doch noch was nicht funktioniert. Oder ein neues Issue mit akteullen Daten öffnen.

closed 2024-07-29 by mcm1957