iobroker-community-adapters/ioBroker.fronius #375: Update auf 3.28.1-3 - Keine Datenübertragung mehr

jonaslinhose, 2023-10-08:
Ich habe den Fronius Wechselrichter auf die Softwareversion 3.28.1-3 aktualisiert. Seit dem werden keine Daten mehr übertragen.

Wechselrichter: Fronius Symo 8.2-3-M

jonaslinhose, 2023-10-08:
Ich habe den Datalogger auf dem Wechselrichter neu gestartet, seit dem habe ich in den Logs die Fehlermeldung:

fronius.0 | 2023-10-08 09:38:11.233 | error | IP is not a Fronius inverter

geigl, 2023-10-08:
Ich habe diese Fehlermeldung (neben anderen) ebenfalls.
Fronius Symo Hybrid 5-0-3-S

nkleber78, 2023-10-09:
Was passiert bei einem Adapter restart? GGF. im Debug mode starten und prüfen was dort ausgegeben wird.
Was liefert die API auf der IP die beim Adapter angegeben wurde?

nkleber78, 2023-10-09:
API calls sind in der Readme dokumentiert. Der check der API wird regelmäßig ausgeführt. Daher nehme ich an dass das Problem jetzt permanent gelöst ist wenn es am Tag funktioniert

jonaslinhose, 2023-10-10:
Eine Fehlermeldung habe ich in der Log nicht mehr, aber Daten werden weiterhin nicht übertragen...

nkleber78, 2023-10-10:
Bitte poste die log files im debug mode wenn der Adapter neu gestartet wird

geigl, 2023-10-10:
aktuell keine Fehlermeldung, hier aber das LOG
<html><body>

2023-10-10 09:29:40.618  - info: host.raspberrypi "system.adapter.fronius.0" enabled
--
2023-10-10 09:29:40.685  - info: host.raspberrypi instance system.adapter.fronius.0 started with pid 2315681
2023-10-10 09:29:42.755  - debug: fronius.0 (2315681) Redis Objects: Use Redis connection: 127.0.0.1:9001
2023-10-10 09:29:42.812  - debug: fronius.0 (2315681) Objects client ready ... initialize now
2023-10-10 09:29:42.815  - debug: fronius.0 (2315681) Objects create System PubSub Client
2023-10-10 09:29:42.817  - debug: fronius.0 (2315681) Objects create User PubSub Client
2023-10-10 09:29:42.927  - debug: fronius.0 (2315681) Objects client initialize lua scripts
2023-10-10 09:29:42.936  - debug: fronius.0 (2315681) Objects connected to redis: 127.0.0.1:9001
2023-10-10 09:29:43.003  - debug: fronius.0 (2315681) Redis States: Use Redis connection: 127.0.0.1:9000
2023-10-10 09:29:43.125  - debug: fronius.0 (2315681) States create System PubSub Client
2023-10-10 09:29:43.127  - debug: fronius.0 (2315681) States create User PubSub Client
2023-10-10 09:29:43.254  - debug: fronius.0 (2315681) States connected to redis: 127.0.0.1:9000
2023-10-10 09:29:43.701  - info:  fronius.0 (2315681) starting. Version 2.0.2 in  /opt/iobroker/node_modules/iobroker.fronius, node: v18.18.0,  js-controller: 5.0.12
2023-10-10 09:29:43.900  - debug: fronius.0 (2315681) requestType https:// is not working! Now trying with http://
2023-10-10 09:29:44.002  - debug:  fronius.0 (2315681) Response to  http://192.168.25.180/solar_api/v1/GetActiveDeviceInfo.cgi?DeviceClass=System:   {"Body":{"Data":{"Inverter":{"1":{"DT":99}},"Meter":{"1":{"DT":-1,"Serial":"2131035562"}},"Ohmpilot":{},"SensorCard":{},"Storage":{"1":{"DT":-1,"Serial":"28340625"}},"StringControl":{}}},"Head":{"RequestArguments":{"DeviceClass":"System"},"Status":{"Code":0,"Reason":"","UserMessage":""},"Timestamp":"2023-10-10T09:29:43+02:00"}}
2023-10-10 09:29:44.003  - debug:  fronius.0 (2315681) Current result of System deviceINFO:  {"Inverter":{"1":{"DT":99}},"Meter":{"1":{"DT":-1,"Serial":"2131035562"}},"Ohmpilot":{},"SensorCard":{},"Storage":{"1":{"DT":-1,"Serial":"28340625"}},"StringControl":{}}
2023-10-10 09:29:44.005  - debug: fronius.0 (2315681) The current system configuration is up to date
2023-10-10 09:29:44.048  - debug:  fronius.0 (2315681) Response to  http://192.168.25.180/solar_api/v1/GetLoggerInfo.cgi:  {"Body":{"LoggerInfo":{"CO2Factor":0.5299999713897705,"CO2Unit":"kg","CashCurrency":"EUR","CashFactor":0.13600000739097595,"DefaultLanguage":"en","DeliveryFactor":0.5,"HWVersion":"2.4D","PlatformID":"wilma","ProductID":"fronius-hybrid","SWVersion":"1.28.1-3","TimezoneLocation":"Vienna","TimezoneName":"CEST","UTCOffset":7200,"UniqueID":"239.25145"}},"Head":{"RequestArguments":{},"Status":{"Code":0,"Reason":"","UserMessage":""},"Timestamp":"2023-10-10T09:29:44+02:00"}}
2023-10-10 09:29:44.110  - debug:  fronius.0 (2315681) Response to  http://192.168.25.180/solar_api/v1/GetLoggerInfo.cgi:  {"Body":{"LoggerInfo":{"CO2Factor":0.5299999713897705,"CO2Unit":"kg","CashCurrency":"EUR","CashFactor":0.13600000739097595,"DefaultLanguage":"en","DeliveryFactor":0.5,"HWVersion":"2.4D","PlatformID":"wilma","ProductID":"fronius-hybrid","SWVersion":"1.28.1-3","TimezoneLocation":"Vienna","TimezoneName":"CEST","UTCOffset":7200,"UniqueID":"239.25145"}},"Head":{"RequestArguments":{},"Status":{"Code":0,"Reason":"","UserMessage":""},"Timestamp":"2023-10-10T09:29:44+02:00"}}
2023-10-10 09:29:44.126  - debug: fronius.0 (2315681) Object creation will be done for 1 times
2023-10-10 09:29:44.408  - debug:  fronius.0 (2315681) Response to  http://192.168.25.180/solar_api/v1/GetInverterInfo.cgi:  {"Body":{"Data":{"1":{"CustomName":"Dr. Thomas  Geigl","DT":99,"ErrorCode":0,"PVPower":5660,"Show":1,"StatusCode":7,"UniqueID":"1123205"}}},"Head":{"RequestArguments":{},"Status":{"Code":0,"Reason":"","UserMessage":""},"Timestamp":"2023-10-10T09:29:44+02:00"}}
2023-10-10 09:29:44.433  - debug:  fronius.0 (2315681) Response to  http://192.168.25.180/solar_api/v1/GetOhmPilotRealtimeData.cgi?Scope=System:   {"Body":{"Data":{}},"Head":{"RequestArguments":{"DeviceClass":"OhmPilot","Scope":"System"},"Status":{"Code":0,"Reason":"","UserMessage":""},"Timestamp":"2023-10-10T09:29:44+02:00"}}
2023-10-10 09:29:44.486  - debug:  fronius.0 (2315681) Response to  http://192.168.25.180/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DeviceId=1&DataCollection=3PInverterData:   {"Body":{"Data":{"IAC_L1":{"Unit":"A","Value":2.04},"IAC_L2":{"Unit":"A","Value":2.04},"IAC_L3":{"Unit":"A","Value":2.04},"UAC_L1":{"Unit":"V","Value":230.9},"UAC_L2":{"Unit":"V","Value":229.3},"UAC_L3":{"Unit":"V","Value":231.8}}},"Head":{"RequestArguments":{"DataCollection":"3PInverterData","DeviceClass":"Inverter","DeviceId":"1","Scope":"Device"},"Status":{"Code":0,"Reason":"","UserMessage":""},"Timestamp":"2023-10-10T09:29:44+02:00"}}
2023-10-10 09:29:44.599  - debug:  fronius.0 (2315681) Response to  http://192.168.25.180/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DeviceId=1&DataCollection=CommonInverterData:   {"Body":{"Data":{"DAY_ENERGY":{"Unit":"Wh","Value":1792},"DeviceStatus":{"ErrorCode":0,"LEDColor":0,"LEDState":0,"MgmtTimerRemainingTime":-1,"StateToReset":false,"StatusCode":7},"FAC":{"Unit":"Hz","Value":50.02},"IAC":{"Unit":"A","Value":6.12},"IDC":{"Unit":"A","Value":0.75},"PAC":{"Unit":"W","Value":1416},"SAC":{"Unit":"VA","Value":1416},"TOTAL_ENE [... cut ...] usCode":7,"UniqueID":"1123205"}}},"Head":{"RequestArguments":{},"Status":{"Code":0,"Reason":"","UserMessage":""},"Timestamp":"2023-10-10T09:29:54+02:00"}}
2023-10-10 09:29:54.457  - debug:  fronius.0 (2315681) Response to  http://192.168.25.180/solar_api/v1/GetOhmPilotRealtimeData.cgi?Scope=System:   {"Body":{"Data":{}},"Head":{"RequestArguments":{"DeviceClass":"OhmPilot","Scope":"System"},"Status":{"Code":0,"Reason":"","UserMessage":""},"Timestamp":"2023-10-10T09:29:54+02:00"}}

</body>
</html>

nkleber78, 2023-10-10:
Die daten die geliefert werden passen. Somit kann ich mir nicht vorstellen dass die Objekte nicht upgedated werden. Welche datenpunkte prüfst du da? Schau dir mal P_PV im folder site an.

jonaslinhose, 2023-10-10:
Die Daten scheinen jetzt unter **fronius.0.site** und nich tmehr unter **fronius.0.powerflow**.

nkleber78, 2023-10-10:
Ja so ist das auch im Readme dokumentiert…

jonaslinhose, 2023-10-10:
oh sorry.. das hatte ich nicht gesehen. Ich hatte die ChangeLogs durchgeschaut und eigentlich auch kein Update des Adapters gemacht, deshalb hatte ich das nicht in Verbindung gebracht.

nkleber78, 2023-10-10:
Können wir dann das issue schließen?

jonaslinhose, 2023-10-13:
von meiner Seite aus kann er geschlossen werden... vielen Dank noch mal und sorry für die Arbeit

nkleber78, 2023-10-13:
@jonaslinhose Alles klar, danke
@geigl dass die Meldung ggf. beim Start des Adapters mal auftaucht kann schon sein. Trotzdem sollte das ganze auch ohne debug modus sauber laufen... Sollte trotzdem noch was sein dann bitte einfach hier dazufügen, auch wenn schon geschlossen

closed 2023-10-13 by nkleber78