iobroker-community-adapters/ioBroker.fronius #431: IP is not a Fronius Inverter

geigl, 2024-01-16:
Erhalte immer wieder die Meldung, dass mein Fronius Symo Hybrid kein Fronius Inverter ist.

`<html><body>

fronius.0 | 2024-01-16 14:56:27.179 | error | IP is not a Fronius inverter
-- | -- | -- | --
fronius.0 | 2024-01-16 14:56:25.944 | info | starting. Version 2.0.2 in /opt/iobroker/node_modules/iobroker.fronius, node: v18.19.0, js-controller: 5.0.17
fronius.0 | 2024-01-16 14:56:21.381 | info | Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
fronius.0 | 2024-01-16 14:56:21.380 | info | terminating
fronius.0 | 2024-01-16 14:56:20.877 | info | Got terminate signal TERMINATE_YOURSELF
fronius.0 | 2024-01-16 14:56:20.798 | info | The current system configuration is not up to date. Settings are updated and adapter restarted!

</body>
</html>`

nkleber78, 2024-01-16:
Das Thema passiert in der Regel wenn ein Reboot des Adapters ausserhalb der Betriebszeiten des Inverters passiert. Interessant ist dass er eine Änderung der Konfiguration erkennt. Kann es sein dass hier entweder ein Update gemacht wurde oder sonst irgendwas im Umfeld sich geändert hat? Da er die erkennung periodisch macht (incl. Prüfung der configuration) löst sich das problem von selbst. Ev. könnte man überlegen den Error nicht beim ersten Versuch bei der automatischen erkennung zu machen...

geigl, 2024-01-16:
Reboot des Adapters habe ich nicht gemacht, aber heute wurde das Unigate und die Batteriesteuerung getauscht. Die Fehlermeldung habe ich aber vorher schon öfters gehabt. Ich beobachte einmal, ob sie nur wie von Dir oben beschrieben auftritt, ansonsten schreibe ich wieder hier.

Danke für die rasche Antwort

closed 2024-03-19 by geigl