iobroker-community-adapters/ioBroker.fronius #443: Additional data points for MPPT AC Power

MNEWI, 2024-03-11:
As I am having an West/East System installed, devided on the 2 MPPT's it would be helpful to have the actual AC Power per MPPT as data points.
The information is already in the System, if you press on the Fronius Main Screen indicating the total power on the most right button.
It will show you the actual AC "Ausgangsleistung" per MPPT.

This inormation is required as a data point.

Cheers,
Michael

nkleber78, 2024-03-11:
Is that datapoint availlable on the REST API? Can you share a screenshot where this info is shown? I cannot find this information, only the DC power per MPPT

MNEWI, 2024-03-12:
The information is not given in the Rest-API, nor in the Modbus values.
Or at least I have not found them.

The info available is on the screen of the Fronius Inverter itself.
Tomorrow I will post some fotos.

mcm1957, 2024-03-12:
Ähmm
If the Info ist not available at the api ... do you expect that iobroker should read the display of the inverter?  

😉 😉 😉

MNEWI, 2024-03-12:
Of cours not, question is whether the api content can be extended ?