iobroker-community-adapters/ioBroker.fronius #460: Adapter not starting after upgrading to 2.1.0

mi3dke, 2024-05-22:
Hi,
I just installed the upgrade to 2.1.0 from 2.0.2. Everything was working before.
This was the only installation today. All adapters, node etc. have been updated earlier already on a regular base 
After the installation the adapter is not connecting. 
[image]
In the log there is an error-message.
[image]
But I have admin 6.13.16 installed and running. (since longer already and rebooted etc. ) 
[image]
Also the frameworks are updated.
[image]
I tried to downgrad to 2.0.2 but no success. Not starting and also showing the adapter size with 0MB. 

As the adapter is sending data to influx I can see it stopped exactly when I did the update so it is pretty obvious, the upgrade of the adapter lead to the problem. And the root cause seems to be a faulty detection of a version dependency with admin.

Any idea?

Thanks
Michael

mi3dke, 2024-05-23:
I tried different things but finally it got really messy so that the installation of the adapter worked but did not show up in instances anymore ... I decided to restore a regular backup that was luckily only an hour old and that worked, ie. after the restore was finished I was able to upgrade to version 2.1.0 without a problem.
Not sure what went wrong with the upgrade attempt I did earlier.

mcm1957, 2024-05-23:
Adapter has not been started as admin dependencies did not match. This check is not perfomed by adapter but by js-controller which logs the message too. So this is never a issue of the adapter.

Why js-controller reported admin to be older than installed cannot be investigated as it seems to work now. And this would also be a issue of admin / js-controller. Maybe an iob upload goit lost somewhere.

closed 2024-05-23 by mcm1957

mi3dke, 2024-05-23:
Thanks for the explanation, Martin.
This could help next time ...

mcm1957, 2024-05-23:
One addition:
If using a multihost system the admin dependency must be fulfilled at ALL hosts. 

In this case the error message is not optimal.
See https://github.com/ioBroker/ioBroker.js-controller/issues/2762