iobroker-community-adapters/ioBroker.fronius #498: lookup table for status code (and other) missing

lu-as, 2024-09-06:
**Describe the bug**  
PV inverter says "status code: 7" in IOBroker, which probably means "everything is fine". I'd like to see a table that translates these values in "OK", "warning", "error" etc.
Same question for "Show" and "ErrorCode"

**To Reproduce**  
Steps to reproduce the behavior:
In IOBroker, go to Objects>fronius>0>inverter>1>StatusCode. A value like "7" is shown, but without description.

**Expected behavior**  
I'd like to see a table (probably here in documentation) that points out which state means what.

**Screenshots & Logfiles**  

**Versions:**  
 - Adapter version: 2.1.1
 - JS-Controller version: 6.0.11
 - Node version: 18.17.1
 - Operating system: RaspBian 11

**Additional context**

mcm1957, 2024-09-06:
Do you have a table of values and corresponding description?
Or a link to such a docu?

lu-as, 2024-09-07:
No, I didn't find this.
What came to my mind is that 7 is 111 in binary, so this could be a 3-bit-code. In [Fronius ModBus documentation](https://www.fronius.com/~/downloads/Solar%20Energy/Operating%20Instructions/gen24-modbus-api-external-docs.zip) I find something that might support this idea: 

`"Customer" event severity level bitfield  bit 0 -> Error  bit 1 -> Warning  bit 2 -> InfoThis bitfield is composed as a superposition for the severity-levels of all currently active events on view level "Customer".If a bit is set it means that currently there is at least one active event with this severity level. Multiple bits can be set at once indicating several active events with different severity-levels.`

This might go together with errorcode in Fronius IOBroker adapter, which is 0 in my case.

Oh, and in https://manuals.fronius.com/html/4204102649/de.html#0_m_0000014565, I find a table that mentions "operating codes" from 1 to 8, where 7 would point out an error. But the inverter's web interface just says "status: running".

nkleber78, 2024-09-07:
@lu-as 
These codes are already translated to messages. The InverterErrorState translates the ErrorCode and the InverterState translates the StatusCode
[image]

lu-as, 2024-09-08:
Oh great, thank you for providing the complete overview!
Up to now, my Grafana dashboard shows the status of SMA inverters, mapping SMA's status codes in intuitively comprehensible colours. Your table helps me to integrate Fronius states in the same scheme.

closed 2024-09-08 by lu-as