iobroker-community-adapters/ioBroker.fronius #509: SOC ändern/schreiben

bjoern77, 2024-10-06:
Moin
Erstmal vielen Dank für die tolle Arbeit an diesem Adapter.
Bei mir nähert sich der erste Winter mit dem Fronius und meiner neuen Photovoltaikanlage.
Ich habe mich etwas in das Batteriemanagement eingelesen und würde gerne den SOCmin dynamisch ändern.
Leider kann man im Adapter den Datenpunkt SOCmin nur lesen.
Den bisher einzigen Weg zum beschreiben der SOC habe ich nur über Modbus gedunden. Was aber scheinbar sehr komplex ist.
Nun kam gerade Fronius 1.33.7-1 raus. Hier darf auch der normale Kunde die SOC Werte ändern, nicht nur der Technican.
Währe es jetzt möglich, den SOC Wert in dem Adapter schreibbar zu definieren? Das wäre der Hammer.
Vielen Dank für die Mühe und Gruss
Björn

nkleber78, 2024-10-06:
Hallo @bjoern77 
Leider ist das nicht möglich. Die REST API welche die daten per http abholt kann keine Daten modifizieren. Dies ist eine read-only API. Eine modifikation der Daten kann nur über Modbus gemacht werden, ist aber auch dort nicht so kompliziert. Du musst hierzu nur das Register 40358_StorCtl_Mod /*Activate hold/discharge/charge storage control mode  Bitfield value bit 0: DISCHARGE bit 1: CHARGE*/ richtig setzen. Dazu noch das 40365_OutWRte /*Percent of max charge rate  For force charging it must a negativ value*/ und 40365_InWRte /* Percent of max discharge rate*/ setzen Wenn man sich da mal etwas eingelesen hat ist das schon machbar
gruß Norbert

mcm1957, 2024-10-06:
Da das REST Api keine Änderungen zulässt kann / muss hier geschlossen werden.
Sorry

closed 2024-10-06 by mcm1957

bjoern77, 2024-10-08:
MoinDanke für die Info.Ich habe es geschaft kontakt zum Smart Meter per Modbus zu bekommen (ID200).Aber der Wechselrichter (ID1) bleibt stumm :-(Danke und GrussBjörn